displaylink-rpm/displaylink-rpm #218: Stopped working after update - Fedora 36 Workstation Gnome 42.2

hahndouglasf, 2022-06-17:
I recently started using Fedora Workstation 36 (about 2 weeks ago), it was working fine with: HP USB TravelDock Rev:B, but in today's last update, it stopped working!

Identification link:
https://support.hp.com/br-pt/product/hp-usb-c-travel-dock/10127474/product-info

We look forward to the update...

Crashdummyy, 2022-06-18:
You need evdi 1.11 to work which got merged in #207 .

AltF02, 2022-06-20:
Any idea when a github release will be created for 5.6.0?

Crashdummyy, 2022-06-20:
The CI/CD publishes them the moment someone creates a github tag/release.

I couldnt wait either and compiled it myself ( well without the i386 target ).
If youre crazy enough to trust some random guy in github you can use my [RPM File](https://nextcloud.codepotatoes.de/s/T9KFeSg8gEZe2pT) as well 😜 

Compiling it yourself is however not a problem.
If you have a look at the [SpecFile](https://github.com/displaylink-rpm/displaylink-rpm/blob/master/displaylink.spec) I guess you might only be missing the libdrm-devel package.

You could just install that,
remove all i386 references in the makefile,
type `make github-release`

hahndouglasf, 2022-06-20:
@Crashdummyy 

I downloaded your package and installed the same, it is working perfectly.
Thanks

hahndouglasf, 2022-06-20:
For those who are using Fedora Workstation 36 evdi 1.11 is needed, it will work perfectly.

musthafak, 2022-07-13:
You can use the [official Ubuntu display link driver](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu) for Fedora. It is working fine for me.

elguero, 2022-07-22:
A new release has been tested on F36, F36 KDE, Rocky 8 and Rocky 9.  Give that a try and see if your issue is resolved.

elguero, 2022-08-12:
I believe that this has been resolved.  Closing for now due to lack of response.

closed 2022-08-12 by elguero