displaylink-rpm/displaylink-rpm #221: Fedora 36, displaylink-driver.service: Failed with result 'exit-code'.

btk-42, 2022-07-04:
Hello, 

I installed display link on my fedora (kde plasma) and I get that error:
Process: 25045 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

Then I did research and I did that thing: [https://github.com/displaylink-rpm/displaylink-rpm/issues/48#issuecomment-357425447](url)
I used the newest version of evdi which is 1.11.0 
> 
    create /usr/src/evdi-1.5.0 if it doesn't exist already (1.5.0 is the latest release)
    download latest evdi-release https://github.com/DisplayLink/evdi/releases
    unzip it
    cd into unzipped evdi-folder
    sudo make
    cd into (created) modules-folder. it contains the needed dkms.conf and other useful files
    sudo mv * /usr/src/evdi-1.5.0/
    sudo dkms add -m evdi -v 1.5.0
    sudo dkms build -m evdi -v 1.5.0
    sudo dkms install -m evdi -v 1.5.0

After that I'm not getting that error message but when I plug usb c cable to the laptop, the screen goes black. And I'm able to log in to the system by terminal after using shortcut CTRL + ALT + F3

alex20465, 2022-07-05:
Same here fedora-36 workstation.

btk-42, 2022-07-07:
Hello, 

It crashed after one day.

elguero, 2022-07-22:
There is a new release out on this project.  You should try cleaning everything out that you have installed and try the latest package for F36.

btk-42, 2022-07-23:
> There is a new release out on this project. You should try cleaning everything out that you have installed and try the latest package for F36.

Hi, it doesn't work. When I'm trying use external monitor the settings reload.

https://user-images.githubusercontent.com/106232568/180600208-ebd3859e-cfc2-4045-98d6-cafba94a8c4d.mp4

elguero, 2022-07-23:
Do you have any logs?  For instance, `dmesg`, `systemctl status displaylink-driver`.

btk-42, 2022-07-23:
> Do you have any logs? For instance, `dmesg`, `systemctl status displaylink-driver`.

Sure, here are: 
[displaylink-driver.log](https://github.com/displaylink-rpm/displaylink-rpm/files/9173934/displaylink-driver.log)
[dmesg.log](https://github.com/displaylink-rpm/displaylink-rpm/files/9173935/dmesg.log)

elguero, 2022-07-23:
This is Fedora 36 KDE x86_64, correct?  You installed only the RPM from this repo?

Please provide `lsusb -v` and `modinfo evdi`.  It looks like DisplayLink manager is core dumping and that software is closed source.

What hardware do you have?  Can you try not having anything else plugged into your dock?

btk-42, 2022-07-23:
Fedora 36 KDE x86_64 it is correct. I downloaded package from github and instilled by rpm -i p_name.rpm
I had plugged two usb cables to the dock station. I have lenovo t14s ryzen 5 pro + lenovo thinkpad hybrid usb c

[modinfoe.log](https://github.com/displaylink-rpm/displaylink-rpm/files/9174447/modinfoe.log)
[lsusb.log](https://github.com/displaylink-rpm/displaylink-rpm/files/9174446/lsusb.log)

closed 2022-09-03 by btk-42