displaylink-rpm/displaylink-rpm #228: Screen doesn't lock on timer with v5.6.1 on Fedora

CryptoKiddies, 2022-08-19:
I have a potential security issue. Using a Yoga i9 Lenovo w/ Fedora 36 & Gnome 42.3, the latest driver is preventing the autolock configured from my Gnome settings from running when my Asus portable monitor is connected. It is not an issue when the external monitor is disconnected. I can still manually lock and sleep my computer.

elguero, 2022-11-29:
Can you provide some logs?  Does `dmesg` provide anything?  Maybe `journalctl -xu displaylink-driver`?  I would even check if SELinux is doing something, `sudo grep "displaylink" /var/log/audit/audit.log`.

CryptoKiddies, 2022-12-01:
This last happened 3 or 4 kernel updates ago, once or twice, so I think we can close this unless others have reported the same issue.