displaylink-rpm/displaylink-rpm #229: Support Fedora 37

ebousse, 2022-09-26:
Fedora 37 is arriving [in about one month](https://fedorapeople.org/groups/schedule/f-37/f-37-key-tasks.html), and the Fedora 37 beta is already available and used by quite a handful of people. I'm hoping it will be possible to have a working displaylink-rpm build. Thanks for the work done and good luck!

noavolution, 2022-11-16:
It is very important for me too. I tried to build the driver myself but I didn't succeed.

simcax, 2022-11-17:
+1 - just upgraded and now don't have any displays on my docking station :-( 
Tried building from latest main, but it is of course the same result (I should have realised :-) )

Reventlow, 2022-11-17:
Would be great with fedora 37 support :-)

dominikzogg, 2022-11-18:
I suggest todo it like that in the meanwhile :

Use the fedora 36 version and manually build the master branch of evdi

https://github.com/DisplayLink/evdi/issues/384#issuecomment-1301739514

milesje, 2022-11-18:
I've gotten this rpm to build and work using
`make devel` and then `make github-release`
This produces an rpm that installs and then the dock will work as long as it is plugged in during boot. Plug-N-Play is hit and miss, some times it works and sometimes it doesn't. There are some issues when a monitor (TV in my case) goes into a sleep mode and will not be found after powering the monitor (TV) back on. On a few occasions it appears to be found but I will be logged out of the system automatically. I'm assuming this behavior is due to the `devel` branch of the evdi being used.

martijn-sch, 2022-11-29:
Would be nice, now I am stuck on Fedora 36 😢

elguero, 2022-11-29:
I have been working on some patches based on the `devel` branch of the EVDI driver since it is unknown as to when Synaptics is going to release a newer version of their driver.

I have a couple of patches completed and compiling tests are working at least.  Thinking this through though. Some hardware testing will be next.  We have to worry about not just Fedora 37 but make sure we do not break other versions that are supported.

sudopluto, 2022-12-03:
@elguero 

please ping if you want some guinea pigs to help test on hardware, would be happy to with my lenovo dock

cristobal87, 2022-12-20:
@elguero 

I also volunteer myself to do some hardware testing. I have an Asus ZenScreen which was working great on Fedora 36 with Kernel 5.18. Now on Fedora 37 on Kernel 6.0.12-300 it doesnt work. 

I have tried building evdi from the devel and master branches, but I keep getting an error related to drm.
On master branch:
```
/home/cristobal/evdi_temp/main/evdi/module/evdi_drv.h:24:10: fatal error: drm/drm_irq.h: No such file or directory
   24 | #include <drm/drm_irq.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```

And on devel branch:
```
evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory 
 6 | #include <libdrm/drm.h>
   |          ^~~~~~~~~~~~~~
```

I have checked and libdrm is installed and working. I am a medium level linux user, but I have no idea how to fix this. 

Any help is very much appreciated.

cheers!

dominikzogg, 2022-12-20:
@cristobal87 please check my comment https://github.com/displaylink-rpm/displaylink-rpm/issues/229#issuecomment-1319910803

cristobal87, 2022-12-20:
@dominikzogg thank you very much for the response, sadly I tried to compile it on my fedora 36 version, following your steps on comment #384 and still got the same results. Since I just upgraded to 37, I no longer have a version on fedora on kernel 5.xx i am assuming that is the main issue.
Here is the output I get on the console, in case it gives you any insight.

```
[cristobal@fedora evdi-1.12.0]$ sudo make
make -C /lib/modules/6.0.5-200.fc36.x86_64/build M=$PWD
make[1]: Entering directory '/usr/src/kernels/6.0.5-200.fc36.x86_64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.2.1 20220819 (Red Hat 12.2.1-2)
  You are using:           gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
  CC [M]  /usr/src/evdi/module/evdi_drv.o
In file included from /usr/src/evdi/module/evdi_drv.c:20:
/usr/src/evdi/module/evdi_drv.h:24:10: fatal error: drm/drm_irq.h: No such file or directory
   24 | #include <drm/drm_irq.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:249: /usr/src/evdi/module/evdi_drv.o] Error 1
make[1]: *** [Makefile:1856: /usr/src/evdi/module] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.0.5-200.fc36.x86_64'
make: *** [Makefile:71: module] Error 2
[cristobal@fedora evdi-1.12.0]$ 

```

cristobal87, 2022-12-21:
Well, I kept digging around online and after a lot of trials I found this response from somebody on reddit:

[This post](https://www.reddit.com/r/Fedora/comments/yxkm3w/fedora_37_anybody_know_how_to_get_displaylink_to/)
Code:
```
cd ~/Downlaods 
git clone https://github.com/DisplayLink/evdi
cd evdi-devel/evdi/module/
cp * /usr/src/evdi-1.12.0/
sudo dnf install dkms
sudo dkms build -m evdi -v 1.12.0 --force
sudo dkms install -m evdi -v 1.12.0
sudo reboot
```
and it worked like a charm. didnt even have to reboot and my Asus ZenScreen came back to life.
and in case somebody is wondering I am using Fedora 37
Linux fedora 6.0.13-300.fc37.x86_64

hope this helps somebody else.

cheers!

lennertcc, 2023-01-01:
> Well, I kept digging around online and after a lot of trials I found this response from somebody on reddit:
>
> [This [...]

Worked for me too, only used `--force` with the `dkms install` command because it resulted in a 'nothing to do' message without it.
And afterwards I had to run the steps for Fedora secure boot again (from the readme: https://github.com/displaylink-rpm/displaylink-rpm#secure-boot-on-fedora).

dominikzogg, 2023-01-16:
Hi @sreber84

Have you tried the new release? https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v5.6.1-1
it even works with the 6.1 kernels

ParadoxGuitarist, 2023-03-06:
FWIW - I'm running F27 and the new rpm package is working just fine for me. I think this can be closed.

sreber84, 2023-03-22:
Can confirm that https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v5.6.1-1 is working fine with Fedora 37 and kernel 6.1. With 6.2 compilation is failing though. But will need to spend some more time to understand what is happening.

ParadoxGuitarist, 2023-03-22:
Yeah the driver fails with the 6.2 vs 6.1 kernel, but I'd argue that's outside of the scope of the issue for the ticket. 

Likewise, Fedora 38 is almost here and might need a build, but I wouldn't ask them to keep this ticket open for that as well. 

But it's not my repo, so the maintiners are free to do whatever. I just wanted to let @dominikzogg that it was working as expected. (at least for a while)