displaylink-rpm/displaylink-rpm #231: Support kernel 6

matej-oliva, 2022-10-30:
Fedora 36
Linux fedora 6.0.5-201.fsync.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Oct 28 20:47:27 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

/var/log/displaylink/displaylink.log
Creating symlink /var/lib/dkms/evdi/1.12.0/source -> /usr/src/evdi-1.12.0
Sign command: /lib/modules/5.19.14-201.fsync.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=5.19.14-201.fsync.fc36.x86_64 all INCLUDEDIR=/lib/modules/5.19.14-201.fsync.fc36.x86_64/build/include KVERSION=5.19.14-201.fsync.fc36.x86_64 DKMS_BUILD=1...
Signing module /var/lib/dkms/evdi/1.12.0/build/evdi.ko
Cleaning build area...

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.14-201.fsync.fc36.x86_64/kernel/drivers/gpu/drm/evdi/
depmod.....

After recent update displaylink has stopped working, please add kernel 6 support.

RandieM, 2022-11-01:
Just in case it helps someone, it works for me, when I use the evdi module from the devel branch here: https://github.com/DisplayLink/evdi.

matej-oliva, 2022-11-03:
I downloaded the devel branch of evdi and used the Makefile inside. It said `Can't build drm/drm.h: No such file or directory #6`, so I installed libdrm-devel. Then the Makefile worked, but unfortunately even after reboot, the displaylink is still not working for me.

george-barsa, 2022-11-03:
I am by no means an expert but I managed to get this working by copying the files from
`{?}/evdi/module/`
to 
`/usr/src/evdi-1.12.0/`
after building.  Then I used
`sudo dkms build -m evdi -v 1.12.0 --force`
and 
`sudo dkms install -m evdi -v 1.12.0`

I did get an error with encryption, but it still worked after a reboot.

SimonSchwendele, 2022-11-04:
> Can't build drm/drm.h: No such file or directory #6

`sudo dnf install libdrm-devel` ;)

matej-oliva, 2022-11-04:
it worked! Thank you very much for your help!

lennertcc, 2022-11-07:
> I am by no means an expert but I managed to get this working by copying the files from `{?}/evdi/module/` to `/usr/src/evdi-1.12.0/` after building. Then I used `sudo dkms build -m evdi -v 1.12.0 [...]

Worked for me too. Didn't get an error with encryption and no need for reboot. I did have to run the steps for Fedora secure boot again (from the readme: https://github.com/displaylink-rpm/displaylink-rpm#secure-boot-on-fedora)

drando, 2022-11-09:
Same here as @lennertcc worked for me without rebooting. Just signed the module for secure boot and worked straight away.

aakolov, 2022-12-30:
Can anyone please help a noob?

This is what I see when trying to build from current master (Fedora 37):

```
make[1]: Entering directory '/var/tmp/displayport9QqCjI/displaylink-1.12.0/evdi-1.12.0/library'
gcc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64 -m32 -march=i386   -c -o evdi_lib.o evdi_lib.c
In file included from /usr/include/features.h:514,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdint.h:26,
                 from /usr/lib/gcc/x86_64-redhat-linux/12/include/stdint.h:9,
                 from evdi_lib.c:5:
/usr/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-32.h: No such file or directory
    7 | # include <gnu/stubs-32.h>
      |           ^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [<builtin>: evdi_lib.o] Error 1
make[1]: Leaving directory '/var/tmp/displayport9QqCjI/displaylink-1.12.0/evdi-1.12.0/library'
error: Bad exit status from /var/tmp/displayportWuvHX2/rpm-tmp.zy4V1J (%build)
```

I understand this error is somehow connected to evdi but is seems to b a different error than discussed above. What shall I do? Any advise highly appreciated.

FunkyRobot73, 2023-02-01:
Thank you...  I bought this adapter over a year ago and finally got it to work on My Fedora 36 laptop today.  Thank you.