displaylink-rpm/displaylink-rpm #232: install fail on Rocky Linux 9.1 , root cause identified

commandline-be, 2023-01-08:
* Fedora / CentOS / **Rocky Linux**
  * Version 9.1
  * Kernel version `(uname -a)` Linux machine 5.14.0-162.6.1.el9_1.0.1.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Nov 28 18:44:09 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`
  Building module:
Cleaning build area...
make -j8 KERNELRELEASE=5.14.0-162.6.1.el9_1.0.1.x86_64 all INCLUDEDIR=/lib/modules/5.14.0-162.6.1.el9_1.0.1.x86_64/build/include KVERSION=5.14.0-162.6.1.el9_1.0.1.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.0-162.6.1.el9_1.0.1.x86_64 (x86_64)
Consu lt /var/lib/dkms/evdi/1.12.0/build/make.log for more information.
  
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
DKMS make.log for evdi-1.12.0 for kernel 5.14.0-162.6.1.el9_1.0.1.x86_64 (x86_64)
zo 08 jan 2023 23:00:43 CET
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.12.0/build SUBDIRS=/var/lib/dkms/evdi/1.12.0/build SRCROOT=/var/lib/dkms/evdi/1.12.0/build CONFIG_MODULE_SIG= -C /lib/modules/5.14.0-162.6.1.el9_1.0.1.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-162.6.1.el9_1.0.1.x86_64'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;	
  ....
  /var/lib/dkms/evdi/1.12.0/build/evdi_gem.c:15:10: fatal error: linux/dma-buf-map.h: No such file or directory
   15 | #include <linux/dma-buf-map.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:295: /var/lib/dkms/evdi/1.12.0/build/evdi_gem.o] Error 1
make[2]: *** Waiting for unfinished jobs....

Main cause appears to be the absence of linux/dma-buf-map.h in Rocky Linux 9.1 packages kernel-headers and kernel-devel

It is unclear how to work around this

Try any way of intalling from rpm or from source for the latest build.

commandline-be, 2023-01-08:
Here the modified evdi_gem.c file which compiles with iosys-map.h, sorry, to this day no clue how to use git
----

// SPDX-License-Identifier: GPL-2.0-only
/*
 * Copyright (C) 2012 Red Hat
 * Copyright (c) 2015 - 2020 DisplayLink (UK) Ltd.
 *
 * This file is subject to the terms and conditions of the GNU General Public
 * License v2. See the file COPYING in the main directory of this archive for
 * more details.
 */

#include <linux/sched.h>
#include <linux/version.h>
#if KERNEL_VERSION(5, 18, 0) <= LINUX_VERSION_CODE
#elif KERNEL_VERSION(5, 11, 0) <= LINUX_VERSION_CODE
#include <linux/iosys-map.h>
#endif
#if KERNEL_VERSION(5, 16, 0) <= LINUX_VERSION_CODE || defined(EL9)
#include <drm/drm_prime.h>
#include <drm/drm_file.h>
#elif KERNEL_VERSION(5, 5, 0) <= LINUX_VERSION_CODE || defined(EL8)
#else
#include <drm/drmP.h>
#endif
#include "evdi_drm_drv.h"
#include "evdi_params.h"
#include <linux/shmem_fs.h>
#include <linux/dma-buf.h>
#include <drm/drm_cache.h>

#if KERNEL_VERSION(5, 16, 0) <= LINUX_VERSION_CODE || defined(EL9)
MODULE_IMPORT_NS(DMA_BUF);
#endif

#if KERNEL_VERSION(5, 11, 0) <= LINUX_VERSION_CODE || defined(EL8)
static int evdi_prime_pin(struct drm_gem_object *obj);
static void evdi_prime_unpin(struct drm_gem_object *obj);

static const struct vm_operations_struct evdi_gem_vm_ops = {
	.fault = evdi_gem_fault,
	.open = drm_gem_vm_open,
	.close = drm_gem_vm_close,
};

static struct drm_gem_object_funcs gem_obj_funcs = {
	.free = evdi_gem_free_object,
	.pin = evdi_prime_pin,
	.unpin = evdi_prime_unpin,
	.vm_ops = &evdi_gem_vm_ops,
	.export = drm_gem_prime_export,
	.get_sg_table = evdi_prime_get_sg_table,
};
#endif

static bool evdi_was_called_by_mutter(void)
{
	char task_comm[TASK_COMM_LEN] = { 0 };

	get_task_comm(task_comm, current);

	return strcmp(task_comm, "gnome-shell") == 0;
}

static bool evdi_drm_gem_object_use_import_attach(struct drm_gem_object *obj)
{
	if (!obj || !obj->import_attach)
		return false;

[... 386 lines cut ...]
#if KERNEL_VERSION(5, 11, 0) <= LINUX_VERSION_CODE || defined(EL8)
static int evdi_prime_pin(struct drm_gem_object *obj)
{
	struct evdi_gem_object *bo = to_evdi_bo(obj);

	return evdi_pin_pages(bo);
}

static void evdi_prime_unpin(struct drm_gem_object *obj)
{
	struct evdi_gem_object *bo = to_evdi_bo(obj);

	evdi_unpin_pages(bo);
}
#endif

struct sg_table *evdi_prime_get_sg_table(struct drm_gem_object *obj)
{
	struct evdi_gem_object *bo = to_evdi_bo(obj);
	#if KERNEL_VERSION(5, 10, 0) <= LINUX_VERSION_CODE || defined(EL8)
		return drm_prime_pages_to_sg(obj->dev, bo->pages, bo->base.size >> PAGE_SHIFT);
	#else
		return drm_prime_pages_to_sg(bo->pages, bo->base.size >> PAGE_SHIFT);
	#endif
}

commandline-be, 2023-01-08:
dkms requires signed with MOK, closing as most likely fixed

commandline-be, 2023-01-09:
reopened pending maintainer feedback

elguero, 2023-01-10:
Thank you for your help.  I have been working on patches for a bit now and apologize for the delay.  I just committed them after I was able to test the built rpm on a Rocky 9.0 system today.  I upgraded it from 9.0 to 9.1 and everything is working still.

Regarding the Wiki page you added, thank you for the time you put into that.  From my experience, I have not needed to modify the system unit file and I am using multiple USB adapters.

I am running in a secure boot environment as well.  I only needed to enroll the MOK key once on the system when I first set it up.  After that, DKMS takes care of signing the newly built modules on kernel upgrades.  I have not only Displaylink (EVDI module) that was signed but also NVIDIA kernel modules that were built for the newer kernel and signed when I did the kernel update today.  After a reboot, everything was working without any extra effort.

elguero, 2023-01-10:
Thank you for your help.  I have been working on patches for a bit now and apologize for the delay.  I just committed them after I was able to test the built rpm on a Rocky 9.0 system today.  I upgraded it from 9.0 to 9.1 and everything is still working.

Regarding the Wiki page you added, thank you for the time you put into that.  From my experience, I have not needed to modify the system unit file and I am using multiple USB adapters.

I am running in a secure boot environment as well.  I only needed to enroll the MOK key once on the system.  After that, DKMS takes care of signing the newly built modules on kernel upgrades.  I have not only Displaylink (EVDI module) that was signed but also NVIDIA kernel modules that were built for the newer kernel and signed.  After a reboot, everything was working without any extra effort.

closed 2023-03-25 by elguero

commandline-be, 2023-07-26:
to my surprise, today i try to install 5.7 and get the errors again for which i published the work-around

  CC [M]  /usr/src/evdi-1.13.1/evdi_gem.o
/usr/src/evdi-1.13.1/evdi_gem.c:15:10: fatal error: linux/dma-buf-map.h: No such file or directory
 #include <linux/dma-buf-map.h>