displaylink-rpm/displaylink-rpm #235: Driver is broken on Kernel 6.2

MiguelGuthridge, 2023-03-25:
**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2* OR *DisplayLink 5.4* and *evdi v1.9.1*

  * Are you using the latest DisplayLink driver? Yes
    
  * Are you using the latest EVDI module? Yes
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website? Yes
    
  * Is this issue related to evdi/kernel? I think so
    
  * Is this issue related to DisplayLinkManager? Not sure
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver? Didn't upgrade driver
    

  * Fedora
  * Version 37
  * Kernel version `(uname -a)` 
```
Linux yzma 6.2.7-200.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Mar 17 16:16:00 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

When I boot my laptop using the latest Kernel 6.2, the driver doesn't work correctly. It works as expected using Kernel 6.1

1. Upgrade to Kernel 6.2
2. Reboot and observe it is broken

EdwardAirs, 2023-03-25:
Identical set-up, Identical problem ...

elguero, 2023-03-25:
Hi.  The upstream EVDI driver has some updates for 6.2 support.  They need to release a new DisplayLink driver and release the evdi driver.

elguero, 2023-04-04:
#236

MiguelGuthridge, 2023-04-12:
Is it possible to have this project grab the latest commits from the develop branch of [DisplayLink/evdi](https://github.com/DisplayLink/evdi), since that will keep this driver working more reliably.

AwkwardTech, 2023-04-22:
It looks like they just released v5.7 of the DL driver earlier this week with 6.2 kernel support:
```
C.7  Changes in 5.7 release
-----------------------------
- New driver distribution channel via Synaptics apt repository (https://www.synaptics.com/sites/default/files/Ubuntu/)
- In some cases DisplayLink installer freezes when Secure Boot is enabled. (PS-2455)
- In some cases DisplayLink evdi driver is not signed and loaded when Secure Boot is enabled. (PS-2455)
- Added preliminary support for kernel 6.2
```
@elguero @MiguelGuthridge

elguero, 2023-04-22:
Yes, they did just release on the 20th.  I started working on it last night.  Going to do some testing.

elguero, 2023-04-25:
Hey. Just finished testing this morning.

New release. Tested with Fedora 38 Live ISO and on Rocky Linux 9.

Thanks!

closed 2023-04-25 by elguero

MiguelGuthridge, 2023-04-26:
Thanks for your work, @elguero and glad to have it fixed!