displaylink-rpm/displaylink-rpm #236: Fedora 37 Kernal 6.2 Evdi 1.12.0, RPM Release isn't working.

MarvinMynx, 2023-04-04:
**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2* OR *DisplayLink 5.4* and *evdi v1.9.1*

  * Are you using the latest DisplayLink driver?
    I Downloaded the Latest fedora-37*.rpm from the Release Page

  * Are you using the latest EVDI module?
    I... Don't know? There is literally nothing that Explains what this is, how to get/build/install it. I tought it would be automatically installed with the .rpm File?

  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    No?

  * Is this issue related to evdi/kernel?
     I dont know...

  * Is this issue related to DisplayLinkManager?
     I dont know...

  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    
  * Fedora
  * 37
  * Linux fedora.tailc55b7.ts.net 6.2.8-200.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Mar 22 19:11:02 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

  * ❯ tail /var/log/displaylink/displaylink.log
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...
Key file /var/lib/dkms/mok.key not found and can't be generated, modules won't be signed

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.2.8-200.fc37.x86_64 all INCLUDEDIR=/lib/modules/6.2.8-200.fc37.x86_64/build/include KVERSION=6.2.8-200.fc37.x86_64 DKMS_BUILD=1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.8-200.fc37.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.12.0/build/make.log for more information.

  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
  There is no *kernel* in "/var/lib/dkms/evdi/"
❯ ll /var/lib/dkms/evdi/
total 0
drwxr-xr-x. 1 root root 22  4. Apr 09:53 1.12.0
❯ tail /var/lib/dkms/evdi/1.12.0/build/make.log
  ./scripts/check-local-export /var/lib/dkms/evdi/1.12.0/build/evdi_connector.o
   { echo ; echo '/var/lib/dkms/evdi/1.12.0/build/evdi_platform_dev.o: $(wildcard ./tools/objtool/objtool)' ; } >> /var/lib/dkms/evdi/1.12.0/build/.evdi_platform_dev.o.cmd
  ./scripts/check-local-export /var/lib/dkms/evdi/1.12.0/build/evdi_drm_drv.o
   { echo ; echo '/var/lib/dkms/evdi/1.12.0/build/evdi_connector.o: $(wildcard ./tools/objtool/objtool)' ; } >> /var/lib/dkms/evdi/1.12.0/build/.evdi_connector.o.cmd
   { echo ; echo '/var/lib/dkms/evdi/1.12.0/build/evdi_drm_drv.o: $(wildcard ./tools/objtool/objtool)' ; } >> /var/lib/dkms/evdi/1.12.0/build/.evdi_drm_drv.o.cmd
  ./scripts/check-local-export /var/lib/dkms/evdi/1.12.0/build/evdi_modeset.o
   { echo ; echo '/var/lib/dkms/evdi/1.12.0/build/evdi_modeset.o: $(wildcard ./tools/objtool/objtool)' ; } >> /var/lib/dkms/evdi/1.12.0/build/.evdi_modeset.o.cmd
make[1]: *** [Makefile:2025: /var/lib/dkms/evdi/1.12.0/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.2.8-200.fc37.x86_64'
make: *** [Makefile:38: all] Error 2

I Installed the Latest Release of the Fedora 37 x86 RPM and Rebooted. But the Servie isn't comming up.
Modprob isn't working.

Install Fedora 37 Workstation x86. Install Latest RPM File accordingly. Done.

I Installed the Latest Release of the Fedora 37 x86 RPM and Rebooted. But the Servie isn't comming up.
Apr 04 10:15:49 fedora.tailc55b7.ts.net systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
Apr 04 10:15:49 fedora.tailc55b7.ts.net modprobe[7150]: modprobe: FATAL: Module evdi not found in directory /lib/modules/6.2.8-200.fc37.x86_64
Apr 04 10:15:49 fedora.tailc55b7.ts.net systemd[1]: displaylink-driver.service: Control process exited, code=exited, status=1/FAILURE
Apr 04 10:15:49 fedora.tailc55b7.ts.net systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
Apr 04 10:15:49 fedora.tailc55b7.ts.net systemd[1]: Failed to start displaylink-driver.service - DisplayLink Driver Service.
Modprob isn't working.

fabrizio-ottati, 2023-04-22:
The same applies for Fedora 36. Reverting to kernel 6.1.15 solved the issue for me at the moment.

elguero, 2023-04-25:
Hey.  Just finished testing this morning.

New release.  Tested with Fedora 38 Live ISO and on Rocky Linux 9.

Thanks!

closed 2023-04-25 by elguero