displaylink-rpm/displaylink-rpm #237: Please ass Fedora 38 support

jamesRUS52, 2023-04-24:
Please ass Fedora 38 support

I tried to use rpm for fedora 37 and got an errors

 james  ~  sudo systemctl start displaylink-driver                                                                                                                                                                                        1 
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xeu displaylink-driver.service" for details.
 james  ~  systemctl status displaylink-driver.service                                                                                                                                                                                    1 
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: activating (auto-restart) (Result: exit-code) since Mon 2023-04-24 10:20:39 MSK; 3s ago
    Process: 8561 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
        CPU: 4ms
 james  ~  journalctl -xeu displaylink-driver.service                                                                                                                                                                                     3 
░░ 
░░ The unit displaylink-driver.service has entered the 'failed' state with result 'exit-code'.
апр 24 10:22:24 fedora systemd[1]: Failed to start displaylink-driver.service - DisplayLink Driver Service.
░░ Subject: Ошибка юнита displaylink-driver.service
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ Произошел сбой юнита displaylink-driver.service.
░░ 
░░ Результат: failed.
апр 24 10:22:29 fedora systemd[1]: displaylink-driver.service: Scheduled restart job, restart counter is at 33.
░░ Subject: Назначен автоматический перезапуск юнита
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ Назначен автоматический перезапуск юнита displaylink-driver.service, так как для него был задан 
░░ параметр Restart=.
апр 24 10:22:29 fedora systemd[1]: Stopped displaylink-driver.service - DisplayLink Driver Service.
░░ Subject: Завершена остановка юнита displaylink-driver.service.
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ Процесс остановки юнита displaylink-driver.service был завершен.
апр 24 10:22:29 fedora systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
░░ Subject: Начинается запуск юнита displaylink-driver.service
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ Начат процесс запуска юнита displaylink-driver.service.
апр 24 10:22:29 fedora modprobe[10737]: modprobe: FATAL: Module evdi not found in directory /lib/modules/6.2.11-300.fc38.x86_64
апр 24 10:22:29 fedora systemd[1]: displaylink-driver.service: Control process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ An ExecStartPre= process belonging to unit displaylink-driver.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
апр 24 10:22:29 fedora systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit displaylink-driver.service has entered the 'failed' state with result 'exit-code'.
апр 24 10:22:29 fedora systemd[1]: Failed to start displaylink-driver.service - DisplayLink Driver Service.
░░ Subject: Ошибка юнита displaylink-driver.service
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ Произошел сбой юнита displaylink-driver.service.
░░ 
░░ Результат: failed.

DusterTheFirst, 2023-04-24:
@Crashdummyy's COPR works for me on Fedora Linux 37

bartwo, 2023-04-24:
@Crashdummyy I can confirm it also works on Fedora 38, great job!

jamesRUS52, 2023-04-24:
@Crashdummyy thanks a lot, it work now !!!

SimonSchwendele, 2023-04-24:
@jamesRUS52 
You should really reopen this issue again.
I forked away of this repository and altered some files.

This repo has yet to update its release!

elguero, 2023-04-25:
New release.  Tested with Fedora 38 Live ISO and on Rocky Linux 9.

Thanks!

jamesRUS52, 2023-04-26:
@Crashdummyy after first time install your package my D3100 works properly, but today it not working again.
The status of service is Inactive (dead) after startup laptop. When I try to start it manually the laptop is hanging.
After reboot it, the service not started again.
@elguero I try to install your driver and problem the same (service not started or hangs on manually start)
Do you know what may be is wrong now? Why it working after first install and not working now (after several reboot and working one day from another place without dock)

jamesRUS52, 2023-04-26:
dkms status                                                                                                                                                                            
evdi/1.13.1, 6.2.11-300.fc38.x86_64, x86_64: installed

jamesRUS52, 2023-04-26:
uname -r                                                                                                                                                                               
6.2.11-300.fc38.x86_64

sudo systemctl status displaylink-driver                                                                                                                                           
○ displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: inactive (dead)

jamesRUS52, 2023-04-26:
I have updated and reinstalled all just now, but it not helped

jamesRUS52, 2023-04-26:
The laptop is hang
only hard reset after that

jamesRUS52, 2023-04-27:
The old version not working for me now too.
I am confused.
I tried several kernels 6.2.9-6.2.12 with full reinstall displaylink-1.13.1-2
The laptop is hang when running 
modprob evdi

jamesRUS52, 2023-04-28:
Messages after start modprob evdi in journal
may be you can see something wrong

апр 27 23:48:45 fedora sudo[8797]:    james : TTY=pts/0 ; PWD=/home/james ; USER=root ; COMMAND=/usr/sbin/modprobe evdi
апр 27 23:48:45 fedora sudo[8797]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
апр 27 23:48:45 fedora sudo[8797]: pam_unix(sudo:session): session closed for user root
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(0): EDID vendor "SDC", prod id 16737
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(0): Printing DDC gathered Modelines:
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(0): Modeline "1920x1080"x0.0  138.77  1920 1968 2000 2080  1080 1088 1096 1112 +hsync -vsync (66.7 kHz eP)
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) config/udev: removing GPU device /sys/devices/platform/evdi.1/drm/card3 /dev/dri/card3
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) config/udev: Adding drm device (/dev/dri/card3)
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) xfree86: Adding drm device (/dev/dri/card3)
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) systemd-logind: got fd for /dev/dri/card3 226:3 fd 118 paused 0
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) LoadModule: "modesetting"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Loading /usr/lib64/xorg/modules/drivers/modesetting_drv.so
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Module modesetting: vendor="X.Org Foundation"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         compiled for 1.20.14, module version = 1.20.14
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         Module class: X.Org Video Driver
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         ABI class: X.Org Video Driver, version 24.1
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) UnloadModule: "modesetting"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Unloading modesetting
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Failed to load module "modesetting" (already loaded, 0)
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(G1): using drv /dev/dri/card3
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(G1): Creating default Display subsection in Screen section
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         "Default Screen Section" for depth/fbbpp 24/32
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (==) modeset(G1): Depth 24, (==) framebuffer bpp 32
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Applying OutputClass "DisplayLink" options to /dev/dri/card3
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (**) modeset(G1): Option "SWcursor" "on"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (**) modeset(G1): Option "ShadowFB" "true"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (**) modeset(G1): Option "AccelMethod" "none"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (**) modeset(G1): Option "PageFlip" "off"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (==) modeset(G1): RGB weight 888
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (==) modeset(G1): Default visual is TrueColor
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (**) modeset(G1): glamor disabled
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(G1): ShadowFB: preferred NO, enabled YES
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(G1): Double-buffered shadow updates: off
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(G1): Output DVI-I-2-2 has no monitor section
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (==) modeset(G1): Using gamma correction (1.0, 1.0, 1.0)
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (**) modeset(G1): Display dimensions: (340, 190) mm
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (**) modeset(G1): DPI set to (76, 102)
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Loading sub module "fb"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) LoadModule: "fb"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Loading /usr/lib64/xorg/modules/libfb.so
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Module fb: vendor="X.Org Foundation"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         compiled for 1.20.14, module version = 1.0.0
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         ABI class: X.Org ANSI C Emulation, version 0.4
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Loading sub module "shadow"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) LoadModule: "shadow"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Loading /usr/lib64/xorg/modules/libshadow.so
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (II) Module shadow: vendor="X.Org Foundation"
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         compiled for 1.20.14, module version = 1.1.0
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]:         ABI class: X.Org ANSI C Emulation, version 0.4
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: (==) modeset(G1): Backing store enabled
[... cut ...]
апр 27 23:48:45 fedora /usr/libexec/gdm-x-session[3011]: xf86: found device 6
апр 27 23:48:46 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(0): EDID vendor "SDC", prod id 16737
апр 27 23:48:46 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(0): Printing DDC gathered Modelines:
апр 27 23:48:46 fedora /usr/libexec/gdm-x-session[3011]: (II) modeset(0): Modeline "1920x1080"x0.0  138.77  1920 1968 2000 2080  1080 1088 1096 1112 +hsync -vsync (66.7 kHz eP)
-- Boot 4269853f561947768c7dc32834374613 --

jamesRUS52, 2023-04-28:
Wow
It work on Wayland
I swithed to Xorg because some app does not work on wayland.

Thanks for help