displaylink-rpm/displaylink-rpm #243: Fedora 38 (and 37) rpm installation is still broken (ThinkPad)

CryptoKiddies, 2023-06-23:
These are errors I don't see in any of the past issues.

Description: error when installing _fedora-38-displaylink-1.13.1-1.x86_64.rpm_ via dnf.

  * Fedora 38 
  * Kernel version: Linux fedora 6.2.15-300.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Thu May 11 17:37:39 UTC 2023 x86_64 GNU/Linux
 * Device: Lenovo Thinkpad X1

Attempt installation with `sudo dnf install ~/Downloads/fedora-38-displaylink-1.13.1-1.x86_64.rpm --skip-broken` and immediately receive these errors:
```
Error: 
 Problem 1: package kernel-6.2.8-200.fc37.x86_64 from @System requires kernel-core-uname-r = 6.2.8-200.fc37.x86_64, but none of the providers can be installed
  - conflicting requests
 Problem 2: package kernel-modules-extra-6.2.8-200.fc37.x86_64 from @System requires kernel-uname-r = 6.2.8-200.fc37.x86_64, but none of the providers can be installed
  - conflicting requests
 Problem 3: package kernel-modules-6.2.8-200.fc37.x86_64 from @System requires kernel-modules-core-uname-r = 6.2.8-200.fc37.x86_64, but none of the providers can be installed
  - conflicting requests
```

CryptoKiddies, 2023-06-23:
Strange behavior... Gnome software app shows everything up to date, but a `sudo dnf update` fixed most of the issues. The only left to do is build the evdi module and sign. However, @elguero i didn't need to do this with Fedora 36 rpm package. Unless I'm forgetting something?

CryptoKiddies, 2023-06-23:
I think it wasn't recognized because of secureboot. Once the module was signed and registered, my commands ran successfully.

closed 2023-06-23 by CryptoKiddies

elguero, 2023-06-24:
Hi.  I would suggest that you don't use `--skip-broken`.

You shouldn't have to download the evdi module separately if you have `dkms` installed.  Your usage of `--skip-broken` may be causing you issues.

CryptoKiddies, 2023-07-24:
Thanks @elguero - I had tried both with and without the flag. It seems extraneous to the issue, which resolves once the signing is successful.