displaylink-rpm/displaylink-rpm #244: installation fails on Fedora 38 evdi-dkms conflict

mxschumacher, 2023-08-12:
Hello all,

my goal is to use an external display via Display Port on my Fedora 38 linux machine. It's using Wayland.

Installing displaylink-rpm for Fedora 38 fails due to a conflict with "evdi-dkms-1.13.1-1.noarch".

**Please see the full output of the installation command below.**

"lsmod | grep evdi" returns nothing

According to systemd, no display link driver is currently running
 
 `sudo rpm -qa | grep displaylink` does not return anything for me
 

            /var/lib/dkms/evdi
            /usr/libexec/displaylink
            /usr/src/evdi-*
            
            The folders above all don't contain evdi

     Both locations below are also empty:
     
            /usr/lib/modules/6.4.7-200.fc38.x86_64/weak-updates
            /usr/lib/modules/6.4.7-200.fc38.x86_64/extra
            
            

          Reboot
          Reinstall the RPM package -->

  * Fedora
  * Version: 38
  * Kernel version: Linux fedora **6.4.7-200**.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Jul 27 20:01:18 UTC 2023 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
  => both of these files don't exist on my machine

Download 
**fedora-38-displaylink-1.13.1-2.x86_64.rpm**  from [displaylink-rpm release page](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v5.7.0-1)
then run "sudo dnf install fedora-38-displaylink-1.13.1-2.x86_64.rpm"

```
Total size: 5.2 M
Installed size: 13 M
Is this ok [y/N]: y
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Error: Transaction test error:
  file /usr/src/evdi-1.13.1/Kconfig from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/LICENSE from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/Makefile from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/dkms.conf from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_connector.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_cursor.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_cursor.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_debug.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_debug.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_drm.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_drm_drv.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_drm_drv.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_encoder.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_fb.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_gem.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_i2c.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_i2c.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_ioc32.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_modeset.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_painter.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_params.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_params.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_platform_dev.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_platform_dev.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_platform_drv.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_platform_drv.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_sysfs.c from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
  file /usr/src/evdi-1.13.1/evdi_sysfs.h from install of displaylink-1.13.1-2.x86_64 conflicts with file from package evdi-dkms-1.13.1-1.noarch
```

Does somebody please have any guidance how I can proceed with the troubleshooting?

elguero, 2023-08-12:
Hi @mxschumacher ,

It looks like you have installed an RPM package from another source and not from this repo.  That package is called `evdi-dkms-1.13.1-1.noarch` and needs to be uninstalled if you plan on using the RPM from this repo.

mxschumacher, 2023-08-12:
Hello @elguero , thank you for taking the time to help. I really appreciate it.

I will try to remove it. So far, both

> dkms remove evdi/1.13.1

and

> sudo dnf remove evdi-dkms-1.13.1-1.noarch

fail to remove the package

Error message:

```
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                1/1
  Running scriptlet: evdi-dkms-1.13.1-1.noarch                                      1/1
Error! The module/version combo: evdi-1.13.1 is not located in the DKMS tree.
error: %preun(evdi-dkms-1.13.1-1.noarch) scriptlet failed, exit status 3

Error in PREUN scriptlet in rpm package evdi-dkms
  Verifying        : evdi-dkms-1.13.1-1.noarch                                      1/1

Failed:
  evdi-dkms-1.13.1-1.noarch

Error: Transaction failed
```

I will keep trying! Thanks for your help

ssaavedra, 2023-08-27:
@mxschumacher did you finally get to remove this?
Your package is failing because the PREUN script (in RPM parlance this is the script that gets run before uninstalling) is failing because you already removed evdi-1.13.1 from the DKMS tree manually. You can remove the package without running the script by using the `--noscripts` flag. (See e.g., https://unix.stackexchange.com/a/599016)

ssaavedra, 2023-08-27:
I'm gonna mark this as closed, but if you are still struggling with this, please reopen :)

closed 2023-08-27 by ssaavedra