displaylink-rpm/displaylink-rpm #245: unable to build project

mxschumacher, 2023-08-12:
Hello all,

because I was unable to install the pre-built package from the release page (see my ticket here: https://github.com/displaylink-rpm/displaylink-rpm/issues/244 ) I attempted to build from source.

Both the regular and the make-devel build fails. Please see the full log below:

```
(master) λ make github-release
                                  
CFLAGS='-m32 -march=i386' \
LDFLAGS='-m32 -march=i386' \
rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 2" --define "_daemon_version 5.7.0-61.129" --define "_version 1.13.1" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`" --define "_github 1" displaylink.spec --target=i386
Building target platforms: i386
Building for target i386
setting SOURCE_DATE_EPOCH=1687564800
Executing(%prep): /bin/sh -e /var/tmp/displayport2fSg82/rpm-tmp.OZIrwP
+ umask 022
+ cd /var/tmp/displayportDEpeV1
+ cd /var/tmp/displayportDEpeV1
+ rm -rf displaylink-1.13.1
+ /usr/bin/mkdir -p displaylink-1.13.1
+ cd displaylink-1.13.1
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ cd /var/tmp/displayportDEpeV1
+ cd displaylink-1.13.1
+ /usr/lib/rpm/rpmuncompress -x /home/mxschumacher/code/displaylink-rpm/DisplayLink_USB_Graphics_Software_for_Ubuntu_5.7.0-61.129.zip
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ chmod +x displaylink-driver-5.7.0-61.129.run
+ ./displaylink-driver-5.7.0-61.129.run --noexec --keep
Creating directory displaylink-driver-5.7.0-61.129
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.7.0-61.129  100%
+ mkdir -p evdi-1.13.1
+ cd /var/tmp/displayportDEpeV1
+ cd displaylink-1.13.1
+ /usr/lib/rpm/rpmuncompress -x /home/mxschumacher/code/displaylink-rpm/v1.13.1.tar.gz
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ cd evdi-1.13.1
+ echo 'Patch #0 (evdi-el-fixes-1.13.1.diff):'
Patch #0 (evdi-el-fixes-1.13.1.diff):
+ /usr/bin/patch --no-backup-if-mismatch -f -p1 --fuzz=0
patching file module/evdi_drm_drv.h
Hunk #1 FAILED at 18.
Hunk #2 FAILED at 26.
2 out of 2 hunks FAILED -- saving rejects to file module/evdi_drm_drv.h.rej
patching file module/evdi_gem.c
Hunk #1 FAILED at 10.
Hunk #2 FAILED at 295.
Hunk #3 FAILED at 328.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej
error: Bad exit status from /var/tmp/displayport2fSg82/rpm-tmp.OZIrwP (%prep)

RPM build errors:
    Bad exit status from /var/tmp/displayport2fSg82/rpm-tmp.OZIrwP (%prep)
make: *** [Makefile:163: i386/displaylink-1.13.1-2-github_evdi.i386.rpm] Error 1
```

I would appreciate any guidance you might have to make the build succeed. What are steps I can take for troubleshooting?

elguero, 2023-08-12:
The EVDI project had released an updated version of the EVDI drivers on their GitHub page but had not released the binary parts to their website that work with it.  The two work together.  We have been waiting for the DisplayLink drivers (binaries that come from displaylink.com) to drop.  What is happening when trying to build your own, is, you are trying to use version 1.14.1 since that is the current release on GitHub and the patches being applied were against 1.13.  That is failing because the patches were submitted upstream and are now part of their latest release.

I just went to the DisplayLink website and saw an updated version that was just released 4 days ago.  Which means we have a bit of work to do to bring out a new release here.  We need to check all code changes between versions and do some QC in order to release a newer version.

Hang tight. We will start working on an update here.

mxschumacher, 2023-08-12:
Fantastic, thanks for the explanation.

elguero, 2023-08-14:
@mxschumacher There is a new release.

It was much smoother this time around since not too much had changed between versions.

closed 2023-08-27 by ssaavedra