displaylink-rpm/displaylink-rpm #247: Display Link 1.14 fails to run service on Fedora 38

CryptoKiddies, 2023-08-31:
The displaylink driver service fails to start/enable when installing the latest package in Fedora 38. I also cannot run make in evdi repo as that fails as well.

  * Fedora 38: Linux fedora 6.4.12-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Aug 23 17:46:49 UTC 2023 x86_64 GNU/Linux

Install output:
``` bash
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                               1/1 
  Installing       : openssl-1:3.0.9-2.fc38.x86_64                                                                 1/4 
  Installing       : kernel-devel-matched-6.4.12-200.fc38.x86_64                                                   2/4 
  Installing       : dkms-3.0.11-1.fc38.noarch                                                                     3/4 
  Running scriptlet: dkms-3.0.11-1.fc38.noarch                                                                     3/4 
Created symlink /etc/systemd/system/multi-user.target.wants/dkms.service → /usr/lib/systemd/system/dkms.service.

  Installing       : displaylink-1.14.1-1.x86_64                                                                   4/4 
  Running scriptlet: displaylink-1.14.1-1.x86_64                                                                   4/4 
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xeu displaylink-driver.service" for details.
warning: %post(displaylink-1.14.1-1.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink
  Verifying        : dkms-3.0.11-1.fc38.noarch                                                                     1/4 
  Verifying        : kernel-devel-matched-6.4.12-200.fc38.x86_64                                                   2/4 
  Verifying        : openssl-1:3.0.9-2.fc38.x86_64                                                                 3/4 
  Verifying        : displaylink-1.14.1-1.x86_64                                                                   4/4 

Installed:
  displaylink-1.14.1-1.x86_64         dkms-3.0.11-1.fc38.noarch       kernel-devel-matched-6.4.12-200.fc38.x86_64      
  openssl-1:3.0.9-2.fc38.x86_64  
  ```

  * `/var/log/displaylink/displaylink.log` : 
  ``` bash
Deleting module evdi-1.14.1 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.14.1/source -> /usr/src/evdi-1.14.1
Sign command: /lib/modules/6.4.12-200.fc38.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j16 KERNELRELEASE=6.4.12-200.fc38.x86_64 all INCLUDEDIR=/lib/modules/6.4.12-200.fc38.x86_64/build/include KVERSION=6.4.12-200.fc38.x86_64 DKMS_BUILD=1...
Signing module /var/lib/dkms/evdi/1.14.1/build/evdi.ko
Cleaning build area...

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.4.12-200.fc38.x86_64/extra/
depmod....
```
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`: 
  ``` bash
...
# cmd_check_local_export /var/lib/dkms/evdi/1.14.1/build/evdi_debug.o
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_debug.o
# cmd_check_local_export /var/lib/dkms/evdi/1.14.1/build/evdi_params.o
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_params.o
# cmd_check_local_export /var/lib/dkms/evdi/1.14.1/build/evdi_sysfs.o
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_sysfs.o
# cmd_check_local_export /var/lib/dkms/evdi/1.14.1/build/evdi_encoder.o
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_encoder.o
# cmd_check_local_export /var/lib/dkms/evdi/1.14.1/build/evdi_platform_drv.o
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_platform_drv.o
# cmd_check_local_export /var/lib/dkms/evdi/1.14.1/build/evdi_platform_dev.o
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_platform_dev.o
[... 15 lines cut ...]
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_gem.o
# cmd_check_local_export /var/lib/dkms/evdi/1.14.1/build/evdi_painter.o
  ./scripts/check-local-export /var/lib/dkms/evdi/1.14.1/build/evdi_painter.o
# LD [M]  /var/lib/dkms/evdi/1.14.1/build/evdi.o
  ld -m elf_x86_64 -z noexecstack --no-warn-rwx-segments   -r -o /var/lib/dkms/evdi/1.14.1/build/evdi.o @/var/lib/dkms/evdi/1.14.1/build/evdi.mod  ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --hacks=skylake --ibt --orc --retpoline --rethunk --sls --static-call --uaccess --prefix=16  --link  --module /var/lib/dkms/evdi/1.14.1/build/evdi.o
# cmd_gen_objtooldep /var/lib/dkms/evdi/1.14.1/build/evdi.o
   { echo ; echo '/var/lib/dkms/evdi/1.14.1/build/evdi.o: $(wildcard ./tools/objtool/objtool)' ; } >> /var/lib/dkms/evdi/1.14.1/build/.evdi.o.cmd
# cmd_modules_order /var/lib/dkms/evdi/1.14.1/build/modules.order
  {   echo /var/lib/dkms/evdi/1.14.1/build/evdi.o; :; } > /var/lib/dkms/evdi/1.14.1/build/modules.order
sh ./scripts/modules-check.sh /var/lib/dkms/evdi/1.14.1/build/modules.order
make -f ./scripts/Makefile.modpost
# MODPOST /var/lib/dkms/evdi/1.14.1/build/Module.symvers
   scripts/mod/modpost       -o /var/lib/dkms/evdi/1.14.1/build/Module.symvers -T /var/lib/dkms/evdi/1.14.1/build/modules.order -i Module.symvers -e 
make -f ./scripts/Makefile.modfinal
# CC [M]  /var/lib/dkms/evdi/1.14.1/build/evdi.mod.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.14.1/build/.evdi.mod.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -funsigned-char -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -falign-functions=16 -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -DMODULE -DKBUILD_BASENAME='"evdi.mod"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.14.1/build/evdi.mod.o /var/lib/dkms/evdi/1.14.1/build/evdi.mod.c
# LD [M]  /var/lib/dkms/evdi/1.14.1/build/evdi.ko
  ld -r -m elf_x86_64 -z noexecstack --no-warn-rwx-segments --build-id=sha1  -T scripts/module.lds -o /var/lib/dkms/evdi/1.14.1/build/evdi.ko /var/lib/dkms/evdi/1.14.1/build/evdi.o /var/lib/dkms/evdi/1.14.1/build/evdi.mod.o;  true
# BTF [M] /var/lib/dkms/evdi/1.14.1/build/evdi.ko
  if [ ! -f vmlinux ]; then printf "Skipping BTF generation for %s due to unavailability of vmlinux\n" /var/lib/dkms/evdi/1.14.1/build/evdi.ko 1>&2; elif [ -n "" ] && ./scripts/is_rust_module.sh /var/lib/dkms/evdi/1.14.1/build/evdi.ko; then printf "Skipping BTF generation for %s because it's a Rust module\n" /var/lib/dkms/evdi/1.14.1/build/evdi.ko 1>&2; else LLVM_OBJCOPY="objcopy" pahole -J  --btf_base vmlinux /var/lib/dkms/evdi/1.14.1/build/evdi.ko; ./tools/bpf/resolve [... cut ...] /lib/dkms/evdi/1.14.1/build/evdi.ko; fi;
Skipping BTF generation for /var/lib/dkms/evdi/1.14.1/build/evdi.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.4.12-200.fc38.x86_64'
```

DKMS and EVDI seem to be installed. I also manually install kernel-headers and kernel-dev libraries and retired installation. I did a clean reinstall of displaylink after restarting machine. Still fails consistently with same issue. Wondering how to fix this. Any help is appreciated.

elguero, 2023-08-31:
Hi!

What does  `systemctl status displaylink-driver.service` and `journalctl -xeu displaylink-driver.service` reveal about the error from trying to start the service?

CryptoKiddies, 2023-08-31:
Hi here they are:
```
displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: activating (auto-restart) (Result: exit-code) since Thu 2023-08-31>
    Process: 7727 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILU>
        CPU: 7ms

Aug 31 15:55:45 fedora systemd[1]: Failed to start displaylink-driver.service ->
```
```
Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ An ExecStartPre= process belonging to unit displaylink-driver.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Aug 31 15:56:17 fedora systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit displaylink-driver.service has entered the 'failed' state with result 'exit-code'.
Aug 31 15:56:17 fedora systemd[1]: Failed to start displaylink-driver.service - DisplayLink Driver Service.
░░ Subject: A start job for unit displaylink-driver.service has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit displaylink-driver.service has finished with a failure.
░░ 
░░ The job identifier is 20693 and the job result is failed.
```
Not much info I'm afraid. I tried registering a fresh key and enrolled it, but that didn't seem to help.

elguero, 2023-08-31:
That was going to be my next question about whether or not you have SecureBoot enabled and the key DKMS used to sign the module enrolled.

Just to cover the basics, what output do you get from running:

`mokutil --sb-state`

`lsmod | grep evdi`

`dmesg | grep evdi`

CryptoKiddies, 2023-08-31:
`mokutil --sb-state`: SecureBoot enabled
`lsmod | grep evdi`: empty
`dmesg | grep evdi`: empty

I updated my kernel today and perhaps evdi needs an update. When I run make in the evdi devel branch, i get:
```
Package libdrm was not found in the pkg-config search path.
Perhaps you should add the directory containing `libdrm.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libdrm', required by 'virtual:world', not found
evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory
    6 | #include <libdrm/drm.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [<builtin>: evdi_lib.o] Error 1
make[1]: Leaving directory '/home/garen/projects/evdi/library'
make: *** [Makefile:11: all] Error 2
```

elguero, 2023-08-31:
Which kernel did you upgrade from?

CryptoKiddies, 2023-08-31:
Either 6.4.10 or 6.4.11

elguero, 2023-09-01:
I do not have Fedora installed on a BareMetal machine.

I created a container with Fedora 38.  I updated to the latest and installed the kernel you just updated to.  I was able to build `evdi` using dkms against that kernel and install the module.

I would try doing a clean install again.  Remove the RPM, make sure that there is nothing under `/var/lib/dkms` related to evdi.  I would make sure that under `/lib/modules/{KERNELVERSION}/extra` does not have the evdi kernel module there.

Do the following for a clean driver install:
           Find the current package installed - `sudo rpm -qa | grep displaylink`
           Remove what is returned: `sudo rpm -evh [package]`

Clear out anything left over:
```
/var/lib/dkms/evdi
/usr/libexec/displaylink
/usr/src/evdi-*

/etc/X11/xorg.conf.d/20-displaylink.conf
/etc/logrotate.d/displaylink
/usr/lib/systemd/system/displaylink-driver.service
/usr/lib/systemd/system-preset/95-displaylink.preset
/usr/lib/systemd/system-sleep/displaylink.sh
/etc/udev/rules.d/99-displaylink.rules
/usr/share/doc/displaylink
/usr/share/doc/displaylink/LICENSE

/usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz
/usr/lib/modules/`uname -r`/extra/evdi.ko.xz
```

Reboot
Reinstall the RPM package

elguero, 2023-09-01:
I do not have Fedora installed on a BareMetal machine.

I created a container with Fedora 38.  I updated to the latest and installed the kernel you just updated to.  I was able to build `evdi` using dkms against that kernel and install the module.

I would try doing a clean install again.  Remove the RPM, make sure that there is nothing under `/var/lib/dkms` related to evdi.  I would make sure that under `/lib/modules/{KERNELVERSION}/extra` does not have the evdi kernel module there.

Do the following for a clean driver install:
           Find the current package installed - `sudo rpm -qa | grep displaylink`
           Remove what is returned: `sudo rpm -evh [package]`

Clear out anything left over:
```/var/lib/dkms/evdi
/usr/libexec/displaylink
/usr/src/evdi-*

/etc/X11/xorg.conf.d/20-displaylink.conf
/etc/logrotate.d/displaylink
/usr/lib/systemd/system/displaylink-driver.service
/usr/lib/systemd/system-preset/95-displaylink.preset
/usr/lib/systemd/system-sleep/displaylink.sh
/etc/udev/rules.d/99-displaylink.rules
/usr/share/doc/displaylink
/usr/share/doc/displaylink/LICENSE

/usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz
/usr/lib/modules/`uname -r`/extra/evdi.ko.xz
```

Reboot
Reinstall the RPM package

CryptoKiddies, 2023-09-01:
I get the error `dkms is needed by displaylink-1.14.1-1.x86_64`. I think that's progress... The evdi folder in _/var/lib/dkms_ was referencing an old kernel version fyi. I'm trying to avoid dnf install, perhaps it is retrieving an out of date dep version. How should I grab the dkms lib? Thanks again.

CryptoKiddies, 2023-09-01:
It failed with the same error unfortunately.

CryptoKiddies, 2023-09-01:
Well I installed the necessary python and libdrm dev libs to build the evdi. However, I still get the same error.

ngmarchant, 2023-09-03:
I'm encountering the same issue on Fedora 38 with kernel version 6.4.13. Secure boot is disabled and I installed the released RPM.

scorpp, 2023-09-04:
Fedora 38 Workstation
- 6.4.10 doesn't work (but i believe it used to!)
- 6.4.11 works with some manual configuration
- 6.4.12 doesn't work
- 6.4.13 doesn't work

CryptoKiddies, 2023-09-20:
I'm on 6.4.15-200 now and in journalctl i see: `fedora modprobe[9028]: modprobe: ERROR: could not insert 'evdi': Key was rejected by service` @elguero

CryptoKiddies, 2023-09-20:
@elguero rpm installs without error on 6.4.15-200. However even with reboot, my ASUS ZenScreen monitor status shows no signal.

CryptoKiddies, 2023-09-20:
looks like systemctl issue
```
Sep 20 09:30:45 fedora systemd[1]: displaylink-driver.service: Control process exited, code=exited, status=1/FAILURE
Sep 20 09:30:45 fedora systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
Sep 20 09:30:45 fedora systemd[1]: Failed to start displaylink-driver.service - DisplayLink Driver Service.
Sep 20 09:30:48 fedora systemd[1]: Stopped displaylink-driver.service - DisplayLink Driver Service.
[n@fedora ~]$ sudo systemctl start displaylink-driver.service
```

no details in journalctl

ghost-72, 2023-10-04:
OS Name: Fedora Linux 38.20231003.0 (Silverblue)
OS Type: 64-bit
GNOME Version: 44.5
Windowing System: Wayland
Kernel Version: Linux 6.5.5-200.fc38.x86_64

Log Start
Txn Upgrade on /org/projectatomic/rpmostree1/fedora failed: 

error: Could not depsolve transaction; 1 problem detected:
 Problem: package displaylink-1.14.1-1.x86_64 from @commandline requires kernel-devel >= 4.15, but none of the providers can be installed
  - package kernel-devel-6.5.5-200.fc38.x86_64 from updates requires gcc, but none of the providers can be installed
  - package kernel-devel-6.2.9-300.fc38.x86_64 from fedora requires gcc, but none of the providers can be installed
  - package kernel-devel-6.2.10-300.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.2.11-300.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.2.12-300.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.2.13-300.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.2.14-300.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.2.15-300.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.3.11-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.3.12-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.3.4-201.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.3.5-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.3.6-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.3.7-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.3.8-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.10-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.11-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.12-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.13-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.14-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.15-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.4-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.6-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.7-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.8-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package kernel-devel-6.4.9-200.fc38.x86_64 from updates-archive requires gcc, but none of the providers can be installed
  - package gcc-13.2.1-1.fc38.x86_64 from updates requires glibc-devel >= 2.2.90-12, but none of the providers can be installed
  - package gcc-13.2.1-1.fc38.x86_64 from updates-archive requires glibc-devel >= 2.2.90-12, but none of the providers can be installed
  - package glibc-devel-2.37-5.fc38.i686 from updates requires glibc = 2.37-5.fc38, but none of the providers can be installed
  - package glibc-devel-2.37-5.fc38.x86_64 from updates requires glibc = 2.37-5.fc38, but none of the providers can be installed
  - package glibc-devel-2.37-5.fc38.x86_64 from updates-archive requires glibc = 2.37-5.fc38, but none of the providers can be installed
  - package glibc-devel-2.37-1.fc38.i686 from fedora requires glibc = 2.37-1.fc38, but none of the providers can be installed
  - package glibc-devel-2.37-1.fc38.x86_64 from fedora requires glibc = 2.37-1.fc38, but none of the providers can be installed
  - glibc-2.37-5.fc38.i686 from updates  does not belong to a distupgrade repository
  - glibc-2.37-1.fc38.i686 from fedora  does not belong to a distupgrade repository
  - package glibc-devel-2.37-4.fc38.x86_64 from updates-archive requires glibc = 2.37-4.fc38, but none of the providers can be installed
  - package gcc-13.0.1-0.12.fc38.x86_64 from fedora requires cpp = 13.0.1-0.12.fc38, but none of the providers can be installed
  - package gcc-13.1.1-1.fc38.x86_64 from updates-archive requires cpp = 13.1.1-1.fc38, but none of the providers can be installed
  - package gcc-13.1.1-2.fc38.x86_64 from updates-archive requires cpp = 13.1.1-2.fc38, but none of the providers can be installed
  - package gcc-13.1.1-4.fc38.x86_64 from updates-archive requires cpp = 13.1.1-4.fc38, but none of the providers can be installed
[... cut ...]
  - cannot install both glibc-2.37-5.fc38.x86_64 from updates-archive and glibc-2.37-10.fc38.x86_64 from @System
  - cannot install both cpp-13.0.1-0.12.fc38.x86_64 from fedora and cpp-13.2.1-1.fc38.x86_64 from @System
  - cannot install both cpp-13.1.1-1.fc38.x86_64 from updates-archive and cpp-13.2.1-1.fc38.x86_64 from @System
  - cannot install both cpp-13.1.1-2.fc38.x86_64 from updates-archive and cpp-13.2.1-1.fc38.x86_64 from @System
  - cannot install both cpp-13.1.1-4.fc38.x86_64 from updates-archive and cpp-13.2.1-1.fc38.x86_64 from @System
  - conflicting requests

aschreque, 2023-10-07:
I got the same. I did a clean remove of every files @elguero mentionned before reinstalling the RPM. The problem resides that the key which signed the module is not accepted. 

`77.777726] usb 2-3: new SuperSpeed USB device number 6 using xhci_hcd
[   77.790300] usb 2-3: New USB device found, idVendor=17e9, idProduct=4301, bcdDevice=36.26
[   77.790304] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   77.790305] usb 2-3: Product: USB3.0 UHD DisplayPort Adapter
[   77.790306] usb 2-3: Manufacturer: DisplayLink
[   77.790306] usb 2-3: SerialNumber: 000200122090641
[   77.798098] usb 2-3: Warning! Unlikely big volume range (=511), cval->res is probably wrong.
[   77.798100] usb 2-3: [13] FU [DisplayLink Audio Playback Volume] ch = 2, val = -8176/0/16
[   77.838497] Loading of module with unavailable key is rejected
[   83.046389] Loading of module with unavailable key is rejected
[   88.291971] Loading of module with unavailable key is rejected
[   93.543262] Loading of module with unavailable key is rejected `

Keys are present : 
`localhost dkms]$ ls
mok.key  mok.pub
`

ghost-72, 2023-10-08:
Forgive my ignorance......I'm still new to Linux

Considering the Speed at which Fedora releases it's Kernels and Versions.....
https://en.wikipedia.org/wiki/Fedora_Linux_release_history

Would it be easier for Application Support to switch from a .RPM to a Universal Containerised App such as Flatpaks or the Community Based AppImages?
https://youtu.be/IuxHidCdbPo?si=QddojEfmZQrSsxiG&t=640
https://youtu.be/Wy63jwjpNg4?si=qxnv68nUjUfJXSpo

Or do these Sandbox Enviroments cause Hardware issues?

aschreque, 2023-10-08:
> Would it be easier for Application Support to switch from a .RPM to a Universal Containerised App such as Flatpaks or the Community Based AppImages?
> 
> Or do these Sandbox Enviroments cause Hardware issues?
It's a module of the kernel : it needs to be recompile and linked against each version of the installed kernel. Packaging does not change the frequency of the operation.  A source RPM is doing the stuff.

The problem is : there is an error somewhere in signing the module or installed keys. As I have no clue on how this is done, I asked for some help.

aschreque, 2023-10-09:
ok, the answer is RTFM 

I followed the steps to sign de evdi module and it was fine : it works well

see this https://github.com/displaylink-rpm/displaylink-rpm#secure-boot-on-fedora

Thanks @elguero , sorry to not have read the README.MD

scorpp, 2023-10-10:
> Fedora 38 Workstation
> 
> * 6.4.10 doesn't work (but i believe it used to!)
> * 6.4.11 works with some manual configuration
> * 6.4.12 doesn't work
> * 6.4.13 doesn't work

ok, so today i managed to fix this with 6.5.5 kernel and evdi-1.14.1.

the problem was caused by evdi-1.13 leftovers which broke dkms and prevented it from building a module for newer kernel. this appeared to be also the problem with 6.4.10 (which used to work, but after some install/reinstall refused to).

so essentially the solution was to just clean everything in `/var/lib/dkms/evdi` which referenced 1.13.1 version and rerun `sudo dkms autoinstall`

elguero, 2024-01-07:
@CryptoKiddies Is this issue still present for you?

@reinouts I believe this is resolved in the latest release.  Support for kernel 6.6 has been added by patching a change from upstream.

elguero, 2024-04-11:
Closing this out as it would appear to be resolved.

closed 2024-04-11 by elguero