displaylink-rpm/displaylink-rpm #249: Issues with Kernel 6.6.0

PrestonSam, 2023-10-21:
### Issue
It seems that I'm unable to get the package working - it errors when I install the `fedora-38-displaylink-1.14.1-1.x86_64.rpm` file using `rpm -i $filename`

### Device:
**Fedora 38**
`Linux barnabas 6.6.0-0.rc6.20231017gt213f8915.247.vanilla.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Oct 17 04:37:56 UTC 2023 x86_64 GNU/Linux`

Logs
[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/13060603/displaylink.log)
[journalctl.log](https://github.com/displaylink-rpm/displaylink-rpm/files/13060604/journalctl.log)
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/13060605/make.log)

The errors seem to express that support has been implemented for kernel 6.5 but not yet for kernel 6.6.

GrzegorzBB, 2023-11-19:
Hi, small update from my side, recently there was a commit to devel branch of evdi ([](https://github.com/DisplayLink/evdi/commit/495e8cffbf62f6809108442999084213b014e9e0)) which added support for kernel 6.6. In order to make it work you should replace  wget command with:
`git clone https://github.com/DisplayLink/evdi.git
cd evdi/module/
sudo mkdir -p /usr/src/evdi-1.14.1/
sudo cp * /usr/src/evdi-1.14.1/
sudo dkms build -m evdi -v 1.14.1
# or
sudo dkms autoinstall`

PrestonSam, 2023-11-21:
Thanks for that update :)

Just writing to comment that this altered version continues to work on Kernel 6.7 (on my copy of Fedora 38 at least)

nathan-weinberg, 2023-12-01:
Just wanted to chime in - I'm on Fedora 38 with kernel `6.6.2-101.fc38.x86_64` - the [current released version upstream](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v5.8.0) for the displaylink-rpm stopped working for me but [this RPM](https://github.com/SimonSchwendele/displaylink-rpm/releases/tag/v1.14.1-3) from @SimonSchwendele fixed the issue

commit e417807c (2023-12-25): Add patches for newer kernels

This commit adds patches from upstream to fix kernel build issues. This should
resolve issues up to kernel 6.6 and the latest EL8/EL9 kernels.

Resolves #256, resolves #253, resolves #249
closed 2023-12-25 by elguero