displaylink-rpm/displaylink-rpm #250: Fedora 39 RPM?

webknjaz, 2023-11-07:
F39 got released today: https://fedoramagazine.org/announcing-fedora-linux-39/. Is it possible to generate a new RPM for it?

bennothommo, 2023-11-09:
The Fedora 38 driver still works - for me, at least.

webknjaz, 2023-11-09:
I haven't tried reinstalling it upon upgrade. But the previously installed systemd unit crashes not finding a kernel module or something. Haven't looked into it more.

bennothommo, 2023-11-09:
I am following the instructions for [signing the driver for secure boot](https://github.com/displaylink-rpm/displaylink-rpm#secure-boot-on-fedora), I did have a crash as well until I followed those steps.

webknjaz, 2023-11-09:
I never had to go through the secure boot thing as I don't think I have it enabled.

jtkellys, 2023-11-12:
I don't know if it's a crash, but only one of the display ports is working correctly. Any luck on figuring this out?

PR #251 (2023-11-21): Build RPMs for Fedora 39
Resolves #250.
commit 910575f1 (2023-11-21): Build RPMs for Fedora 39 (#251)

Resolves #250.
closed 2023-11-21 by elguero