displaylink-rpm/displaylink-rpm #252: [idea] Integrate Packit

webknjaz, 2023-11-08:
Packit is a service that allows continuously building+testing RPMs. To set it up, you'll need to install this GitHub App https://github.com/marketplace/packit-as-a-service, then add a config: https://packit.dev/docs/guide. You can test against a number of distros, not just Fedora.
There's a quick approval process for when you first set it up, but when all's configured, you get a CI-like thing that reports statuses in PRs + some integration possibilities for making RPMs and putting them in Copr or official repos, but that's optional (I saw in other issues that the license is the blocker).

I have experience of setting integrating it myself and there's a room on the Fedora matrix instance https://chat.fedoraproject.org/#/room/#packit:fedora.im and an IRC room on LiberaChat for questions.

ssaavedra, 2024-01-17:
We cannot use packit for several reasons, but the foremost one is licensing: even though DisplayLink is distributed with a license liberal enough to grant us the right to repackaging and redistributing the repackaged results, this is not [an Acceptable License for Fedora](https://docs.fedoraproject.org/en-US/legal/allowed-licenses), so **we cannot build on copr** on the first place.

Even though I welcome the attitude, if you're working for RH you probably shouldn't be offering us an option that seems would break RH's T&Cs 😅 

As a small piece of feedback, I have just taken a look at the docs and I don't see a clearly supported way of testing different distributions (besides [what Copr offers](https://docs.pagure.org/copr.copr/user_documentation.html#can-i-build-for-different-versions-of-fedora)), and we are currently building for other distros which don't seem to be first citizens for Packit. Is there something I'm missing? I might need better keyword searching

But in short, given the licensing constraints of this project, I believe our best chance of future integration would be with the RPMFusion toolchain, and not Fedora's, unfortunately.

elguero, 2024-11-09:
Hola!  Agree with @ssaavedra .  I do not see this as being the correct solution for this project.

closed 2024-11-09 by elguero

webknjaz, 2024-11-09:
I wonder if it'd be usable as a package called in the regular CI w/o their hosted service...