displaylink-rpm/displaylink-rpm #253: Module Build failed to install on RHEL 8.9 and RHEL 9.2 / 9.3

jodalein, 2023-11-22:
Hi,
in the latest RHEL Systems its not possible to install the module anymore.
It fails with this errors

```
cd /usr/src/evdi-1.14.1/
make
make -C /lib/modules/4.18.0-513.5.1.el8_9.x86_64/build M=$PWD
make[1]: Verzeichnis „/usr/src/kernels/4.18.0-513.5.1.el8_9.x86_64“ wird betreten
  CC [M]  /usr/src/evdi-1.14.1/evdi_fb.o
/usr/src/evdi-1.14.1/evdi_fb.c: In Funktion »evdifb_create«:
/usr/src/evdi-1.14.1/evdi_fb.c:410:17: Fehler: »struct drm_fb_helper« hat kein Element namens »fbdev«; meinten Sie »dev«?
  efbdev->helper.fbdev = info;
                 ^~~~~
                 dev
/usr/src/evdi-1.14.1/evdi_fb.c: In Funktion »evdi_fbdev_destroy«:
/usr/src/evdi-1.14.1/evdi_fb.c:470:21: Fehler: »struct drm_fb_helper« hat kein Element namens »fbdev«; meinten Sie »dev«?
  if (efbdev->helper.fbdev) {
                     ^~~~~
                     dev
/usr/src/evdi-1.14.1/evdi_fb.c:471:25: Fehler: »struct drm_fb_helper« hat kein Element namens »fbdev«; meinten Sie »dev«?
   info = efbdev->helper.fbdev;
                         ^~~~~
                         dev
/usr/src/evdi-1.14.1/evdi_fb.c: In Funktion »evdi_fbdev_init«:
/usr/src/evdi-1.14.1/evdi_fb.c:506:46: Warnung: Übergabe des Arguments 3 von »drm_fb_helper_prepare« wandelt eine Zahl in einen Zeiger um, ohne explizite Typkonvertierung [-Wint-conversion]
  drm_fb_helper_prepare(dev, &efbdev->helper, &evdi_fb_helper_funcs);
                                              ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/evdi-1.14.1/evdi_fb.c:26:
./include/drm/drm_fb_helper.h:244:20: Anmerkung: »unsigned int« erwartet, aber Argument hat Typ »struct drm_fb_helper_funcs *«
       unsigned int preferred_bpp,
       ~~~~~~~~~~~~~^~~~~~~~~~~~~
/usr/src/evdi-1.14.1/evdi_fb.c:506:2: Fehler: Zu wenige Argumente für Funktion »drm_fb_helper_prepare«
  drm_fb_helper_prepare(dev, &efbdev->helper, &evdi_fb_helper_funcs);
  ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/evdi-1.14.1/evdi_fb.c:26:
./include/drm/drm_fb_helper.h:243:6: Anmerkung: hier deklariert
 void drm_fb_helper_prepare(struct drm_device *dev, struct drm_fb_helper *helper,
      ^~~~~~~~~~~~~~~~~~~~~
/usr/src/evdi-1.14.1/evdi_fb.c:527:8: Fehler: zu viele Argumente für Funktion »drm_fb_helper_initial_config«
  ret = drm_fb_helper_initial_config(&efbdev->helper, 32);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/evdi-1.14.1/evdi_fb.c:26:
./include/drm/drm_fb_helper.h:300:5: Anmerkung: hier deklariert
 int drm_fb_helper_initial_config(struct drm_fb_helper *fb_helper);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/evdi-1.14.1/evdi_fb.c: In Funktion »evdi_fbdev_unplug«:
/usr/src/evdi-1.14.1/evdi_fb.c:564:21: Fehler: »struct drm_fb_helper« hat kein Element namens »fbdev«; meinten Sie »dev«?
  if (efbdev->helper.fbdev) {
                     ^~~~~
                     dev
/usr/src/evdi-1.14.1/evdi_fb.c:567:25: Fehler: »struct drm_fb_helper« hat kein Element namens »fbdev«; meinten Sie »dev«?
   info = efbdev->helper.fbdev;
                         ^~~~~
                         dev
make[2]: *** [scripts/Makefile.build:318: /usr/src/evdi-1.14.1/evdi_fb.o] Fehler 1
make[1]: *** [Makefile:1619: _module_/usr/src/evdi-1.14.1] Fehler 2
make[1]: Verzeichnis „/usr/src/kernels/4.18.0-513.5.1.el8_9.x86_64“ wird verlassen
make: *** [Makefile:86: module] Fehler 2
```

elguero, 2023-11-22:
Hi @jodalein ,

Yes, I just worked on that yesterday and have a fix.  The updated kernel in 8.9 and 9.3  have DRM backports in it going all the way up to 6.3.  Although, some patches from 6.3 were reverted later, which is interesting.

Anyways, I will be working on a patch for EVDI upstream and working on an updated release here to include that patch.

Thanks!

elguero, 2023-11-23:
Submitted a PR upstream for the EVDI part of this: https://github.com/DisplayLink/evdi/pull/440

Will work on applying this patch for the RPM.

commit e417807c (2023-12-25): Add patches for newer kernels

This commit adds patches from upstream to fix kernel build issues. This should
resolve issues up to kernel 6.6 and the latest EL8/EL9 kernels.

Resolves #256, resolves #253, resolves #249