displaylink-rpm/displaylink-rpm #255: Missing current Fedora version 39 support

skewty, 2023-12-03:
No rpms have been released for current version of Fedora 39.  I tried it and current kernel doesn't seem supported.

Issue #210 is more-or-less the same issue (deprecated versions should be removed and current versions added)

It is likely Fedora 39 will also have issue(s) of #247.

mcoyle1960, 2023-12-03:
Yup. Fedora 38 rpm won't install on 39 with kernel 6.6.2.

`sudo dnf install '/home/coyle/Downloads/fedora-38-displaylink-1.14.1-1.x86_64.rpm' 
[sudo] password for coyle: 
Last metadata expiration check: 2:17:19 ago on Sun 03 Dec 2023 01:22:56 PM EST.
Dependencies resolved.
========================================================================================================================
 Package                       Architecture             Version                    Repository                      Size
========================================================================================================================
Installing:
 displaylink                   x86_64                   1.14.1-1                   @commandline                   6.3 M

Transaction Summary
========================================================================================================================
Install  1 Package

Total size: 6.3 M
Installed size: 15 M
Is this ok [y/N]: y
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                1/1 
  Installing       : displaylink-1.14.1-1.x86_64                                                                    1/1 
  Running scriptlet: displaylink-1.14.1-1.x86_64                                                                    1/1 
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xeu displaylink-driver.service" for details.
warning: %post(displaylink-1.14.1-1.x86_64) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package displaylink
  Verifying        : displaylink-1.14.1-1.x86_64                                                                    1/1 

Installed:
  displaylink-1.14.1-1.x86_64  `

In the past, I'm pretty sure I see listings for both the displaylink driver and evdi, but evdi seems to be missing.

regisf, 2023-12-04:
Hi, 
I fix the same issue on my Ryzen 7 4800H laptop by installing evdi by hand.

skewty, 2023-12-04:
> Hi, I fix the same issue on my Ryzen 7 4800H laptop by installing evdi by hand.

How did you install evdi into Fedora 39?

regisf, 2023-12-04:
I was inspired by the response from reddit r/fedora about installing on fedora 38

Assuming displaylink rpm is installed (type `sudo systemctl status displaylink-driver`) 
Just
* ` git clone https://github.com/DisplayLink/evdi` 
* `cd evdi` 
*  `export CPLUS_INCLUDE_PATH="/usr/include/python3.12:$CPLUS_INCLUDE_PATH"` 
* ` sudo dnf install pybind11-devel`  
* ` make`
* `sudo make install` 

I check with `systemctl`  the driver status and I restarted the driver with `systemctl start displaylink-driver` 

Et voilà!

skewty, 2023-12-04:
So checking out "master" branch is enough? We shouldn't be checking out the "devel" branch?

Also, on Fedora 39, there seems to be additional requirements.

See `evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory` below during `make`.

```shell
$ make
CFLAGS="-isystem./include -isystem./include/uapi -Werror -Wextra -Wall -Wno-error=missing-field-initializers -Werror=sign-compare -Wmissing-prototypes -Wstrict-prototypes -Werror=discarded-qualifiers " make -C module 
make[1]: Entering directory '/home/myuser/Projects/3rd_party/evdi/module'
make -C /lib/modules/6.5.12-300.fc39.x86_64/build M=$PWD
make[2]: Entering directory '/usr/src/kernels/6.5.12-300.fc39.x86_64'
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_platform_drv.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_platform_dev.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_sysfs.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_modeset.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_connector.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_encoder.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_drm_drv.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_fb.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_gem.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_painter.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_params.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_cursor.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_debug.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_i2c.o
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi_ioc32.o
  LD [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi.o
  MODPOST /home/myuser/Projects/3rd_party/evdi/module/Module.symvers
  CC [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi.mod.o
  LD [M]  /home/myuser/Projects/3rd_party/evdi/module/evdi.ko
  BTF [M] /home/myuser/Projects/3rd_party/evdi/module/evdi.ko
Skipping BTF generation for /home/myuser/Projects/3rd_party/evdi/module/evdi.ko due to unavailability of vmlinux
make[2]: Leaving directory '/usr/src/kernels/6.5.12-300.fc39.x86_64'
make[1]: Leaving directory '/home/myuser/Projects/3rd_party/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wno-error=missing-field-initializers -Werror=sign-compare -Wmissing-prototypes -Wstrict-prototypes -Werror=discarded-qualifiers " make -C library 
make[1]: Entering directory '/home/myuser/Projects/3rd_party/evdi/library'
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64 -I../module -Werror -Wextra -Wall -Wno-error=missing-field-initializers -Werror=sign-compare -Wmissing-prototypes -Wstrict-prototypes -Werror=discarded-qualifiers  $(pkg-config --cflags-only-I libdrm)   -c -o evdi_lib.o evdi_lib.c
Package libdrm was not found in the pkg-config search path.
Perhaps you should add the directory containing `libdrm.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libdrm', required by 'virtual:world', not found
evdi_lib.c:6:10: fatal error: libdrm/drm.h: No such file or directory
    6 | #include <libdrm/drm.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [<builtin>: evdi_lib.o] Error 1
make[1]: Leaving directory '/home/myuser/Projects/3rd_party/evdi/library'
make: *** [Makefile:11: all] Error 2
```

So I did `sudo dnf install libdrm libdrm-devel`

Re-running make now completes.

mcoyle1960, 2023-12-04:
Thank you for the solution. This worked for me:

Remove displaylink first. ( _sudo dnf remove displaylink_ )

Then I build evdi per the instructions, no problems there, then reinstalled the Fedora 38 Displaylink rpm. 

During the reboot I get an error that dkms wasn't able to start, but the driver loaded and my external monitors are working. Whew. I thought I was going to have to downgrade to Fedora 38 cause I have a job coming up and I needed my external monitors!!

mcoyle1960, 2023-12-04:
Until the dkms problem is fixed, we'll probably have to reinstall after ever kernel update.

artemave, 2023-12-11:
> Thank you for the solution. This worked for me:
> 
> Remove displaylink first. ( _sudo dnf remove displaylink_ )
> 
> Then I build evdi per the instructions, no problems there, then reinstalled the Fedora 38 Displaylink rpm.

I followed this, but to no avail. displaylink-driver appears to be healthy, but plugging in external screen via usb-c still doesn't work (works with hdmi). It was working before upgrading to f39.

```
❯ sudo systemctl status displaylink-driver
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Mon 2023-12-11 17:14:54 CET; 19s ago
    Process: 8472 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
   Main PID: 8478 (DisplayLinkMana)
      Tasks: 13 (limit: 36937)
     Memory: 17.7M
        CPU: 943ms
     CGroup: /system.slice/displaylink-driver.service
             └─8478 /usr/libexec/displaylink/DisplayLinkManager

Dec 11 17:14:54 fedora systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
Dec 11 17:14:54 fedora systemd[1]: Started displaylink-driver.service - DisplayLink Driver Service.
```

mcoyle1960, 2023-12-11:
I am no wizard at this, but I managed to hack around and put together a solution for myself that worked in Fedora 39 and a fresh install of Rawhide. I put it all in a text doc which you can [download here](https://www.jennaudio.com/nextcloud/index.php/s/Nsqaqmy53n2CFYi).

artemave, 2023-12-12:
I followed your instructions again (thanks for compiling them!), but, yeah, that's what I was already doing and so the result was still negative. The only thing to add is that after reboot the driver is not active.

```
❯ sudo systemctl status displaylink-driver
○ displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: inactive (dead)
[1]    5632 exit 3     sudo systemctl status displaylink-driver
```

However I am able to `sudo systemctl start displaylink-driver` and after that it appears healthy (but not working).

mcoyle1960, 2023-12-12:
> I followed your instructions again...
Ugh. Sorry. Tell us more about your system. Gnome, X11, Nvidia drivers (which version)? 

Also, I seem to rememeber something about removing old versions of edvi.
So look in /var/lib/dkms/evdi and delete anything older than 1.14.1

artemave, 2023-12-12:
> Tell us more about your system. Gnome, X11, Nvidia drivers (which version)?

ThinkPad T14s Gen 3 (AMD Ryzen 7 PRO 6850U with Radeon Graphics), i3wm (X11).

Let me know if I missed some obvious piece of information - I am rather inexperienced in troubleshooting these kind of things.

> Also, I seem to remember something about removing old versions of edvi

There's only 1.14.1 there.

mciancia, 2023-12-21:
Same issue for me on fedora 38, kernel 6.6.7
Tried installing latest evdi, but didn't help

mcoyle1960, 2023-12-21:
I did my 6.6.7 update this morning on Fedora 39. It took a couple tries. 

Try rebooting every step of the way:

Uninstall displaylink - reboot
Build and install evdi - reboot
install dislaylink rpm - reboot. (or force reboot if like me you lockup hard, but give it a couple minutes incase it's rebuilding any modules.)

That got me working.

jeffreygrant717, 2023-12-29:
@regisf & @mcoyle1960 y'all are amazing smart hackers, got this working on Fedora 39:

> $ cat /etc/redhat-release 
Fedora release 39 (Thirty Nine)

> Hostname:		unix-laptop
uptime:			6:57 2
Manufacturer:		Dell Inc.
Product Name:		Latitude 3540
Version:		
Serial Number:		C5XQ4Y3
Machine Type:		Physical
Operating System:	Operating System: Fedora Linux 39 (Thirty Nine)
Kernel:			6.6.8-100.fc38.x86_64
Architecture:		x86_64
Processor Name:		13th Gen Intel(R) Core(TM) i5-1335U

> $ lsusb
...
Bus 004 Device 002: ID 17e9:4301 DisplayLink [unknown]

> $ systemctl status displaylink-driver.service
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Fri 2023-12-29 10:27:24 EST; 6h ago
...
Dec 29 10:27:24 c00000818-unx-lptp systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
Dec 29 10:27:24 c00000818-unx-lptp systemd[1]: Started displaylink-driver.service - DisplayLink Driver Service.

Build steps:

1. git clone https://github.com/DisplayLink/evdi
2. cd evdi
3. export CPLUS_INCLUDE_PATH="/usr/include/python3.12:$CPLUS_INCLUDE_PATH"
4. sudo dnf install pybind11-devel
5. make
6. sudo make install
7. reboot

Thank you very much for contributing to this thread!

:computer: :sparkles: Happy Hacking & Happy New Year :champagne: :clinking_glasses:

fishingskywalker, 2023-12-29:
Thank you @regisf  & @mcoyle1960 for your instructions, I successfully built EVDI on my Fedora 39 laptop, and now I have my second monitor!

System:

> Hostname:		unix-laptop
uptime:			6:57 2
Manufacturer:		Dell Inc.
Product Name:		Latitude 3540
Version:		
Serial Number:		C5XQ4Y3
Machine Type:		Physical
Operating System:	Operating System: Fedora Linux 39 (Thirty Nine)
Kernel:			6.6.8-100.fc38.x86_64
Architecture:		x86_64
Processor Name:		13th Gen Intel(R) Core(TM) i5-1335U

> $ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
...
Bus 004 Device 002: ID 17e9:4301 DisplayLink [unknown]

Build Instructions:

**1. Remove previous versions:**

$ sudo dnf remove displaylink

**2. Install developer tools:**

$ sudo dnf groupinstall 'Development Tools'
$ sudo dnf install gcc-g++
$ sudo dnf install python-devel
$ sudo dnf install pybind11-devel
$ sudo dnf install libdrm-devel

**3. Clone git repository, set env variables & build:**

$ git clone https://github.com/DisplayLink/evdi
$ cd evdi
$ export CPLUS_INCLUDE_PATH="/usr/include/python3.12:$CPLUS_INCLUDE_PATH"
$ make
$ sudo make install
$ sudo reboot

4. Check service after reboot:

> $ systemctl status displaylink-driver.service
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Fri 2023-12-29 10:27:24 EST; 7h ago
    Process: 2379 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
...
Dec 29 10:27:24 c00000818-unx-lptp systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
Dec 29 10:27:24 c00000818-unx-lptp systemd[1]: Started displaylink-driver.service - DisplayLink Driver Service.

:computer: :sparkles: Happy Hacking & Happy New Year :clinking_glasses: :champagne:

reinouts, 2024-01-02:
@fishingskywalker Looks like you're missing the step where you need to reinstall the displaylink rpm? Or am I overlooking something?

fishingskywalker, 2024-01-02:
> @fishingskywalker Looks like you're missing the step where you need to reinstall the displaylink rpm? Or am I overlooking something?

> 
...

$ sudo make install
$ sudo dnf install Downloads/fedora-30-displaylink-1.14.1-1.i386.rpm

    X will probably freeze, swap to a console with CTRL+ALT+F3/4/5/6 ...

$ sudo reboot

...

🦊🍂

diepeterpan, 2024-01-06:
Building on Rocky Linux there were some slight changes to the process, but it's working now.

yum remove displaylink

sudo dnf groupinstall 'Development Tools'
sudo dnf install gcc-g++
sudo dnf install python-devel
sudo dnf install pybind11-devel
sudo dnf install libdrm-devel
git clone https://github.com/DisplayLink/evdi
cd evdi
export CPLUS_INCLUDE_PATH="/usr/include/python3.12:$CPLUS_INCLUDE_PATH"

make

**pip3 install pybind11**
sudo make install

**cd /lib/modules/$(uname -r)/build/certs
sudo tee x509.genkey > /dev/null << 'EOF'
[ req ]
default_bits = 4096
distinguished_name = req_distinguished_name
prompt = no
string_mask = utf8only
x509_extensions = myexts
[ req_distinguished_name ]
CN = Modules
[ myexts ]
basicConstraints=critical,CA:FALSE
keyUsage=digitalSignature
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid
EOF

sudo openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform DER -out signing_key.x509 -keyout signing_key.pem**
cd ~/evdi/
sudo make install

cd ..
yum install rocky-9-displaylink-1.14.1-1.x86_64.rpm 
systemctl status displaylink-driver.service
_systemctl enable displaylink-driver.service_

elguero, 2024-01-07:
A new Fedora 39 package has been released.

closed 2024-01-07 by elguero

stentammemae, 2024-01-21:
> $ make
> CFLAGS="-isystem./include -isystem./include/uapi -Werror -Wextra -Wall -Wno-error=missing-field-initializers > -Werror=sign-compare -Wmissing-prototypes -Wstrict-prototypes -Werror=discarded-qualifiers " make -C module 
> make[1]: Entering directory '/home/user/displaylink/evdi/module'
> make -C /lib/modules/6.6.9-200.fc39.x86_64/build M=$PWD
> make[2]: Entering directory '/home/user/displaylink/evdi/module'
> make[2]: *** /lib/modules/6.6.9-200.fc39.x86_64/build: No such file or directory.  Stop.
> make[2]: Leaving directory '/home/user/displaylink/evdi/module'
> make[1]: *** [Makefile:86: module] Error 2
> make[1]: Leaving directory '/home/user/displaylink/evdi/module'
> make: *** [Makefile:10: all] Error 2

Tried also with 6.6.9 and 6.5.6. Couldn't get past this.