displaylink-rpm/displaylink-rpm #258: Cursor artifacts after RPM install

biohavregryn, 2024-02-08:
* Are you using the latest DisplayLink driver? - **Yes, 5.8.0-1** 
  * Are you using the latest EVDI module? - **Yes, 1.14.1**
  * If you are using a DisplayLink device, have you checked 'troubleshooting' on DisplayLink's website? - **Yes, no matching topic found.**
  * Is this issue related to evdi/kernel? - **I don´t know how to identify this.**
  * Is this issue related to DisplayLinkManager? - **I don´t know how to identify this.**
  * If you are having an issue upgrading, have you tried to do a clean install of the driver? - **It was a clean install of 5.8.0-1**

  * Fedora
  * Version 39
  * Kernel version 6.7.3-200.fc39.x86_64

[displaylink.log](https://github.com/displaylink-rpm/displaylink-rpm/files/14213933/displaylink.log)
[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/14213934/make.log)

- I installed the DisplayLink driver to get my new external monitor to be recognized (connected via USB) -> That worked.
- However, at the same time I noticed three other changes:
1. My LUKS password query during bootup has a different look. Still works, so I guess that's fine.
2. My cursor occasionally starts blinking (quickly appearing and disappearing). I can´t identify under which circumstances.
3. After a mouseclick, a roughly 100x100 pixels area around my cursor still shows the image of whatever it is that was visible before the click. 
-  Example: A button is green and when pressed, it turns red. After installing the RPM, when I click the green button, it turns red, but the 100x100 area around my cursor still has shows the green button. The artifact shrinks when I move the cursor and disappears completely when I leave the 100x100 area with my mouse cursor.
- [image]

I am a novice linux user, but will try anything I can do fix this. I may also be completely wrong here, but I don't know where else to start, since my only change was installing the latest RPM. 

Thanks for your help!

vavadik, 2024-02-13:
Facing the same issue

vavadik, 2024-02-14:
Hey @biohavregryn, just have managed to fix this issue, installed EVDI and displaylink drivers manually following [this guide](https://kcore.org/2023/01/01/installing-evdi-fedora37/). Replaced 1.12.0 EVDI version with the latest 1.14.2.

The issue is probably related to EVDI incompatibility with the latest kernel (6.7.4-200.fc39.x86_64). It works only on X11 though, seeing black screen on Wayland.

I have Nvidia card and 545 driver version.

Hope that helps!

biohavregryn, 2024-02-18:
Yes, it did indeed work! Thanks for pointing out this path!

I somehow didn´t get RPMSphere to work, but I found a [workaround on reddit](https://redlib.r4fo.com/r/Fedora/comments/yxkm3w/fedora_37_anybody_know_how_to_get_displaylink_to/iy5mh3g/), which covers basically the same approach.

closed 2024-02-18 by biohavregryn