displaylink-rpm/displaylink-rpm #259: Displaylink-RPM fails to install on Fedora 40

rutgerputter, 2024-03-06:
**Please note:** The only working combination is *DisplayLink 5.3.1* and *evdi v1.7.2* OR *DisplayLink 5.4* and *evdi v1.9.1*

  * Are you using the latest DisplayLink driver?
    
Yes, fedora-39-displaylink-1.14.1-2.x86_64.rpm
  * Are you using the latest EVDI module?
    
Yes, from the fedora-39-displaylink-1.14.1-2.x86_64.rpm file
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
  * Is this issue related to evdi/kernel?
    
Yes, the bundled version in fedora-39-displaylink-1.14.1-2.x86_64.rpm is v1.14.1 and v1.14.2 is released with support for kernel 6.8, which is included in Fedora 40
  * Is this issue related to DisplayLinkManager?
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora / CentOS / Rocky Linux
  * Version
  * Kernel version `(uname -a)`

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

Fedora 40 has been branched off and I'm trying to find out any issues before release. Fedora 40 includes kernel 6.8 and probable a bump in EVDI version is needed.

Upgrade to Fedora 40, install fedora-39-displaylink-1.14.1-2.x86_64.rpm from the releases section.

Could you please bump the EVDI version in your releases, this would probably fix the issue.

elguero, 2024-03-09:
Hi @rutgerputter ,

Sure.  I can start looking at updating the EVDI module.  Hopefully the Displaylink driver works with it.

rutgerputter, 2024-03-11:
> Hi @rutgerputter ,
> 
> Sure. I can start looking at updating the EVDI module. Hopefully the Displaylink driver works with it.

Much appreciated! If you need any help testing, give me a shout :smile: .

mhjacks, 2024-03-26:
I can also help test if needed.

MastaG, 2024-04-04:
any update on this?

michal-os, 2024-04-05:
Hi, I'm running Fedora 38 using the linux-surface custom kernel, and I just updated to kernel 6.8 today: kernel-surface-6.8.1-1.surface.fc38.x86_64.

The latest displaylink-rpm release v5.8.0-1 (fedora-38-displaylink-1.14.1-2.x86_64.rpm) does not work with my kernel, and if I build this project locally from the master branch using the default target ( git clone displaylink-rpm.git ; make ) I get build errors with evdi.

The issue with building the default target on master is currently master expects evdi version 1.14.2, but the version of evdi in the 'DisplayLink USB Graphics Software for Ubuntu5.8-EXE.zip' file from synaptics.com contains version 1.14.1. The default target uses the evdi version supplied in the ZIP file, and the version mismatch causes the build failure.

The fundamental issue with displaylink not running on kernel 6.8 is due to some obsolete linux kernel symbols being used in evdi 1.14.1: DRM_UNLOCKED and I2C_CLASS_DDC, which has been fixed in evdi 1.14.2: https://github.com/DisplayLink/evdi/issues/448, https://github.com/DisplayLink/evdi/pull/450

To build a functional displaylink-rpm package I had to build this project locally using the rpm-github target: **make rpm-github**, which pulls evdi version 1.14.2 from github.

With the rpm-github target the build is successful, I could install the locally built rpm without errors and my displaylink dock started working normally again.

MastaG, 2024-04-05:
so the release just needs updating against the latest evdi version.
i hope the maintainer can do this soon.

rutgerputter, 2024-04-05:
Confirming that `make rpm-github` does not cause any dkms errors anymore and the the built rpm installs just fine.

I'll have access to a dock next week and I'll test if it actually works.

jee-r, 2024-04-06:
Confirming that  `make github-release` woks on my Fedora 39 `6.8.4-200.fc39.x86_64`  with Dell D3100 Docking Station

cheers

jasperges, 2024-04-08:
After I updated the kernel this morning I also encountered this issue. Doing `make github-release` on my Fedora 39 with kernel `6.8.4-200.fc39.x86_64` made it work again. Thanks!

rutgerputter, 2024-04-09:
> Confirming that `make rpm-github` does not cause any dkms errors anymore and the the built rpm installs just fine.
> 
> I'll have access to a dock next week and I'll test if it actually works.

Verified that the dock actually works after `make rpm-github` on Fedora 40 as well.

So the next steps would be to include EVDI 1.14.2 in a release and also tageting Fedora 40 in the future.

timo-a, 2024-04-10:
I'm also affected on `6.8.4-200.fc39.x86_64` and hope to see a fix released soon. I tried to copy what others in this thread have done, but I cannot install the src.rpm that is built: 
```
$ git clone https://github.com/displaylink-rpm/displaylink-rpm.git
$ cd displaylink-rpm/
$ sudo dnf install fedora-packager rpmdevtools gcc
$ sudo dnf install libdrm-devel
$ make github-release
[...]
RPM build warnings:
    absolute symlink: /usr/libexec/displaylink/libevdi.so -> /usr/libexec/displaylink/libevdi.so.1.14.3
rpmbuild -bs --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 1" --define "_daemon_version 5.8.0-63.33" --define "_version 1.14.3" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`" --define "_github 1" displaylink.spec
setting SOURCE_DATE_EPOCH=1712707200
Wrote: /home/[...]/displaylink-rpm/displaylink-1.14.3-1.github_evdi.src.rpm
$ sudo dnf install displaylink-1.14.3-1.github_evdi.src.rpm
Last metadata expiration check: 1:03:03 ago on Mi 10 Apr 2024 19:39:21.
Error: Will not install a source rpm package (displaylink-1.14.3-1.github_evdi.src).
```
Can you please help me out here, did I build it wrong, did I try to install it the wrong way?
Possibly relevant: evdi [v1.14.3](https://github.com/DisplayLink/evdi/releases/tag/v1.14.3) has been released a few hours ago.

adil192, 2024-04-10:
@timo-a Run `make rpm-github` then install the rpm at `x86_64/displaylink-1.14.2-1.github_evdi.x86_64.rpm`

elguero, 2024-04-11:
Hey 👋 .  Does the latest release resolve this issue?  Thanks!

nathan-weinberg, 2024-04-12:
@elguero I'm on Fedora 38 but can confirm the latest release did fix compatibility for me using kernel `6.8.4-100.fc38.x86_64` :pray: thank you for your continued maintenance of this package!

rutgerputter, 2024-04-12:
> Hey 👋 . Does the latest release resolve this issue? Thanks!

Yes! It works great on Fedora 40, thank you for fixing this issue :smile:

closed 2024-04-12 by rutgerputter

gavinbaumanis, 2024-06-27:
YesterdaY I did an update using yum update.
So I am "up to date" from a "system" perspective.

I have very little installed
Flatpack
MS Teams for Linux
Slack
VS-code for linux.
Git.

Today;
I downloaded the Fedora 40 RPM and did a local install so that I can use a USB Dock to connect my external monitors.

Subsequently;
My Dell Precision 3581 - is now stck at the bios screen.

I cannot enter the bios.
The power button does NOT work either.

So it would seem that I am stuck at waiting for the battery to run out...

Once I manage to get back up and running I will update this message with kernel versions etc.

federstedt, 2024-09-04:
> I'm also affected on `6.8.4-200.fc39.x86_64` and hope to see a fix released soon. I tried to copy what others in this thread have done, but I cannot install the src.rpm that is built:
> 
> ```
> $ git clone https://github.com/displaylink-rpm/displaylink-rpm.git
> $ cd displaylink-rpm/
> $ sudo dnf install fedora-packager rpmdevtools gcc
> $ sudo dnf install libdrm-devel
> $ make github-release
> [...]
> RPM build warnings:
>     absolute symlink: /usr/libexec/displaylink/libevdi.so -> /usr/libexec/displaylink/libevdi.so.1.14.3
> rpmbuild -bs --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 1" --define "_daemon_version 5.8.0-63.33" --define "_version 1.14.3" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`" --define "_github 1" displaylink.spec
> setting SOURCE_DATE_EPOCH=1712707200
> Wrote: /home/[...]/displaylink-rpm/displaylink-1.14.3-1.github_evdi.src.rpm
> $ sudo dnf install displaylink-1.14.3-1.github_evdi.src.rpm
> Last metadata expiration check: 1:03:03 ago on Mi 10 Apr 2024 19:39:21.
> Error: Will not install a source rpm package (displaylink-1.14.3-1.github_evdi.src).
> ```
> 
> Can you please help me out here, did I build it wrong, did I try to install it the wrong way? Possibly relevant: evdi [v1.14.3](https://github.com/DisplayLink/evdi/releases/tag/v1.14.3) has been released a few hours ago.
> 
> Update:
> 
> ```
> make rpm-github
> sudo dnf install x86_64/displaylink-1.14.3-1.github_evdi.x86_64.rpm
> ```
> 
> did the trick, thanks @adil192!

This solved my issue aswell. Thanks!