displaylink-rpm/displaylink-rpm #260: HP Universal Dock G2 not recognized

Verhoeckx, 2024-03-12:
At my office we have the [HP Universal Dock G2](https://support.hp.com/us-en/document/c06386384) and I'm trying to make it work with Fedore 39 Silverblue. After installing the package `displaylink-1.14.1-2.x86_64.rpm` (with rpm-ostree) I hoped that it would recognize the G2 docking station, but sadly it doesn't.

Questions 

1. Is this the correct package that I need to enable support of Displaylink on the HP Universal Dock G2? 
2. Is the `HP Universal Dock G2` supported or not?

Package: displaylink-1.14.1-2.x86_64.rpm (installed with rpm-ostree)
Distribution: Fedora 39 Silverblue
Kernel: 6.7

The log file `/var/log/displaylink/displaylink.log` doesn't exist.

elguero, 2024-04-11:
It looks like the DisplayLink chipset should be supported.

It seems that Silverblue does not support DKMS which is how the evdi kernel module is being installed. With this package.

https://docs.fedoraproject.org/en-US/fedora-silverblue/troubleshooting/#_out_of_tree_kernel_modules_and_drivers_using_dkms

There is a project that was started to build akmod version of this but it stalled out.  I will have to see if I can find some time to help push that forward. https://github.com/displaylink-rpm/displaylink-rpm/pull/166

Verhoeckx, 2024-04-12:
Thanks for reaching out!

I resolved the issue temporarily by using two USB-C to DisplayLink adapters. Not perfect, but it works! I shall close this issue now.

closed 2024-04-12 by Verhoeckx