displaylink-rpm/displaylink-rpm #262: Error building ci for fedora

lennertcc, 2024-04-16:
* Are you using the latest DisplayLink driver?
    
  * Are you using the latest EVDI module?
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
  * Is this issue related to evdi/kernel?
    
  * Is this issue related to DisplayLinkManager?
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora 
  * Version 39
  * Kernel version `(uname -a)`
    Linux framework 6.8.4-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Apr  4 20:45:21 UTC 2024 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

After uninstalling and clearing up old build artifacts, I get an error running `sudo ./ci/fedora.sh`

```
[lennert@framework displaylink-rpm]$ sudo ./ci/fedora.sh 
Last metadata expiration check: 0:14:16 ago on Tue 16 Apr 2024 11:50:41 AM CEST.
Package rpm-build-4.19.1.1-1.fc39.x86_64 is already installed.
Package make-1:4.4.1-2.fc39.x86_64 is already installed.
Package gcc-13.2.1-7.fc39.x86_64 is already installed.
Package gcc-c++-13.2.1-7.fc39.x86_64 is already installed.
Package libdrm-devel-2.4.120-1.fc39.x86_64 is already installed.
Package systemd-rpm-macros-254.10-1.fc39.noarch is already installed.
Package glibc-devel-2.38-17.fc39.x86_64 is already installed.
Package wget-1.21.4-1.fc39.x86_64 is already installed.
Package git-2.44.0-1.fc39.x86_64 is already installed.
Package dnf-plugins-core-4.6.0-1.fc39.noarch is already installed.
Dependencies resolved.
Nothing to do.
Complete!
Last metadata expiration check: 0:14:17 ago on Tue 16 Apr 2024 11:50:41 AM CEST.
Package gcc-c++-13.2.1-7.fc39.x86_64 is already installed.
Package libdrm-devel-2.4.120-1.fc39.x86_64 is already installed.
Package make-1:4.4.1-2.fc39.x86_64 is already installed.
Package systemd-rpm-macros-254.10-1.fc39.noarch is already installed.
Dependencies resolved.
Nothing to do.
Complete!
rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 1" --define "_daemon_version 5.8.0-63.33" --define "_version 1.14.4" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`"  displaylink.spec --target=x86_64
Building target platforms: x86_64
Building for target x86_64
warning: %patchN is deprecated (2 usages found), use %patch N (or %patch -P N)
setting SOURCE_DATE_EPOCH=1712793600
Executing(%prep): /bin/sh -e /var/tmp/displayportVss9SZ/rpm-tmp.It3F7Y
+ umask 022
+ cd /var/tmp/displayportvo70xa
+ cd /var/tmp/displayportvo70xa
+ rm -rf displaylink-1.14.4
+ /usr/bin/mkdir -p displaylink-1.14.4
+ cd displaylink-1.14.4
+ rm -rf /var/tmp/displayportvo70xa/displaylink-1.14.4-SPECPARTS
+ /usr/bin/mkdir -p /var/tmp/displayportvo70xa/displaylink-1.14.4-SPECPARTS
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ cd /var/tmp/displayportvo70xa
+ cd displaylink-1.14.4
+ rm -rf /var/tmp/displayportvo70xa/displaylink-1.14.4-SPECPARTS
+ /usr/bin/mkdir -p /var/tmp/displayportvo70xa/displaylink-1.14.4-SPECPARTS
+ /usr/lib/rpm/rpmuncompress -x /home/lennert/code/displaylink-rpm/DisplayLink_USB_Graphics_Software_for_Ubuntu_5.8.0-63.33.zip
+ STATUS=0
+ '[' 0 -ne 0 ']'
[... 213 lines cut ...]
+ export CXX
+ cd displaylink-1.14.4
+ mkdir -p /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/libexec/displaylink/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/src/evdi-1.14.4/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/lib/systemd/system/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/lib/systemd/system-preset/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/lib/systemd/system-sleep/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/etc/logrotate.d/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/etc/modprobe.d/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/etc/udev/rules.d/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/etc/X11/xorg.conf.d/ /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/var/log/displaylink/
+ pushd /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/src/evdi-1.14.4
/var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/src/evdi-1.14.4 /var/tmp/displayportvo70xa/displaylink-1.14.4
+ cp -a /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/Kconfig /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/LICENSE /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/Makefile /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/dkms.conf /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_connector.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_cursor.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_cursor.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_debug.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_debug.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_drm.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_drm_drv.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_drm_drv.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_encoder.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_fb.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_gem.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_i2c.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_i2c.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_ioc32.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_modeset.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_painter.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_params.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_params.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_platform_dev.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_platform_dev.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_platform_drv.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_platform_drv.h /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_sysfs.c /var/tmp/displayportvo70xa/displaylink-1.14.4/evdi-1.14.4/module/evdi_sysfs.h .
+ popd
/var/tmp/displayportvo70xa/displaylink-1.14.4
+ echo NO_WEAK_MODULES=yes
+ cp -a evdi-1.14.4/library/libevdi.so.1.14.4 /var/tmp/displayportAEBMaI/displaylink-1.14.4-1.x86_64/usr/libexec/displaylink/
cp: cannot stat 'evdi-1.14.4/library/libevdi.so.1.14.4': No such file or directory
error: Bad exit status from /var/tmp/displayportVss9SZ/rpm-tmp.WRVD7n (%install)

RPM build warnings:
    %patchN is deprecated (2 usages found), use %patch N (or %patch -P N)

RPM build errors:
    Bad exit status from /var/tmp/displayportVss9SZ/rpm-tmp.WRVD7n (%install)
make: *** [Makefile:151: x86_64/displaylink-1.14.4-1.x86_64.rpm] Error 1
```

- uninstall displaylink from fedora
- clean up previous build artifacts
- pull latest from this repo
- `sudo ./ci/fedora.sh`

lennertcc, 2024-04-16:
/var/lib/dkms/evdi/kernel-[version]/log/make.log

DKMS make.log for evdi-1.14.1 for kernel 6.8.5-201.fc39.x86_64 (x86_64)
Tue Apr 16 11:50:15 AM CEST 2024
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.14.1/build SUBDIRS=/var/lib/dkms/evdi/1.14.1/build SRCROOT=/var/lib/dkms/evdi/1.14.1/build CONFIG_MODULE_SIG= -C /lib/modules/6.8.5-201.fc39.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/6.8.5-201.fc39.x86_64'
make --no-print-directory -C /usr/src/kernels/6.8.5-201.fc39.x86_64 \
-f /usr/src/kernels/6.8.5-201.fc39.x86_64/Makefile modules
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/1.14.1/build need-builtin=1 need-modorder=1 
# cmd_mod /var/lib/dkms/evdi/1.14.1/build/evdi.mod
  printf '%s\n'   evdi_platform_drv.o evdi_platform_dev.o evdi_sysfs.o evdi_modeset.o evdi_connector.o evdi_encoder.o evdi_drm_drv.o evdi_fb.o evdi_gem.o evdi_painter.o evdi_params.o evdi_cursor.o evdi_debug.o evdi_i2c.o evdi_ioc32.o | awk '!x[$0]++ { print("/var/lib/dkms/evdi/1.14.1/build/"$0) }' > /var/lib/dkms/evdi/1.14.1/build/evdi.mod
# CC [M]  /var/lib/dkms/evdi/1.14.1/build/evdi_platform_drv.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.14.1/build/.evdi_platform_drv.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-overflow -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -g -Iinclude/uapi/drm -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_platform_drv"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.14.1/build/evdi_platform_drv.o /var/lib/dkms/evdi/1.14.1/build/evdi_platform_drv.c  
# CC [M]  /var/lib/dkms/evdi/1.14.1/build/evdi_platform_dev.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.14.1/build/.evdi_platform_dev.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-overflow -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-varia [... cut ...]
./include/drm/drm_ioctl.h:155:26: note: in definition of macro ‘DRM_IOCTL_DEF_DRV’
  155 |                 .flags = _flags,                                        \
      |                          ^~~~~~
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/evdi/1.14.1/build/evdi_drm_drv.o] Error 1
make[2]: *** [/usr/src/kernels/6.8.5-201.fc39.x86_64/Makefile:1939: /var/lib/dkms/evdi/1.14.1/build] Error 2
make[1]: *** [Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.8.5-201.fc39.x86_64'
make: *** [Makefile:38: all] Error 2

lennertcc, 2024-04-17:
I dug in a bit deeper and found that the file at the location is not `libevdi.so.1.14.4`  but `libevdi.so.1.14.1`
` ./var/tmp/displayportfBONp1/displaylink-1.14.4/evdi-1.14.4/library/libevdi.so.1.14.1`

I removed all traces of previous evdi installations and I cannot find anything named `*1.14.1*` on my machine, still this file appears during install.

elguero, 2024-04-17:
Hi @lennertcc ,

The intention is not really to run those scripts directly.

The issue is that DisplayLink has not updated their official driver package release to use the evdi 1.14.4 release.

So, the version included in the official release will be evdi 1.14.1 by running the script the way you are.

The ci scripts are intended to be used as part of our  CI when we do code commits and releases.

If you are building locally, just use `make github-release` to build your RPM to use the evdi release found on GitHub.  You can use the ci script as a guide for understanding the build dependencies.

Hope that helps.

closed 2024-05-10 by elguero