displaylink-rpm/displaylink-rpm #263: Step by step install instructions

m0byn, 2024-04-25:
* Are you using the latest DisplayLink driver?
Do I have to use it instead of rpm package?
  * Are you using the latest EVDI module?
How do I know?
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
Using [Targus Dock 120](https://us.targus.com/blogs/discover-targus/new-dock120-targus-universal-docking-station)
  * Is this issue related to evdi/kernel?
I do not know!
  * Is this issue related to DisplayLinkManager?
I do not know
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    _Do the following for a clean driver install:
           Find the current package installed - `sudo rpm -qa | grep displaylink`_
I get the following result: 

`[manuel@linny ~]$ sudo rpm -qa | grep displaylink
Fehler: Verifying a signature using certificate A88441BD4864F95BEE08E63A71EB474019940E11 (Proton Technologies AG <opensource@proton.me>):
  1. Certificiate 71EB474019940E11 invalid: certificate is not alive
      because: The primary key is not live
      because: Expired on 2022-10-28T13:52:11Z
  2. Key 71EB474019940E11 invalid: key is not alive
      because: The primary key is not live
      because: Expired on 2022-10-28T13:52:11Z
displaylink-1.14.4-1.github_evdi.x86_64
`

_Remove what is returned: `sudo rpm -evh [package]`_
Only remove the displaylink package or the Certificates/Keys as well??

           Clear out anything left over:
            /var/lib/dkms/evdi
            /usr/libexec/displaylink
            /usr/src/evdi-*

            /etc/X11/xorg.conf.d/20-displaylink.conf
            /etc/logrotate.d/displaylink
            /usr/lib/systemd/system/displaylink-driver.service
            /usr/lib/systemd/system-preset/95-displaylink.preset
            /usr/lib/systemd/system-sleep/displaylink.sh
            /etc/udev/rules.d/99-displaylink.rules
            /usr/share/doc/displaylink
            /usr/share/doc/displaylink/LICENSE

            /usr/lib/modules/`uname -r`/weak-updates/evdi.ko.xz
            /usr/lib/modules/`uname -r`/extra/evdi.ko.xz
          Reboot
          Reinstall the RPM package -->

  * Fedora 
  * 40
  * Kernel version 6.8.7-300.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Apr 17 19:21:08 UTC 2024 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`

Result is: `command not found`

  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

Result is: `command not found`

**PROBLEM:**

Rather new Linux user here, so pleeeease help me out! I am trying to get my Targus Dock 120, which uses DisplayLink, up and running. Unfortunately, I did not succeed. I followed several guides found online but none of them lead me to get it up and running. Just to give you Ideas: first I tried installing the Ubuntu drivers - did not work. Then I installed the rpm package and signed the module in several different ways - does not work. 

I also asked on superuser.com [here](https://superuser.com/q/1839875/1653863) but unfortunately, I did not get satisfying asnwers.

So, what I am asking is: could you tell me 

1. step by step guide how to remove anything from previous installations? Please see the ouput provided above as I am not sure what to remove?
2. Please provide a step by step guide on what and how to install the drivers! I am not sure whether just installing the RPM package and signing the module is sufficient, or not. Especially the topic about the evdi module is confusing. I would vote for providing a step by step installation guide in the overview!

LtDragon, 2024-04-28:
I'm also new to Linux, i'm running fedora 40 and Ubuntu 24.04. I like fedora clean interface but I miss the easier install of the displaylink drivers. I tried to install the driver on fedora but I can't seem to get it to work. I think i'm not understanding the steps correctly. Can some write a simpler install procedure and label what variable need to change for each install. Would it also be possible to set up repository in the software install app?

mkadrlik, 2024-04-28:
Looking around at some previous issues, I found [these installation instructions](https://github.com/displaylink-rpm/displaylink-rpm/issues/22#issuecomment-284261155); following those steps got me up and running.  Note: When you get down to step _"All good? ok, install the rpm:"_, remember to sudo the command, otherwise it won't run. 

Happy installing!

krishna05008, 2024-05-05:
i am using fedora 40 as well and new to linux i have done everything but i am getting the following error after the evdi thing `sudo modinfo -n evdi /lib/modules/5.10.19-200.fc33.x86_64/extra/evdi.ko.xz

sudo unxz $(modinfo -n evdi)

sudo /usr/src/kernels/$(uname -r)/scripts/sign-file sha256 ./MOK.priv \
  ./MOK.der /lib/modules/$(uname -r)/extra/evdi.ko

sudo xz -f /lib/modules/$(uname -r)/extra/evdi.ko
modinfo: ERROR: Module evdi not found.
modinfo: ERROR: Module /lib/modules/5.10.19-200.fc33.x86_64/extra/evdi.ko.xz not found.
modinfo: ERROR: Module evdi not found.
unxz: Compressed data cannot be read from a terminal
sudo: /usr/src/kernels/6.8.8-300.fc40.x86_64/scripts/sign-file: command not found
xz: /lib/modules/6.8.8-300.fc40.x86_64/extra/evdi.ko: No such file or directory`

m0byn, 2024-05-09:
Thank you for your reply @GrzegorzBB I followed your instructions, but unfortunately, I am idling at step 3. Whenever I execute `mokutil --import /var/lib/dkms/mok.pub` I get the following error: `Failed to get file status, /var/lib/dkms/mok.pub`

I verified, whether the file exists or not and apparently, the file does NOT exist in the given directory. 

This might be related to some steps I did some time ago. In more detail, when someone opens a new issue here, there is a detailed guide what to do in order to delete all traces from previous installs. There, it is written to delete the directory `/var/lib/dkms/evdi`. Unfortunately, I deleted everything in the `/dkms` folder and not just the evdi folder. As far as I remember, the required file `mok.pub` was in there as well. 

In order to solve the error I am getting, I tried to reinstall mokutil and dkms hoping for the file to return. This did not work, however. So, could you give me a hint on how to proceed?

m0byn, 2024-05-12:
Thanks for getting back to me @GrzegorzBB 

Here, you are going to find the output of the specified commands.

1. `sudo dnf info dkms`
```
Installierte Pakete
Name         : dkms
Version      : 3.0.13
Release      : 1.fc40
Architektur  : noarch
Größe        : 183 k
Quelle       : dkms-3.0.13-1.fc40.src.rpm
Paketquelle  : @System
Aus Paketque : fedora
Zusammenfass : Dynamic Kernel Module Support Framework
URL          : http://linux.dell.com/dkms
Lizenz       : GPLv2+
Beschreibung : This package contains the framework for the Dynamic Kernel Module
             : Support (DKMS) method for installing module RPMS as originally
             : developed by Dell.
```

2.`ls -al /var/lib/dkms`
```
insgesamt 0
drwxr-xr-x. 1 root root    0  6. Mär 01:00 .
drwxr-xr-x. 1 root root 1024 24. Apr 15:08 ..
```
where `insgesamt` means `total`.

3. `cat /etc/dkms/framework.conf`
```
# This configuration file modifies the behavior of DKMS (Dynamic Kernel Module
# Support) and is sourced in by DKMS every time it is run.

# Source Tree Location (default: /usr/src):
# source_tree="/usr/src"

# DKMS Tree Location (default: /var/lib/dkms):
# dkms_tree="/var/lib/dkms"

# Install Tree Location (default: /lib/modules):
# install_tree="/lib/modules"

# Temporary folder Location (default: /tmp):
# tmp_location="/tmp"

# Verbosity setting, will be active if set to a non-null value:
# verbose=""

# This creates symlinks from the install_tree into the dkms_tree instead of
# copying the modules. This preserves some space on the costs of being less
# safe. Symlinking will be active if set to a non-null value:
# symlink_modules=""

# Automatic installation and upgrade for all installed kernels if set to a
# non-null value:
# autoinstall_all_kernels=""

# Location of the sign-file kernel binary. $kernelver can be used in path to
# represent the target kernel version. (default: depends on distribution):
# sign_file="/path/to/sign-file"

# Location of the key and certificate files used for Secure boot. $kernelver
# can be used in path to represent the target kernel version.
#
# NOTE: If any of the files specified by `mok_signing_key` and
# `mok_certificate` are non-existant, dkms will re-create both files.
#
# mok_signing_key can also be a "pkcs11:..." string for PKCS#11 engine, as
# long as the sign_file program supports it.
# (default: /var/lib/dkms):
# mok_signing_key=/var/lib/dkms/mok.key
# mok_certificate=/var/lib/dkms/mok.pub

# Automatically modprobe the built modules upon successful installation:
modprobe_on_install="true"
```

4. The command I used to reinstall the dkms package is: `sudo dnf reinstall dkms mokutil` 

Hopefully, the information helps. If anything is unclear because of the language, I am happy to help!

GrzegorzBB, 2024-05-12:
Hi @m0byn, everything looks good, I think that we need to change the order.
Before you start please remove displaylink driver rpm that you installed previously. Its name can be collected using `rpm -qa |grep displaylink`. After package removal please follow below:
 
1. Get latest rpm package for your distribution from [release section](https://github.com/displaylink-rpm/displaylink-rpm/releases).
2. Go to the directory where you downloaded the package and install it using `sudo dnf install packagename.rpm`. Ignore potential errors.
3. Install mokutil tool from repository `sudo dnf install mokutil`.
4. Verify that your system using SecureBoot using command `mokutil --sb-state`.
5. If the SecureBoot is enabled execute `mokutil --import /var/lib/dkms/mok.pub` and follow the instruction available [here](https://github.com/dell/dkms?tab=readme-ov-file#secure-boot) (you will need to restart your computer ), otherwise skip this point.
6. Execute `dkms autoinstall`.
7. Evdi module should be correctly signed. You can verify that by executing `sudo dkms status` or `sudo systemctl status displaylink-driver.service`.

LtDragon, 2024-05-16:
@GrzegorzBB It look like your instructions worked for be but now it showing from step 7 
sudo systemctl status displaylink-driver.service
○ displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: inactive (dead)

Also when I try dkms autoinstall (it says the below)
Error! You must be root to use this command.

Running this commond. (ls -al /var/lib/dkms)
I get the following
total 8
drwxr-xr-x. 1 root root   36 May 15 22:43 .
drwxr-xr-x. 1 root root 1006 Apr 25 01:28 ..
drwxr-xr-x. 1 root root   82 May 15 22:43 evdi
-rw-------. 1 root root 1704 Apr 25 01:28 mok.key
-rw-r--r--. 1 root root  811 Apr 25 01:28 mok.pub

m0byn, 2024-05-16:
Thank you for your message.

>     1. Get latest rpm package for your distribution from [release section](https://github.com/displaylink-rpm/displaylink-rpm/releases).
> 
>     2. Go to the directory where you downloaded the package and install it using `sudo dnf install packagename.rpm`. Ignore potential errors.

I installed the package and received the following output: 

```
Installieren:
 displaylink      x86_64      1.14.4-1.github_evdi      @commandline      6.3 M

Transaktionszusammenfassung
================================================================================
Installieren  1 Paket

Gesamtgröße: 6.3 M
Installationsgröße: 15 M
Ist dies in Ordnung? [j/N]: j
Pakete werden heruntergeladen:
Transaktionsüberprüfung wird ausgeführt
Transaktionsüberprüfung war erfolgreich.
Transaktion wird getestet
Transaktionstest war erfolgreich.
Transaktion wird ausgeführt
  Vorbereitung läuft    :                                                   1/1 
  Installieren          : displaylink-1.14.4-1.github_evdi.x86_64           1/1 
  Ausgeführtes Scriptlet: displaylink-1.14.4-1.github_evdi.x86_64           1/1 
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xeu displaylink-driver.service" for details.
Warnung: %post(displaylink-1.14.4-1.github_evdi.x86_64) Scriptlet fehlgeschlagen, Beenden-Status 1

Error in POSTIN scriptlet in rpm package displaylink

Installiert:
  displaylink-1.14.4-1.github_evdi.x86_64                                       

Fertig!
```
Is that correct?

> 3. Install mokutil tool from repository `sudo dnf install mokutil`.
>
>     4. Verify that your system using SecureBoot using command `mokutil --sb-state`.
>
>     5. If the SecureBoot is enabled execute [...]

SecureBoot is enabled. However, whenever I execute the mentioned command in step 5 I receive:

```
$ mokutil --import /var/lib/dkms/mok.pub
#Failed to access kernel trusted keyring: Required key not available
input password: 
input password again: 
Failed to enroll new keys
```

and restarting the machine does not lead me to the blue screen in order to enroll the keys. Do you have any hints??

I stopped here and did not continue. Should I continue anyways?

> 
>     6. Execute `dkms autoinstall`.
> 
>     7. Evdi module should be correctly signed. You can verify that by executing `sudo dkms status` or `sudo systemctl status displaylink-driver.service`.

GrzegorzBB, 2024-05-21:
@LtDragon first of all please try to execute `mokutil --list-enrolled` and verify if DKMS key is enrolled, after that try to execute `sudo dkms autoinstall`. 

@m0byn It looks like that key wasn't automatically generated for you -  you can always generate it manually by executing below commands:
`openssl req -new -x509 -newkey rsa:2048 -keyout /var/lib/dkms/mok.key -outform DER -out \
/var/lib/dkms/mok.pub -nodes -days 36500 -subj "/CN=DKMS/"`
After that please follow the rest of the guide.

Googolbyte, 2024-05-21:
> i am using fedora 40 as well and new to linux i have done everything but i am getting the following error after the evdi thing
> 
> ```
> 
> sudo unxz $(modinfo -n evdi)
> 
> sudo /usr/src/kernels/$(uname -r)/scripts/sign-file sha256 ./MOK.priv \
>   ./MOK.der /lib/modules/$(uname -r)/extra/evdi.ko
> 
> sudo xz -f /lib/modules/$(uname -r)/extra/evdi.ko
> modinfo: ERROR: Module evdi not found.
> modinfo: ERROR: Module /lib/modules/5.10.19-200.fc33.x86_64/extra/evdi.ko.xz not found.
> modinfo: ERROR: Module evdi not found.
> unxz: Compressed data cannot be read from a terminal
> sudo: /usr/src/kernels/6.8.8-300.fc40.x86_64/scripts/sign-file: command not found
> xz: /lib/modules/6.8.8-300.fc40.x86_64/extra/evdi.ko: No such file or directory```
> ```

The problem with the instructions is that it is making a hard reference to a specific linux kernel version. 5.10.* was released in 2020, and Fedora 40 ships with version 6.8.*. The releases page (can be found by clicking for "tags" on the front page) contains recent builds. However, per the guide, it seems that these builds need to be made regularly whenever a new kernel is released.

If this is still the case, the instructions and the install scripts to make your own release need to be revised to use some sort of ENV variable that points to the user's machine's linux kernal to ensure everything is built correctly.

LtDragon, 2024-05-22:
@GrzegorzBB 
I tried that but i'm not able to get it to work
below is what i see
kuru@fedora:~$ mokutil --list-enrolled
2bb010e24d fedoraca
d69fe7c2bd kurus-Inspiron-5584 Secure Boot Module Signature key
1292afce26 Displaylink
f4efccc7c4 Displaylink
5b5590b74d Displaylink
kuru@fedora:~$ sudo dkms autoinstall
[sudo] password for kuru: 
kuru@fedora:~$ sudo dkms status
evdi/1.14.4, 6.8.9-300.fc40.x86_64, x86_64: installed
kuru@fedora:~$ sudo systemctl status displaylink-driver.service
○ displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: inactive (dead)

GrzegorzBB, 2024-05-22:
@LtDragon Hi, you have enrolled 3 different keys for signing displaylink module:

> 1292afce26 Displaylink
f4efccc7c4 Displaylink
5b5590b74d Displaylink

Right now we need to verify if the key that we use in DKMS in on this list (system will not load the module signed by unenrolled key). To do that please execute below command and compare 10 first characters with above list.
`openssl dgst -sha1 /var/lib/dkms/mok.pub`

Please post the output in comment.

LtDragon, 2024-05-22:
> @LtDragon Hi, you have enrolled 3 different keys for signing displaylink module:
>
>
> Right now we need to verify if the key that we use in DKMS in on this list (system will not load the module signed [...]

this is the output
Doesn't look like the same key

SHA1(/var/lib/dkms/mok.pub)= 2b27a59d1dc4cc2a98d79a24d49c4158dda092c5

****

GrzegorzBB, 2024-05-28:
@LtDragon Have you tried to enroll the pub key using
'sudo mokutil --import /var/lib/dkms/mok.pub'?

LtDragon, 2024-05-29:
@GrzegorzBB I ran that code and it seems to be working now.
The issue i'm having is when i discount my dock. The screen configuration isn't keep and have to be reset each time.

ssaavedra, 2024-05-29:
@m0byn you have not reported back, but the instructions above should fix your issue. Please let us know and we'll close the issue as fixed :)

m0byn, 2024-05-29:
> @m0byn you have not reported back, but the instructions above should fix your issue. Please let us know and we'll close the issue as fixed :)

I am currently working abroad and only have my portable devices with me. Therefore, I cannot test it. In two weeks I am going to be back home and am going to try it immediately.

ahkdigital, 2024-06-20:
> Hi, readme on the front page is a little bit outdated in my opinion, from dkms version 3.0.4 there is no need to generate your own certificate to sign kernel module - dkms will generate self signed certificate for you and you only need to enroll it to your machine. Basic installation instruction would look like that:
> 
> 1. Install mokutil and dkms tools from repository `sudo dnf install mokutil dkms`.
> 2. Verify if your system using SecureBoot using command `mokutil --sb-state`.
> 3. If the SecureBoot is enabled execute `mokutil --import /var/lib/dkms/mok.pub` and follow the instruction available [here](https://github.com/dell/dkms?tab=readme-ov-file#secure-boot) (you will need to restart your computer ), otherwise skip this point.
> 4. Get latest rpm package for your distribution from [release section](https://github.com/displaylink-rpm/displaylink-rpm/releases).
> 5. Go to the directory where you downloaded the package and install it using `sudo dnf install packagename.rpm`
> 6. Package should be installed without any issues. You can verify that by executing `sudo systemctl status displaylink-driver.service`

Thank you for these clear and easy to follow instructions. :D

m0byn, 2024-06-27:
Sorry for the very late response. As before, I was idling at [this](https://github.com/displaylink-rpm/displaylink-rpm/issues/263#issuecomment-2114686697) step. Then, as you mentioned, I tried to generate the keys manually by executing `openssl req -new -x509 -newkey rsa:2048 -keyout /var/lib/dkms/mok.key -outform DER -out \ /var/lib/dkms/mok.pub -nodes -days 36500 -subj "/CN=DKMS/"` which resulted in 

> sudo openssl req -new -x509 -newkey rsa:2048 -keyout /var/lib/dkms/mok.key -outform DER -out \ /var/lib/dkms/mok.pub -nodes -days 36500 -subj "/CN=DKMS/"
..........+...+.............+++++++++++++++++++++++++++++++++++++++*...+..+...+.......+..+....+.........+...+..+......+....+..+.+..+.+......+.....+......+++++++++++++++++++++++++++++++++++++++*.................+...++++++
...........+..+............+...+++++++++++++++++++++++++++++++++++++++*........+....+++++++++++++++++++++++++++++++++++++++*......+........+......+......+....+..+.......+..................+...+.........+............+...+..............................+...+...+.....+.......+...+..+...+.+.....+...+.......+...+....................+.+......+........+.+..+....+......+..++++++
-----
> Can't open " /var/lib/dkms/mok.pub" for writing, No such file or directory
80921FDEC67F0000:error:80000002:system library:BIO_new_file:No such file or directory:crypto/bio/bss_file.c:67:calling fopen( /var/lib/dkms/mok.pub, wb)
80921FDEC67F0000:error:10000080:BIO routines:BIO_new_file:no such file:crypto/bio/bss_file.c:75:

and I do not know how to proceed. Do you have any hints?

GrzegorzBB, 2024-07-01:
Does the file /var/lib/dkms/mok.pub exist?
Could you publish output of  `stat` and `lsof` command for this file?

bitchkat, 2024-11-06:
> 6. Package should be installed without any issues. You can verify that by executing `sudo systemctl status displaylink-driver.service`

Is the displaylink-driver.service supposed to be enabled? Mine isn't, evdi module is not loaded, but yet when  I plug in my dell USB-docking station to my Carbon X1, it still recognizes the monitors connected to the dock.

elguero, 2026-11-11:
Closing this out since it has been a couple of years. Please open a new issue referencing this one if we still need to address this.

closed 2026-11-11 by elguero