displaylink-rpm/displaylink-rpm #269: External monitor resolution on RL 9

hg2581, 2024-07-23:
I just downloaded the latest release of  your DL driver to use with my Dell Inspiron laptop with a retina display 3840 x 2160 and an external monitor 1920 x 1080 connected to a Dell dock. I can view the apps on the  external monitor but I have not been able to set the desired resolution of it using xrandr. This is the xrandr commands I used:
xrandr --listmonitors
Monitors: 2
 0: +*eDP-1 3840/346x2160/194+0+0  eDP-1
 1: +DVI-I-2-1 3840/509x2160/286+3840+0  DVI-I-2-1

xrandr --output DVI-I-2-1 --panning 1920x1080+3840+0 --scale 1x1 --right-of eDP-1
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  29 (RRSetPanning)
  Serial number of failed request:  59
  Current serial number in output stream:  59

xrandr --listmonitors
Monitors: 2
 0: +*eDP-1 3840/346x2160/194+0+0  eDP-1
 1: +DVI-I-2-1 1920/509x1080/286+3840+0  DVI-I-2-1

Although the last xrandr command seems to have reset the resolution, the output on the display is identical to before running the xrandr command. The xrandr command also resulted in an error message.

Prior to connecting this external monitor to the dock, I had it connected directly to the laptop and used a scaling factor of 2x2 in xrandr which worked fine and gave me desired resolution etc. However, when I run that same command with the monitor connected to the dock, I get the following:

xrandr --output DVI-I-2-1 --panning 1920x1080+3840+0 --scale 2x2 --right-of eDP-1
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  29 (RRSetPanning)
  Serial number of failed request:  58
  Current serial number in output stream:  58

$ xrandr --listmonitors
Monitors: 2
 0: +*eDP-1 3840/346x2160/194+0+0  eDP-1
 1: +DVI-I-2-1 3840/509x2160/286+3840+0  DVI-I-2-1

As you can see, the monitor now changed to 3840 x 1920 despite only having a resolution of 1920 x 1080.

Might there be a bug in the displaylink driver? I would be happy to test a new release for you.

  * Are you using the latest DisplayLink driver?
    
  * Are you using the latest EVDI module?
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
  * Is this issue related to evdi/kernel?
    
  * Is this issue related to DisplayLinkManager?
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora / CentOS Stream / Rocky Linux / AlmaLinux OS
  * Version
  * Kernel version `(uname -a)`

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`

elguero, 2024-11-09:
Hola!

Are you on Wayland?

Please see this article: https://support.displaylink.com/knowledgebase/articles/675136-how-to-configure-displaylink-displays-on-ubuntu

hg2581, 2024-12-29:
Sorry for the very late reply. No, I am using X11.

hg2581, 2025-02-23:
How do I remove the "Need Feedback" label since I have provided the requested information?

elguero, 2026-11-11:
Closing for now since there has been no response.

closed 2026-11-11 by elguero