displaylink-rpm/displaylink-rpm #271: External monitors are really laggy

tsoernes, 2024-08-07:
* Are you using the latest DisplayLink driver?
    
```
dnf list --installed --cacheonly | grep --ignore-case displaylink                                                                                                                 (base) 09:23:01
displaylink.x86_64                                   1.14.4-2.github_evdi                @@commandline   
```
  * Are you using the latest EVDI module?
    
Yes
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
Yes
  * Is this issue related to evdi/kernel?
    
No
  * Is this issue related to DisplayLinkManager?
    
No
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora 
  * Version 40
  * Kernel version Linux torstein-latitude 6.9.12-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Jul 27 15:56:15 UTC 2024 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`
```
Creating symlink /var/lib/dkms/evdi/1.14.4/source -> /usr/src/evdi-1.14.4
Sign command: /lib/modules/6.9.12-200.fc40.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...
Building module(s)...
Signing module /var/lib/dkms/evdi/1.14.4/build/evdi.ko
Cleaning build area...

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.9.12-200.fc40.x86_64/extra/
depmod....
Warning: The unit file, source configuration file or drop-ins of systemd-modules-load.service changed on disk. Run 'systemctl daemon-reload' to reload units.

```
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
 ```
cat /var/lib/dkms/evdi/kernel-6.9.12-200.fc40.x86_64-x86_64/log/make.log                                                                                                          (base) 09:25:59
DKMS make.log for evdi-1.14.4 for kernel 6.9.12-200.fc40.x86_64 (x86_64)
ma. 05. aug. 13:15:28 +0200 2024
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.14.4/build SUBDIRS=/var/lib/dkms/evdi/1.14.4/build SRCROOT=/var/lib/dkms/evdi/1.14.4/build CONFIG_MODULE_SIG= -C /lib/modules/6.9.12-200.fc40.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/6.9.12-200.fc40.x86_64'
make --no-print-directory -C /usr/src/kernels/6.9.12-200.fc40.x86_64 \
-f /usr/src/kernels/6.9.12-200.fc40.x86_64/Makefile modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 14.1.1 20240701 (Red Hat 14.1.1-7)
  You are using:           gcc (GCC) 14.2.1 20240801 (Red Hat 14.2.1-1)
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/1.14.4/build need-builtin=1 need-modorder=1 
# cmd_mod /var/lib/dkms/evdi/1.14.4/build/evdi.mod
  printf '%s\n'   evdi_platform_drv.o evdi_platform_dev.o evdi_sysfs.o evdi_modeset.o evdi_connector.o evdi_encoder.o evdi_drm_drv.o evdi_fb.o evdi_gem.o evdi_painter.o evdi_params.o evdi_cursor.o evdi_debug.o evdi_i2c.o evdi_ioc32.o | awk '!x[$0]++ { print("/var/lib/dkms/evdi/1.14.4/build/"$0) }' > /var/lib/dkms/evdi/1.14.4/build/evdi.mod
# CC [M]  /var/lib/dkms/evdi/1.14.4/build/evdi_platform_drv.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.14.4/build/.evdi_platform_drv.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -fmin-function-alignment=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-overflow -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -g -Iinclude/uapi/drm -Iinclude/drm  -fsanitize=bounds-strict -fsanitize=shift    -DMODULE  -DKBUILD_BASENAME='"evdi_platform_drv"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.14.4/build/evdi_platform_drv.o /var/lib/dkms/evdi/1.14.4/build/evdi_platform_drv.c  
# CC [M]  /var/lib/dkms/evdi/1.14.4/build/evdi_platform_dev.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.14.4/build/.evdi_platform_dev.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -fmin-function-alignment=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-overflow -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -g -Iinclude/uapi/drm -Iinclude/drm  -fsanitize=bounds-strict -fsanitize=shift    -DMODULE  -DKBUILD_BASENAME='"evdi_platform_dev"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.14.4/build/evdi_platform_dev.o /var/lib/dkms/evdi/1.14.4/build/evdi_platform_dev.c  
# CC [M]  /var/lib/dkms/evdi/1.14.4/build/evdi_sysfs.o
[... 24 lines cut ...]
[... cut ...]
Skipping BTF generation for /var/lib/dkms/evdi/1.14.4/build/evdi.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.9.12-200.fc40.x86_64'
```

I have two external monitors (2x 2560x1600 @ 60 Hz) connected to a USB-C hub (Thinkpad USB-C Dock 40AF (https://support.lenovo.com/us/en/solutions/pd500180)) connected to a Dell Latitude 7340 with Intel Integrated graphics running on Wayland.

The external monitors are really laggy. Mouse movement is very visibly laggy. The laptop screen is not lagging when moving mouse cursor.

LtDragon, 2024-08-08:
I have a very similar set-up.
I have 2 1920*1080 @60hz monitors connected to a USB-C Hub TPA-B01.
I'm using a dell 5584.
When I use the laptop monitor I don't see any spike in PC usage, when I use my connected monitor I notice a large spike in CPU usage and it becoming laggy/showing a distorted image where ever the curse is.

mitmarcus, 2024-08-12:
Can confirm, my fan goes haywire if I use the dock with a monitor connected. However, if I disconnect the monitor and still leave the dock the fan returns to normal speeds.

johndrothe, 2024-09-15:
I have a similar set of symptoms when using Firefox. Minimizing the Firefox window drops the usage to 1-2%.

[image]

elguero, 2024-11-09:
Hola!

Is there still an issue with the latest release?

johndrothe, 2024-12-29:
> Hola!
> 
> Is there still an issue with the latest release?

The specific issue with Firefox continues with 1.14.7-4 on Fedora 41.

dmynarski, 2024-12-30:
I can confirm issue with Fedora 41 and Google Chrome 131.0.6778.204 on Lenovo Thinkpad P14s.
System goes really laggy with Displaylink rpm latest release and model DK1523.
RPM package installation went successfully.
This is not related to device its own as testing plug&play on Windows 10 works perfectly. I have other lenovo dock (basic dock 40AG) that did not require extra drivers and it works perfectly.

mctaguma, 2025-02-04:
I can confirm the same high CPU usage occurs on Rocky 9.5 running displaylink-rpm 1.14.7-4, when playing a YouTube video in Firefox on one of my monitors.
If I switch to a different tab during playback, or move the tab to my laptop screen, CPU usage decreases to normal.

Elu43, 2026-08-19:
Hi, same behaviour with latest release, nvidia GPU, and Fedora 44. It's not possible to work on the screen with the lag. Does anyone have a workaround ?