displaylink-rpm/displaylink-rpm #272: evdi module doesn't build against Fedora 40 kernel 6.10

benkay86, 2024-08-09:
The evdi module doesn't build against the latest kernel 6.10.3 on Fedora 40.

Output of `uname -a`:

```
Linux phalaenopsis 6.10.3-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Aug  5 14:30:00 UTC 2024 x86_64 GNU/Linux
```

Contents of `/var/log/displaylink/displaylink.log`:

```
Building module:
Cleaning build area...
Building module(s)...(bad exit status: 2)
Failed command:
make -j14 KERNELRELEASE=6.10.3-200.fc40.x86_64 all INCLUDEDIR=/lib/modules/6.10.3-200.fc40.x86_64/build/include KVERSION=6.10.3-200.fc40.x86_64 DKMS_BUILD=1
Error! Bad return status for module build on kernel: 6.10.3-200.fc40.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.14.4/build/make.log for more information.
```

Tail of `/var/lib/dkms/evdi/1.14.4/build/make.log`:

```
# CC [M]  /var/lib/dkms/evdi/1.14.4/build/evdi_ioc32.o
  gcc -Wp,-MMD,/var/lib/dkms/evdi/1.14.4/build/.evdi_ioc32.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/>
/var/lib/dkms/evdi/1.14.4/build/evdi_gem.c: In function ‘evdi_gem_vmap’:
/var/lib/dkms/evdi/1.14.4/build/evdi_gem.c:319:25: error: implicit declaration of function ‘vmap’; did you mean ‘kmap’? [-Wimplicit-function-declaration]
  319 |         obj->vmapping = vmap(obj->pages, page_count, 0, PAGE_KERNEL);
      |                         ^~~~
      |                         kmap
/var/lib/dkms/evdi/1.14.4/build/evdi_gem.c:319:23: error: assignment to ‘void *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  319 |         obj->vmapping = vmap(obj->pages, page_count, 0, PAGE_KERNEL);
      |                       ^
/var/lib/dkms/evdi/1.14.4/build/evdi_gem.c: In function ‘evdi_gem_vunmap’:
/var/lib/dkms/evdi/1.14.4/build/evdi_gem.c:355:17: error: implicit declaration of function ‘vunmap’; did you mean ‘kunmap’? [-Wimplicit-function-declaration]
  355 |                 vunmap(obj->vmapping);
      |                 ^~~~~~
      |                 kunmap
make[3]: *** [scripts/Makefile.build:244: /var/lib/dkms/evdi/1.14.4/build/evdi_gem.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/kernels/6.10.3-200.fc40.x86_64/Makefile:1946: /var/lib/dkms/evdi/1.14.4/build] Error 2
make[1]: *** [Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.10.3-200.fc40.x86_64'
make: *** [Makefile:38: all] Error 2
```

astromatt42, 2024-08-09:
Thanks for flagging this. I was just about to report the same issue.

astromatt42, 2024-08-09:
This probably relates to https://github.com/DisplayLink/evdi/issues/469
Looks like an upstream problem that is supposed to be fixed in Evdi v1.14.5. Need to update the displaylink-rpm packages.

elguero, 2024-08-09:
Yes, we need to update.  I saw that 1.14.6 is going to be tagged soon and I have been waiting for that.

[image]

astromatt42, 2024-08-09:
Great! Thanks!

benkay86, 2024-08-09:
Submitted PR #273, which fixes the issue for me. Don't know how long it will take for them to tag 1.14.6, but Fedora has just rolled out kernel 6.10 in Fedora 40, so a lot of people are going to encounter this bug when they upgrade. Please consider tagging a release based on 1.14.5 so Fedora users can upgrade right away.

mymistake, 2024-08-10:
I can confirm that this is working. I initially struggled that the EVDI package did not build without excluding the test directory.

benkay86, 2024-08-14:
@elguero, it appears [upstream](https://github.com/DisplayLink/evdi) has tagged version 1.14.6.  I tested on my Fedora 40 install with kernel 6.10.3.  It builds and works without any modifications beyond excluding the tests directory in `displaylink.spec` as in #273.

```
%exclude %{_prefix}/src/evdi-%{version}/tests
```

Hopefully this is a good opportunity for a new release?  That would fix both this issue and #274.

elguero, 2024-08-14:
Yep, I saw that and started testing this morning. thanks for your testing too ,😀

firewallzer0, 2024-08-16:
Just stumbled across the issue from updating my kernel last night. Installed the updated version and got my screens back.

But I now have a rolling shutter effect on a bounding box around my mouse that flickers as I move the mouse. This was not happening last night before the kernel and package update. If I disable the displaylink monitors or uninstall the displaylink package it goes away.

The flickering and bounding box issue is annoying enough that I have disabled my Displaylink monitors for the moment.

Using: displaylink-1.14.6-1.github_evdi.x86_64 from the release page: v6.0.0-1

OS: Fedora 40
`uname -a` Linux fedora 6.10.4-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Aug 11 15:32:50 UTC 2024 x86_64 GNU/Linux

Using Plasma X11

Not sure what additional information to give.
Just let me know what logs/information you want/need.
[image]

firewallzer0, 2024-08-16:
Never mind I looked at old issues, the software mouse was turned on. Same as this issue: https://github.com/displaylink-rpm/displaylink-rpm/issues/171

Turned the software mouse off, rebooted and now I am good!

benkay86, 2024-08-21:
This issue is resolved by the [release of 6.0.0-1](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v6.0.0-1), which upgrades to upstream evdi 1.14.6.

closed 2024-08-21 by benkay86