displaylink-rpm/displaylink-rpm #274: EVDI Module Fails to Build on Fedora Kernel `6.10.3-200.fc40.x86_64` - Missing `make.log` and Build Failures

asbor, 2024-08-10:
### **Summary:**
After upgrading to Fedora kernel `6.10.3-200.fc40.x86_64`, the DisplayLink driver fails to start, and the EVDI module does not build correctly. Despite multiple attempts to rebuild the module, the build process fails without generating a `make.log` file. Rolling back to an older kernel (`6.8.5-301.fc40.x86_64`) allows the module to build and function correctly.

### **System Information:**
- **Operating System:** Fedora 40
- **Kernel Version:**
  - **Working:** `6.8.5-301.fc40.x86_64`
  - **Not Working:** `6.10.3-200.fc40.x86_64`
- **DisplayLink Driver Version:** `1.14.4`
- **EVDI Module Version:** `1.14.4`
- **uname -a Output:**
  ```bash
  Linux fedora 6.10.3-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Aug 7 11:33:33 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
  ```

### **Logs:**
1. **DisplayLink Log Excerpt:**
   ```
   Error! Bad return status for module build on kernel: 6.10.3-200.fc40.x86_64 (x86_64)
   Failed command:
   make -j12 KERNELRELEASE=6.10.3-200.fc40.x86_64 all INCLUDEDIR=/lib/modules/6.10.3-200.fc40.x86_64/build/include KVERSION=6.10.3-200.fc40.x86_64 DKMS_BUILD=1
   ```
   - Full log available if required.

2. **EVDI Make Log:**
   - **Note:** The `make.log` file was not generated during the DKMS build process, indicating an early failure or misconfiguration.

### **Steps to Reproduce:**
1. Update Fedora to kernel `6.10.3-200.fc40.x86_64`.
2. Attempt to use the DisplayLink docking station.
3. Observe that external monitors are not recognized, and the `displaylink-driver` service fails to start.
4. Attempt to manually rebuild the EVDI module using DKMS, which fails without generating a `make.log` file.

### **Detailed Description:**
After upgrading the kernel, the EVDI module fails to build for kernel `6.10.3-200.fc40.x86_64`. Despite several attempts to manually trigger the DKMS build process, the build process fails with a "bad exit status" error, and no `make.log` file is created. Upon rolling back to kernel `6.8.5-301.fc40.x86_64`, the EVDI module builds and installs successfully, and the DisplayLink driver functions correctly.

### **Steps Taken:**
- Performed a clean installation of the DisplayLink driver as per the recommended process, including removing all related files, rebooting, and reinstalling.
- Attempted manual DKMS build and install commands, which failed without generating necessary logs.

### **Temporary Workaround:**
- Downgrade to an older kernel version (`6.8.5-301.fc40.x86_64`) and manually rebuild the EVDI module. This allows the DisplayLink driver to function correctly, but it's not a long-term solution as it prevents the use of newer kernels.

### **Request for Assistance:**
- Please provide guidance on how to resolve the build issues with the EVDI module on Fedora kernel `6.10.3-200.fc40.x86_64`. If there are any known patches or updates that address this issue, that information would be greatly appreciated.

asbor, 2024-08-10:
[displaylink-log.txt](https://github.com/user-attachments/files/16568895/displaylink-log.txt)

mymistake, 2024-08-10:
Hi,

You can either wait until the package here is built or you apply the fix described here https://github.com/displaylink-rpm/displaylink-rpm/issues/272 to "Makefile" and "displaylink.spec" yourself. Afterward, build with "make github-release" and install the updated "evd"i package.

I already did that and can confirm that both my external monitors and my HP docking station are working again with Fedora and the 6.10.3 kernel.

astromatt42, 2024-08-11:
I am not that familiar with packaging systems. It compiles and builds the package, however `dnf install` exits with "Error: Will not install a source rpm package" and `rpm -i` exits without any messages and seemingly without doing anything. Any ideas?

mymistake, 2024-08-11:
I think you used source RPM from the root directory - check the folder, a directory "x86_64" with "displaylink-1.14.5-2.github_evdi.x86_64.rpm" should have been generated if the build process was successful. Assuming you also have x64 architecture, you can install that with "sudo dnf install ./displaylink-1.14.5-2.github_evdi.x86_64.rpm".

astromatt42, 2024-08-11:
Of course! Thank you!

benkay86, 2024-08-21:
This appears to have been a duplicate of #272 and is resolved by the [release of 6.0.0-1](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v6.0.0-1), which upgrades to upstream evdi 1.14.6.  @asbor, could you please test against the latest release and mark the issue as closed?

asbor, 2024-08-27:
Thanks @benkay86 

I have just performed a update on fedora and i lost the monitors again. However, your solution did solve the issue. so everything is operational now again :1st_place_medal:

elguero, 2024-11-09:
Hola!

One thing that the logs show, the kernel headers were not present on the system for that version of the kernel that the drivers were trying to be built against.

It would appear that this has been resolved though.  I will go ahead and close the issue.

Thanks!

closed 2024-11-09 by elguero