displaylink-rpm/displaylink-rpm #275: add support for aarch64 rpm

mrinaldhillon, 2024-08-19:
* Fedora
  * 40
  * `Linux lnx 6.10.5-400.asahi.fc42.aarch64+16k #1 SMP PREEMPT_DYNAMIC Sat Aug 17 22:36:01 UTC 2024 aarch64 GNU/Linux`

We need support for Asahi Linux, which is Fedora on Apple Silicon Mac. Asahi Linux currently does not support display over Thunderbolt, so a display link is an ideal solution.

kiilo, 2024-08-28:
`make github-release` fails with:

```
...
+ /usr/bin/make -O -j8 V=1 VERBOSE=1
make[1]: Entering directory '/var/tmp/displayportC8H37J/displaylink-1.14.6/evdi-1.14.6/library'
gcc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64 -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer  $(pkg-config --cflags-only-I libdrm)   -c -o evdi_lib.o evdi_lib.c
Assembler messages:
Error: unknown architecture `x86-64'

Error: unrecognized option -march=x86-64
cc1: error: unrecognized command-line option ‘-m64’
cc1: error: unknown value ‘x86-64’ for ‘-march’
cc1: note: valid arguments are: armv8-a armv8.1-a armv8.2-a armv8.3-a armv8.4-a armv8.5-a armv8.6-a armv8.7-a armv8.8-a armv8.9-a armv8-r armv9-a armv9.1-a armv9.2-a armv9.3-a armv9.4-a native
cc1: error: ‘-fcf-protection=full’ is not supported for this target
make[1]: *** [<builtin>: evdi_lib.o] Error 1
make[1]: Leaving directory '/var/tmp/displayportC8H37J/displaylink-1.14.6/evdi-1.14.6/library'
error: Bad exit status from /var/tmp/displayportCJh5hj/rpm-tmp.A5TW6X (%build)

RPM build errors:
    Bad exit status from /var/tmp/displayportCJh5hj/rpm-tmp.A5TW6X (%build)
make: *** [Makefile:157: x86_64/displaylink-1.14.6-1-github_evdi.x86_64.rpm] Error 1
```

mrinaldhillon, 2024-08-30:
@kiilo The Makefile architecture is hardcoded to x86_64. It needs simple tweaks to add aarch64 support. Try my patch.

ssaavedra, 2024-09-05:
The pull request is merged. If you find problems with it, please reopen this issue.

closed 2024-09-05 by ssaavedra

kiilo, 2024-09-08:
great - works like a charm
anyone who want to rinse & repeat, did: 
```
git pull 
make github-release
sudo dnf install ./aarch64/displaylink-1.14.6-1.github_evdi.aarch64.rpm
```