displaylink-rpm/displaylink-rpm #277: Relating to updates

mitmarcus, 2024-09-09:
I have the github version installed *1.14.6-1, however dnf tries to update to 6.0.0
[image]

Is it possible to update it on the rpm repo? or should I disable updates entirely to the package?

elguero, 2024-11-09:
Hola @mitmarcus ,

Did you install from a `dnf` repository or did you install from the release here on GitHub?

mitmarcus, 2024-11-09:
installed from here. The one from fedora @system seems outdated.

mitmarcus, 2024-12-05:
Unfortunately, 6.1.0 no longer works. 
The one from GitHub is fine. The one from the official repo *that it always tries to update to* is not.

Am I the only one with this issue?

elguero, 2024-12-05:
Hola @mitmarcus ,

What do you mean "official repo"?  This project doesn't have one.

It sounds like you may need to disable whatever repo you have enabled on your system.

Releases from this project are currently only available here on GitHub.

mitmarcus, 2024-12-05:
[image]
[image]

It seems to be in fedora multimedia? 
[image]

elguero, 2024-12-05:
It looks like you have added a third party repo.  The repo is here: https://negativo17.org/

That has no relation to this project.  They might be repackaging our work from here which is causing the conflict on your system.

mitmarcus, 2024-12-05:
Hola @elguero!

Thank you! I don't know how I missed this until now; I'm not even using that repo anymore.

closed 2024-12-05 by mitmarcus