displaylink-rpm/displaylink-rpm #281: Fedora 41 Update

chiappone, 2024-11-04:
Are there plans to make this available to Fedora 41?

SimonSchwendele, 2024-11-08:
> Are there plans to make this available to Fedora 41?

A friend of mine uses displaylink on F41 without issues.
Regarding the Secureboot process:
Theres already a dkms signing key so the process should be fairly easy.
I described them [in my copr](https://copr.fedorainfracloud.org/coprs/crashdummy/Displaylink)

It's basically those 3 commands:
```bash
# dkms is a dependency of the displaylink package and probably already installed
$ sudo dnf install mokutil dkms

# Enroll the key dkms created automatically and enter a passphrase
$ sudo mokutil --import /var/lib/dkms/mok.pub

# Signing should be done automatically during upgrades.
# If the module is not signed you can invoke it manually 
$ sudo dkms autoinstall
```

aptinstalljosh, 2024-11-09:
> > Are there plans to make this available to Fedora 41?
> 
> A friend of mine uses displaylink on F41 without issues. Regarding the Secureboot process: Theres already a dkms signing key so the process should be fairly easy. I described them [in my copr](https://copr.fedorainfracloud.org/coprs/crashdummy/Displaylink)
> 
> It's basically those 3 commands:
> 
> ```shell
> # dkms is a dependency of the displaylink package and probably already installed
> $ sudo dnf install mokutil dkms
> 
> # Enroll the key dkms created automatically and enter a passphrase
> $ sudo mokutil --import /var/lib/dkms/mok.pub
> 
> # Signing should be done automatically during upgrades.
> # If the module is not signed you can invoke it manually 
> $ sudo dkms autoinstall
> ```

I get `Failed to get file status, /var/lib/dkms/mok.pub`, because there's no file there. How do I generate one? I've confirmed that secure boot is enabled.

elguero, 2024-11-09:
Hola!  A new package for 41 has been released.  The README instructions have been updated to reflect the simplified process for setting up DKMS key signing.  Thanks for all the input and helping each other out.

@aptinstalljosh Normally, that key is automatically generated the first time that DKMS is run.  So, if you have installed the released driver package, it will run DKMS to build the kernel module and the key should have been generated to sign it. https://github.com/dell/dkms?tab=readme-ov-file#module-signing

closed 2024-11-09 by elguero

SimonSchwendele, 2024-11-09:
> Hola! A new package for 41 has been released. The README instructions have been updated to reflect the simplified process for setting up DKMS key signing. Thanks for all the input and helping each [...]

I saw that the [displaylink installer](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu) got updated to 6.1 as well.
It might be a good idea to update that one as well.

```
    - DL−7xxx series: LG 34WK95UP monitor was not working properly on DisplayPort. (VIFR-5046)                                                              
    - DL−7xxx series: Improved interoperability with DisplayPort monitors at 4K 144 Hz. (VIFR-4803)                                                         
    - DL−7xxx series: Failed to get EDID from some HDMI monitors. (VIFR-4762)                                                                               
    - DL−7xxx series: 4K HDMI monitors appear at 30 Hz by default. (VIFR-4712)                                                                              
    - DL-7xxx series: incorrectly offer 4Kp120 mode on HDMI. (VIFR-4619)                                                                                    
    - DL-7xxx series: 4Kp120 modes are not available on DP. (VIFR-4602)                                                                                     
    - Incorrect color profile in some reduced range modes on HDMI display. (VIFR-4094)                                                                      
    - Incorrect scaling and blinking windows on Wayland. (PS-3915)                                                                                          
    - New support tool 
    ```

elguero, 2024-11-10:
Oh wow.  I totally missed that.  I normally check that frequently.  Thank you!!

SimonSchwendele, 2024-11-10:
> Oh wow. I totally missed that. I normally check that frequently. Thank you!!

I do so as well.
That's why I think it's weird that it's mentioned at being updated in September but I check this at least once a week :/