displaylink-rpm/displaylink-rpm #282: Failed rebuilding fedora 41 srpm

enriqueartal, 2024-11-10:
* Are you using the latest DisplayLink driver? The one provided by the src rpm
    
  * Are you using the latest EVDI module? That's the issue, rpmbuild produces 1.14.4 version
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website? No problem with the provided rpm
    
  * Is this issue related to evdi/kernel? See above
    
  * Is this issue related to DisplayLinkManager? See above
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver? Previous version uninstalled
    

  * Fedora 
  * Version 41
  * Kernel version Linux singular 6.11.6-300.fc41.x86_64 SMP PREEMPT_DYNAMIC Fri Nov  1 16:16:00 UTC 2024 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log` Not available
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log` Not available

The rebuilding process produces evdi-1.14.4 and it stops

elguero, 2024-11-10:
Hola!

Can you provide the command being used to rebuild from the SRPM?  Can you provide any messages from your console?

Thank you!

enriqueartal, 2024-11-10:
The command is `rpmbuild --rebuild fedora-41-displaylink-1.14.7-1.github_evdi.src.rpm` and the messages are in [displaylink.log](https://github.com/user-attachments/files/17691819/displaylink.log)
¡Gracias!

elguero, 2024-11-10:
Okay, I see what happened.

The version that DisplayLink released at the time, 6.0, was released with evdi v1.14.6.  When running just `--rebuild` the bundled version of evdi will be used but the spec file is looking for v1.14.7.

If you run the following command `rpmbuild --rebuild fedora-41-displaylink-1.14.7-1.github_evdi.src.rpm --define "_github 1"`, then it would use the release from the evdi repo which contained fixes not present in v1.14.6.

I know this is not ideal and can be confusing.  I am going to think through if there is a more elegant way to handle this.  Since you are using Fedora 41, you are better off using the release from the evdi repo anyways.

enriqueartal, 2024-11-11:
I installed the rpm from this repository, is there any dnf repo elsewhere? I did not need to rebuild the srpm, I wanted to check if it worked. I attach the the log with the errors after `rpmbuild --rebuild fedora-41-displaylink-1.14.7-1.github_evdi.src.rpm --define "_github 1"`. 
On the other side, running just rpmbuild and looking at the BUILD folder, it seems it compiles 1.14.4.
As I said, it works for me, I wanted only to report the errors.
[displaylink.log](https://github.com/user-attachments/files/17698398/displaylink.log)

elguero, 2026-04-28:
¡Hola! Thank you for the info and I apologize for not acknowledging this sooner.