displaylink-rpm/displaylink-rpm #285: RockyLinux 9.5 with DL 6.1.0  (evdi 1.14.7-2): Autoinstall on 5.14.0-503.14.1.el9_5.x86_64 failed for module(s) evdi(10)

jmsdc, 2024-11-24:
Hello :wave:

Installing the RPM package from release version v6.1.0, I encounter an issue at the `dkms autoinstall` step

  * _Are you using the latest driver?_  Yes, from shared RPM in Releases [v6.1.0: DisplayLink 6.1.0](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v6.1.0)
  * _Are you using the latest EVDI version?_  The version is 1.14.7-2 fro the rpm
  * _If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?_  Yes I checked
  * _Is this issue related to evdi/kernel?_ I did not found related to the same issue

```
~ sudo dkms autoinstall                            

Sign command: /lib/modules/5.14.0-503.14.1.el9_5.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=5.14.0-503.14.1.el9_5.x86_64 all INCLUDEDIR=/lib/modules/5.14.0-503.14.1.el9_5.x86_64/build/include KVERSION=5.14.0-503.14.1.el9_5.x86_64 DKMS_BUILD=1
Error! Bad return status for module build on kernel: 5.14.0-503.14.1.el9_5.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.14.7/build/make.log for more information.
Autoinstall on 5.14.0-503.14.1.el9_5.x86_64 failed for module(s) evdi(10).

Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.

```

Here is the execution result log file [make.log](https://github.com/user-attachments/files/17892690/make.log)

The target device is Dell USB3.0 dock D3100

```
~ uname -r
5.14.0-503.14.1.el9_5.x86_64
```

```
NAME="Rocky Linux"
VERSION="9.5 (Blue Onyx)"
ID="rocky"
ID_LIKE="rhel centos fedora"
VERSION_ID="9.5"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Rocky Linux 9.5 (Blue Onyx)"
ANSI_COLOR="0;32"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:rocky:rocky:9::baseos"
HOME_URL="https://rockylinux.org/"
VENDOR_NAME="RESF"
VENDOR_URL="https://resf.org/"
BUG_REPORT_URL="https://bugs.rockylinux.org/"
SUPPORT_END="2032-05-31"
ROCKY_SUPPORT_PRODUCT="Rocky-Linux-9"
ROCKY_SUPPORT_PRODUCT_VERSION="9.5"
REDHAT_SUPPORT_PRODUCT="Rocky Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.5"
```

pierre-at-atos, 2024-11-25:
Same error on AlmaLinux 9.5

elguero, 2024-11-28:
Thank you for the report.  It looks like some upstream kernel changes have made its way into the RHEL / Rocky Linux/ AlmaLinux kernels.

jgunther, 2024-11-28:
I'm experiencing the same issue on Red Hat Enterprise Linux 9.5 running kernel 5.14.0-503.15.1.

closed 2024-11-29 by elguero

elguero, 2024-11-29:
Created a patch to address the new DRM changes from kernel 6.11 that were merged into the RHEL kernel released in 9.5.  Which means it is also in Rocky Linux and AlmaLinux.  A new release has been made to include this patch.

jmsdc, 2024-11-29:
Hello @elguero
Thanks for your quick action, it works fine now with the [v6.1.0-1](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v6.1.0-1) version

commit f45d3702 (2024-11-29): Fix building on EL 9.5 kernels

DRM changes from upstream up to 6.11 were merged into EL 9.5 kernels.

Adding a patch to fix building evdi.

Fixes #285