displaylink-rpm/displaylink-rpm #286: Displays not working on Asahi Fedora 41 / Kernel 6.12.1

ArchitektApx, 2024-12-04:
* Are you using the latest DisplayLink driver?
    

- [X] yes, I tried `make`, `make github-release` and the release rpm for fedora 41

* Are you using the latest EVDI module?
    
- [X] yes, i currently run the rpm built by `make github-release`
  
* If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
- [X] yes I tried everything, the dock works correctly on other devices, has worked yesterday before I upgraded to fedora 41 and the dock is visible to the system: 
```shell 
>lsusb -d 17e9 
Bus 002 Device 004: ID 17e9:6000 DisplayLink Hybrid Dock 1001
``` 

  * Is this issue related to evdi/kernel?
    
- [ ] not sure?
* Is this issue related to DisplayLinkManager?
    

- [ ] not sure?  
* If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

- [X] I did multiple attempts of clean uninstalls/installs as requested by the template

  * Fedora / CentOS Stream / Rocky Linux / AlmaLinux OS 
  `Fedora`
  * Version
  `Asahi Remix 41 (Workstation Edition) aarch64`
  * Kernel version `(uname -a)`
  `Linux KnechRootrecht 6.12.1-404.asahi.fc41.aarch64+16k #1 SMP PREEMPT_DYNAMIC Mon Dec  2 23:05:31 UTC 2024 aarch64 GNU/Linux`

  * `/var/log/displaylink/displaylink.log`
  ```
reating symlink /var/lib/dkms/evdi/1.14.7/source -> /usr/src/evdi-1.14.7

Sign command: /lib/modules/6.12.1-404.asahi.fc40.aarch64+16k/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)... done.
Signing module /var/lib/dkms/evdi/1.14.7/build/evdi.ko
Cleaning build area... done.

Installing /lib/modules/6.12.1-404.asahi.fc40.aarch64+16k/extra/evdi.ko.xz
Running depmod.... done.
Module evdi 1.14.6 is not installed for kernel 6.9.12-401.asahi.fc40.aarch64+16k (aarch64). Skipping...
Module evdi 1.14.6 is not built for kernel 6.9.12-401.asahi.fc40.aarch64+16k (aarch64). Skipping...

Module evdi-1.14.6 for kernel 6.11.6-401.asahi.fc40.aarch64+16k (aarch64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.11.6-401.asahi.fc40.aarch64+16k/extra/evdi.ko.xz
Running depmod.... done.

Module evdi-1.14.6 for kernel 6.11.8-400.asahi.fc40.aarch64+16k (aarch64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.11.8-400.asahi.fc40.aarch64+16k/extra/evdi.ko.xz
Running depmod.... done.

Module evdi-1.14.7 for kernel 6.12.1-404.asahi.fc40.aarch64+16k (aarch64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.12.1-404.asahi.fc40.aarch64+16k/extra/evdi.ko.xz
Running depmod.... done.

[... 193 lines cut ...]

I tried the rpm release, and the srpms produced by both `make` and `make github-release` multiple times, cleaning up any leftovers between attempts. 

Installs seem to work properly and both the dkms module and displaylink service is available, as is the dock itself:
```shell
> dkms status
evdi/1.14.7, 6.12.1-404.asahi.fc41.aarch64+16k, aarch64: installed
> systemctl status displaylink-driver.service 
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/usr/lib/systemd/system/displaylink-driver.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf, 50-keep-warm.conf
     Active: active (running) since Wed 2024-12-04 10:24:09 CET; 23min ago
 Invocation: 8514ab741c9646dda9c0cd40170c3e80
   Main PID: 5458 (DisplayLinkMana)
      Tasks: 47 (limit: 19139)
     Memory: 17.2M (peak: 21.3M)
        CPU: 1.190s
     CGroup: /system.slice/displaylink-driver.service
             └─5458 /usr/libexec/displaylink/DisplayLinkManager

Dez 04 10:24:09 KnechRootrecht systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
Dez 04 10:24:09 KnechRootrecht systemd[1]: Started displaylink-driver.service - DisplayLink Driver Service.
> lsusb -d 17e9:
Bus 002 Device 004: ID 17e9:6000 DisplayLink Hybrid Dock 1001

ArchitektApx, 2024-12-04:
The following workaround fixed my issue for now: 

1. clean uninstall of displaylink package
2. Appling the following patch from https://github.com/DisplayLink/evdi/pull/498 for example by appending to `el9_5-build-fixes-and-el-audit-updates.patch`
```patch
diff --git a/module/evdi_drm_drv.c b/module/evdi_drm_drv.c
index fb6b6a8..13e674b 100644
--- a/module/evdi_drm_drv.c
+++ b/module/evdi_drm_drv.c
@@ -73,6 +73,10 @@ static const struct file_operations evdi_driver_fops = {
 #endif
 
 	.llseek = noop_llseek,
+
+#if defined(FOP_UNSIGNED_OFFSET)
+	.fop_flags = FOP_UNSIGNED_OFFSET,
+#endif
 };
 
 #if KERNEL_VERSION(5, 11, 0) <= LINUX_VERSION_CODE || defined(EL8)
```
3. Build/Install/Reboot
```shell
make github-release
rpmbuild --rebuild displaylink-1.14.7-3.github_evdi.src.rpm --define "_github 1"
sudo dnf install ./aarch64/displaylink-1.14.7-3.github_evdi.aarch64.rpm
sudo reboot
```

elguero, 2024-12-11:
Hi!

Thank you for the update.  We have added the patch from upstream which should resolve this for future installs.

closed 2024-12-11 by elguero