displaylink-rpm/displaylink-rpm #288: EVDI failing to compile/install on RHEL 9.4

mcsosa121, 2024-12-11:
* Are you using the latest DisplayLink driver?
    
Yes
  * Are you using the latest EVDI module?
    
Yes
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
no
  * Is this issue related to evdi/kernel?
    
   Yes
  * Is this issue related to DisplayLinkManager?
    
  no
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    
       N/A

  * Fedora / CentOS Stream / Rocky Linux / AlmaLinux OS
  * Version
  * Kernel version `(uname -a)`
 
RHEL 9.4, Kernel Version: Linux localhost.localdomain 5.14.0-503.16.1.el9_5.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Nov 21 07:26:21 EST 2024 x86_64 x86_64 x86_64 GNU/Linux

  * `/var/log/displaylink/displaylink.log`
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log`
The /var/log/displaylink/displaylink.log
```
Creating symlink /var/lib/dkms/evdi/1.14.7/source -> /usr/src/evdi-1.14.7

Sign command: /lib/modules/5.14.0-503.16.1.el9_5.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=5.14.0-503.16.1.el9_5.x86_64 all INCLUDEDIR=/lib/modules/5.14.0-503.16.1.el9_5.x86_64/build/include KVERSION=5.14.0-503.16.1.el9_5.x86_64 DKMS_BUILD=1

Error! Bad return status for module build on kernel: 5.14.0-503.16.1.el9_5.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.14.7/build/make.log for more information.

Sign command: /lib/modules/5.14.0-503.16.1.el9_5.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=5.14.0-503.16.1.el9_5.x86_64 all INCLUDEDIR=/lib/modules/5.14.0-503.16.1.el9_5.x86_64/build/include KVERSION=5.14.0-503.16.1.el9_5.x86_64 DKMS_BUILD=1

Error! Bad return status for module build on kernel: 5.14.0-503.16.1.el9_5.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.14.7/build/make.log for more information.

Deleting module evdi/1.14.7 completely from the DKMS tree.
Creating symlink /var/lib/dkms/evdi/1.14.7/source -> /usr/src/evdi-1.14.7

Sign command: /lib/modules/5.14.0-503.16.1.el9_5.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=5.14.0-503.16.1.el9_5.x86_64 all INCLUDEDIR=/lib/modules/5.14.0-503.16.1.el9_5.x86_64/build/include KVERSION=5.14.0-503.16.1.el9_5.x86_64 DKMS_BUILD=1

Error! Bad return status for module build on kernel: 5.14.0-503.16.1.el9_5.x86_64 (x86_64)
[... 88 lines cut ...]
ERROR: modpost: module evdi uses symbol dma_buf_vmap from namespace DMA_BUF, but does not import it.
ERROR: modpost: module evdi uses symbol dma_buf_end_cpu_access from namespace DMA_BUF, but does not import it.
ERROR: modpost: module evdi uses symbol dma_buf_begin_cpu_access from namespace DMA_BUF, but does not import it.
ERROR: modpost: module evdi uses symbol dma_buf_vunmap from namespace DMA_BUF, but does not import it.
make[2]: *** [scripts/Makefile.modpost:128: /var/lib/dkms/evdi/1.14.7/build/Module.symvers] Error 1
make[1]: *** [Makefile:1858: modules] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-503.16.1.el9_5.x86_64'
make: *** [Makefile:38: all] Error 2
```

Trying to install displaylink drivers for RHEL 9.4 on a Panasonic Toughbook CF33-MK3

1. Clean install of OS (RHEL 9.4)
2. Enable EPEL Repositories 
3. Run `dnf upgrade`
4. Run `dnf install kernel-headers gcc gcc-c++ libdrm-devel rpmdevtools rpmlint`
5. Run `rpmdev-setuptree`
6. Download the displaylink rpm repo 
7. Run `sudo make github-release` inside the downloaded repo
8. Run `sudo rpmbuild --rebuild <name-of-generated-pkg>.rpm --define "_github 1"
9. Run `sudo dnf install x86_64/<name-of-generated-pkg>.rpm`

Like the title says, I've been trying to get this package installed and am having issues specifically with evdi not building correctly. I've looked at various issues (which led me to take some of the steps above such as #286 #232 #263 and #242 . I've attempted to install from the compiled source and also the "Rocky package" and neither have worked. 

Any help would be appreciated. Thanks!

mcsosa121, 2024-12-11:
Additional helpful logs
Log of running `make github-release`
```
wget --no-verbose -O DisplayLink_USB_Graphics_Software_for_Ubuntu_6.1.0-17.zip \
	"https://www.synaptics.com/sites/default/files/exe_files/2024-10/DisplayLink USB Graphics Software for Ubuntu6.1-EXE.zip"
wget --no-verbose -O v1.14.7.tar.gz \
	https://github.com/DisplayLink/evdi/archive/v1.14.7.tar.gz
rpmbuild -bb --define "_topdir `pwd`" --define "_sourcedir `pwd`" --define "_rpmdir `pwd`" --define "_specdir `pwd`" --define "_srcrpmdir `pwd`" --define "_buildrootdir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_builddir `mktemp -d /var/tmp/displayportXXXXXX`" --define "_release 4" --define "_daemon_version 6.1.0-17" --define "_version 1.14.7" --define "_tmppath `mktemp -d /var/tmp/displayportXXXXXX`" --define "_github 1" displaylink.spec --target=x86_64
Building target platforms: x86_64
Building for target x86_64
setting SOURCE_DATE_EPOCH=1733875200
Executing(%prep): /bin/sh -e /var/tmp/displayport0mUCLC/rpm-tmp.5Edp1g
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 6.1.0
Patch #0 (align-with-linux-v6.11-plus.patch):
patching file module/evdi_modeset.c
Patch #100 (0001-Patch-for-kernel-6.12.patch):
patching file module/evdi_drm_drv.c
Patch #101 (0001-Fix-build-for-6.14-rc3.patch):
patching file module/evdi_gem.c
patching file module/evdi_painter.c
Patch #1 (el9_5-build-fixes-and-el-audit-updates.patch):
patching file module/evdi_drm_drv.c
Hunk #2 succeeded at 130 (offset 4 lines).
patching file module/evdi_drm_drv.h
patching file module/evdi_fb.c
patching file module/evdi_gem.c
patching file module/evdi_i2c.c
patching file module/evdi_modeset.c
patching file module/evdi_painter.c
patching file module/evdi_platform_dev.c
patching file module/evdi_platform_dev.h
Executing(%build): /bin/sh -e /var/tmp/displayport0mUCLC/rpm-tmp.iFgXsl
make[1]: Entering directory '/var/tmp/displayportSIyu9x/displaylink-1.14.7/evdi-1.14.7/library'
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64  $(pkg-config --cflags-only-I libdrm)   -c -o evdi_procfs.o evdi_procfs.c
make[1]: Leaving directory '/var/tmp/displayportSIyu9x/displaylink-1.14.7/evdi-1.14.7/library'
make[1]: Entering directory '/var/tmp/displayportSIyu9x/displaylink-1.14.7/evdi-1.14.7/library'
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64  $(pkg-config --cflags-only-I libdrm)   -c -o evdi_lib.o evdi_lib.c
make[1]: Leaving directory '/var/tmp/displayportSIyu9x/displaylink-1.14.7/evdi-1.14.7/library'
make[1]: Entering directory '/var/tmp/displayportSIyu9x/displaylink-1.14.7/evdi-1.14.7/library'
cc evdi_lib.o evdi_procfs.o -shared -Wl,-soname,libevdi.so.1 -o libevdi.so.1.14.7 -lc -lgcc 
cp libevdi.so.1.14.7 libevdi.so
ln -sf libevdi.so.1.14.7 libevdi.so.1
make[1]: Leaving directory '/var/tmp/displayportSIyu9x/displaylink-1.14.7/evdi-1.14.7/library'
Executing(%install): /bin/sh -e /var/tmp/displayport0mUCLC/rpm-tmp.VF7sTH
/var/tmp/displayportgMgfpn/displaylink-1.14.7-4.github_evdi.x86_64/usr/src/evdi-1.14.7 /var/tmp/displayportSIyu9x/displaylink-1.14.7
/var/tmp/displayportSIyu9x/displaylink-1.14.7
mangling shebang in /usr/src/evdi-1.14.7/dkms_install.sh from /usr/bin/env bash to #!/usr/bin/bash
Processing files: displaylink-1.14.7-4.github_evdi.x86_64
Executing(%license): /bin/sh -e /var/tmp/displayport0mUCLC/rpm-tmp.BcgaLR
Provides: bundled(libevdi) = 1.14.7 config(displaylink) = 1.14.7-4.github_evdi displaylink = 1.14.7-4.github_evdi displaylink(x86-64) = 1.14.7-4.github_evdi libevdi.so.1()(64bit)
Requires(interp): /bin/sh /bin/sh /bin/sh
Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
Requires(post): /bin/sh
Requires(preun): /bin/sh
Requires(postun): /bin/sh
Requires: /usr/bin/bash ld-linux-x86-64.so.2()(64bit) ld-linux-x86-64.so.2(GLIBC_2.3)(64bit) libc.so.6()(64bit) libc.so.6(GLIBC_2.14)(64bit) libc.so.6(GLIBC_2.15)(64bit) libc.so.6(GLIBC_2.2.5)(64bit) libc.so.6(GLIBC_2.3.2)(64bit) libc.so.6(GLIBC_2.3.4)(64bit) libc.so.6(GLIBC_2.33)(64bit) libc.so.6(GLIBC_2.4)(64bit) libc.so.6(GLIBC_2.7)(64bit) libc.so.6(GLIBC_2.8)(64bit) libc.so.6(GLIBC_2.9)(64bit) libdl.so.2()(64bit) libdl.so.2(GLIBC_2.2.5)(64bit) libgcc_s.so.1()(64bit) libgcc_s.so.1(GCC_3.0)(64bit) libm.so.6()(64bit) libm.so.6(GLIBC_2.2.5)(64bit) libm.so.6(GLIBC_2.27)(64bit) libm.so.6(GLIBC_2.29)(64bit) libpthread.so.0()(64bit) libpthread.so.0(GLIBC_2.2.5)(64bit) libpthread.so.0(GLIBC_2.3.2)(64bit) libpthread.so.0(GLIBC_2.3.3)(64bit) libpthread.so.0(GLIBC_2.30)(64bit) librt.so.1()(64bit) librt.so.1(GLIBC_2.2.5)(64bit) libstdc++.so.6()(64bit) libstdc++.so.6(CXXABI_1.3)(64bit) libstdc++.so.6(CXXABI_1.3.3)(64bit) libstdc++.so.6(CXXABI_1.3.9)(64bit) libstdc++.so.6(GLIBCXX_3.4)(64bit) libstdc++.so.6(GLIBCXX_3.4.11)(64bit) libstdc++.so.6(GLIBCXX_3.4.14)(64bit) libstdc++.so.6(GLIBCXX_3.4.15)(64bit) libstdc++.so.6(GLIBCXX_3.4.18)(64bit) libstdc++.so.6(GLIBCXX_3.4.19)(64bit) libstdc++.so.6(GLIBCXX_3.4.20)(64bit) libstdc++.so.6(GLIBCXX_3.4.21)(64bit) libstdc++.so.6(GLIBCXX_3.4.22)(64bit) libstdc++.so.6(GLIBCXX_3.4.25)(64bit) libstdc++.so.6(GLIBCXX_3.4.26)(64bit) libstdc++.so.6(GLIBCXX_3.4.9)(64bit) libusb-1.0.so.0()(64bit) libuuid.so.1()(64bit) libuuid.so.1(UUID_1.0)(64bit) rtld(GNU_HASH)
Conflicts: mutter < 3.32 xorg-x11-server-Xorg = 1.20.1
Checking for unpackaged file(s): /usr/lib/rpm/check-files /var/tmp/displayportgMgfpn/displaylink-1.14.7-4.github_evdi.x86_64
[... cut ...]
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=5.14.0-503.16.1.el9_5.x86_64 all INCLUDEDIR=/lib/modules/5.14.0-503.16.1.el9_5.x86_64/build/include KVERSION=5.14.0-503.16.1.el9_5.x86_64 DKMS_BUILD=1

Error! Bad return status for module build on kernel: 5.14.0-503.16.1.el9_5.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.14.7/build/make.log for more information.
```

elguero, 2024-12-12:
Hi @mcsosa121 ,

Thank you for the detailed report.  I see what the issue is and apologize for missing this and not testing EL9 before rushing out this latest release.

I am working on a fix and testing it.

closed 2024-12-12 by elguero

mcsosa121, 2024-12-13:
The fix worked perfectly. Thank you so much.

commit aa1003e1 (2024-12-12): Add a patch to fix recent change upstream kernel 6.13

The patch upstream for kernel 6.13 breaks older kernel builds.

Fixes #288