displaylink-rpm/displaylink-rpm #289: EVDI not working after Fedora update

timo-a, 2024-12-21:
* Are you using the latest DisplayLink driver?
Yes: displaylink.x86_64 1.14.7-4.github_evdi

    
  * Are you using the latest EVDI module?
    
  * If you are using a DisplayLink device, have you checked 'troubleshooting'
    on DisplayLink's website?
    
  * Is this issue related to evdi/kernel?
    
  * Is this issue related to DisplayLinkManager?
    
  * If you are having an issue upgrading, have you tried to do a clean install of the driver?
    

  * Fedora
  * Version 41
  * Kernel version `(uname -a)`: `Linux my-pc 6.12.5-200.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Dec 15 16:48:23 UTC 2024 x86_64 GNU/Linux`

  * `/var/log/displaylink/displaylink.log` [make.log](https://github.com/user-attachments/files/18219687/make.log)
  * `/var/lib/dkms/evdi/kernel-[version]/log/make.log` [displaylink.log](https://github.com/user-attachments/files/18219690/displaylink.log)

### Please add a good summary of the issue in the Title above
From my limited understanding, I guess that this repo needs to adapt to the latest release of evdi: https://github.com/DisplayLink/evdi/releases/tag/v1.14.8 

### Steps to reproduce
Be on 6.12.4-200.fc41.x86_64, update to 6.12.5-200.fc41.x86_64

### Detailed description
About once every quarter, after a fedora update, I can no longer access my two monitors through my lenovo dockingstation. Habitually, I come to this repository and see if somebody has already filed an Issue, such as: https://github.com/displaylink-rpm/displaylink-rpm/issues/259, https://github.com/displaylink-rpm/displaylink-rpm/issues/272. I guess this time around I'm the one to file the issue.

LtDragon, 2024-12-23:
I have had the same issue updating to 6.12.4 and 6.12.5

eduiam, 2024-12-25:
I went to use my ASUS MB169B+ USB monitor today which I successfully installed in July 2024 and now lost connectivity with Fedora 41 ;-(

LtDragon, 2025-01-12:
Is anyone else still having this issue?

ngmarchant, 2025-01-12:
> Is anyone else still having this issue?

Just tested on 6.12.8-200.fc41.x86_64 and it doesn't work for me on a HP USB-C/A Universal Dock G2. The last working version is 6.11.4-301.fc41.x86_64.

josephdpower, 2025-01-16:
> Is anyone else still having this issue?

Tested on 6.12.9-200.fc41.x86_64 via a Lenovo Thunderbolt 40AF dock, not working for me.

astromatt42, 2025-01-21:
I'm now on 6.12.9-200.fc41.c86_64 and things are working again. Strange problem...

josephdpower, 2025-01-22:
> I also have it working on 6.12.9 but I had to remove it first and then go through and clear out all of my of EVDI packages. There was a lot of of broken packages that had gotten in there. Once I cleaned everything up, it all started working again.

Sorry for noob question but what method did you use to clear up your EVDI packages? I was following [this guide](https://alessandroferrari.live/install-displaylink-fedora39/) but it looks like I still had packages persisting when I got to the final step.

ngmarchant, 2025-01-22:
> Sorry for noob question but what method did you use to clear up your EVDI packages?

I used the instructions from [here](https://github.com/displaylink-rpm/displaylink-rpm/blob/master/.github%2FISSUE_TEMPLATE.md#L16)

Tierran, 2025-01-22:
@josephdpower I did two different things. First I went to :

`/var/lib/dkms/evdi` and removed all of my old versions that were in there. I was still having problems with old versions showing up in DNF though. In order to get rid of them, I had to use the `noscripts` flag like so:

`sudo dnf remove evdi-dkms-<version> --setopt=tsflags=noscripts`

Once I did that I was able to install the driver as normal.

josephdpower, 2025-02-15:
> [@josephdpower](https://github.com/josephdpower) I did two different things. First I went to :
>
> `/var/lib/dkms/evdi` and removed all of my old versions that were in there. I was still having [...]

Hey! Just realised I forgot to say thank you, I reinstalled and removed and now all working :)

elguero, 2025-04-28:
¡Hola!  @timo-a Is this still an issue on the latest release? Thank you.

timo-a, 2025-04-29:
I have not installed the latest release yet. As of now it is working with:
````
$ dnf list installed "displaylink*"
#...
displaylink.x86_64 1.14.9-1.github_evdi @commandline
$ rpm -qa kernel
kernel-6.13.10-200.fc41.x86_64
kernel-6.13.11-200.fc41.x86_64
kernel-6.13.12-200.fc41.x86_64
$ dkms status
#...
evdi/1.14.9, 6.13.10-200.fc41.x86_64, x86_64: installed
evdi/1.14.9, 6.13.11-200.fc41.x86_64, x86_64: installed
evdi/1.14.9, 6.13.12-200.fc41.x86_64, x86_64: installed
````

closed 2025-04-29 by elguero