displaylink-rpm/displaylink-rpm #290: Fedora 41... not working at all

ironhak, 2025-01-24:
Hello, you can check here for a detailed report on my system: https://discussion.fedoraproject.org/t/lenovo-thinkpad-docking-gen-2-not-working-second-monitor/142911

Long story short: I have a ThinkPad T490 (intel) with a Lenovo ThinkPad Docking Station Gen 2. Monitors won't work. I installed the driver from the release page but it does not work... nothing changes at all. 

Please help.

DMario93, 2025-01-27:
Similar issues here with a Dell XPS 16. 

Either installing the pre-built RPM or building it from source (and then installing it) doesn't seem to provide evdi. Indeed, there's no evdi.ko (or evdi.ko.xz) in /lib/modules/$(uname -r)/extra/ or anywhere else in the filesystem

I have been trying to install it manually from the original repo but with scarce results

elguero, 2025-01-27:
@ironhak Sorry for the issues that you are having.  In doing some research on your dock, I am not sure that it has a Displaylink chipset.  The thread that you linked to seems to confirm that.  So, the driver here will not work for you since you are not working with a Displaylink device.  Sorry that we can provide more assistance.

@DMario93 Are you using the same docking station?  If the evdi driver is not being installed, you should look at the logs under `/var/lib/dkms/evdi/kernel-[version]/log/make.log` to see what happened while trying to build the driver.

DMario93, 2025-01-27:
Hi @elguero thank you for helping us out

I am on a Dell D6000 I think. I have used it in the past through the driver provided by this repo. 

So, I tried to:

1. `git clone` this repo; `make`; `sudo dnf install` the RPM.

    **Important:** if I only `make`, I get a `displaylink-1.14.7-4.src.rpm`. If I `make github-release`, I instead get a `displaylink-1.14.7-4.github_evdi.src.rpm`.
    In both cases unfortunately, when I `dnf install` the package I get no evdi anywhere in the filesystem (I checked this via sudo find / -name '\*evdi\*').
    
   Why source rmps? Could they be the issue?
   
   Also, no folder `evdi` is created in `/var/lib/dkms` which makes me think the module was never signed. 

2. `git clone` the original [evdi](https://github.com/DisplayLink/evdi) repo; `make`; `sudo make install`
    
    The evdi.ko file gets written to `/usr/lib/modules/6.12.10-200.fc41.x86_64/kernel/drivers/gpu/drm/evdi/evdi.ko`
    No output in `/var/lib/dkms` which only contains my signing keys.
    
   During `sudo make install` I get this SSL error: BIO routines::no such file: crypto/bio/bss_file.c:75. There are open issues on the original repo about this, but I haven’t found any working solution.

   After "installing" I see no evdi in the output of `lsmod` (I am assuming I should). I even tried disabling Secure Boot: unfortunately same results

DMario93, 2025-01-27:
Ok. So, I followed this [guide](https://alessandroferrari.live/install-displaylink-fedora39/) and installed the x86_64 package for fedora **40** (although I am on 41). Immediately after installation, one of my external screens turned on.

Also, when installing, it pulled these  dependencies as well 

[image]

ironhak, 2025-01-27:
> Ok. So, I followed this [guide](https://alessandroferrari.live/install-displaylink-fedora39/) and installed the x86_64 package for fedora **40** (although I am on 41). Immediately after installation, one of my external screens turned on.
> 
> Also, when installing, it pulled these dependencies as well
> 
> [image]
> 
> [EDIT] the screens sometimes don't refresh properly. Assuming it's not an issue with the nvidia drivers, I guess it would be helpful to have a proper updated module for Fedora 41. For now it works though

Hi, why you installed those x11 dependencies? Are you on xorg or wayland? Also, what DE are you using?

DMario93, 2025-01-27:
Hi @ironhak 

To be honest, I have no idea. Those dependencies get automatically included when dnf installing the display link driver.

I am using Wayland, but I can always start an X11 session (or at least I could until Fedora 40)

ironhak, 2025-01-27:
> Hi [@ironhak](https://github.com/ironhak)
> 
> To be honest, I have no idea. Those dependencies get automatically included when dnf installing the display link driver.
> 
> I am using Wayland, but I can always start an X11 session (or at least I could until Fedora 40)

Got it, and what are the problems you're facing? And were you able to fix them? 
- On KDE: until I unplug the docking station from my laptop Fedora stay stuck on a black screen at boot, when I unplug it takes me to login screen. 
- On gnome: at boot one monitor is just straight up turned off and gets turned on only after I login. 

On Linux Mint I can keep two external monitors at 100hz, here if I keep them at 100hz then one of them simply turns off.

DMario93, 2025-01-27:
I am on Fedora 41 (Gnome with Wayland) and I am using a Dell Docking stations D6000

>  what are the problems you're facing? 

If I try to install the display link for fedora 41 either by downloading the one in releases or by building it from scratch (main branch), it doesn't work at all. 

> And were you able to fix them?

If I instead install the previous version of the display link driver (the one for Fedora 40, that is) then everything works: right now I have keyboard, mouse, two external monitors and ethernet, all connected and working. I followed this [guide](https://alessandroferrari.live/install-displaylink-fedora39/)

However, the external screens do not refresh super quickly. If I type in a black terminal I see the white "trail"

DMario93, 2025-01-31:
Hi @elguero , apologies for chasing you... Is there any update? 

Any other info that I can provide to help you out?

piotrminkina, 2025-03-10:
It's working for me with kernel 6.13.5-200.fc41.x86_64 and `fedora-41-displaylink-1.14.8-1.github_evdi.x86_64.rpm` package (releasted one week ago).

magikmw, 2025-03-19:
I just got it working on Fedora 41 6.13.6-200.fc41.x86_64, but I had to build the package myself from current master branch, as the F41 package from releases didn't work entirely with K17A dock. USB and ethernet did register in dmesg, but no displays.

[As an aside, I had no trouble with DKMS, I already used it for another driver]

I still have no power delivery apparently, but looking at that next.

piotrminkina, 2025-03-20:
@magikmw Regarding the displays, check if it's the same problem as the one I reported in https://github.com/displaylink-rpm/displaylink-rpm/issues/292. Try to call the command `sudo lsusb --verbose`.

magikmw, 2025-03-20:
> [@magikmw](https://github.com/magikmw) Regarding the displays, check if it's the same problem as the one I reported in [#292](https://github.com/displaylink-rpm/displaylink-rpm/issues/292). Try to call the command `sudo lsusb --verbose`.

I did, I scoured all open and closed issues on this repo. Sadly, that one didn't have any visible effect for my case.

elguero, 2025-04-28:
¡Hola!  Sorry for not responding sooner.  Crazy busy lately.

Is there still an issue with the latest release?

closed 2026-04-28 by elguero