displaylink-rpm/displaylink-rpm #292: No display on monitors after reconnect dock station

piotrminkina, 2025-02-04:
Hello,

I would like to report a problem I am having with a DELL Docking Station D6000 on Fedora Workstation 41 (Wayland). I am using version v6.1.0-2 of your driver package.

When I boot the laptop and the docking station is then plugged in, the screen on both connected monitors displays correctly. The monitors are connected via HDMI and DisplayPort. When I unplug the station and plug it back in, the screen on the monitors is no longer displayed. The only thing I've noticed is that the `displaylink-driver.service` after disconnecting from the station is stopped, and after reconnecting the station is not restarted. Even despite manually forcing the service to restart, the screen is still not visible on either of the two monitors. I unfortunately have to restart the laptop to get the screen back.

This is a bit inconvenient because I work in an office where I have to switch to different docking stations when I start a meeting. It's hard to imagine the need to reboot the operating system every time I change workstations. What additional information could I provide that could help solve the above problem?

Regards
Piotr Minkina

rhinehramirez, 2025-03-07:
I'm having the same problem as well.

piotrminkina, 2025-03-10:
All in all, a strange problem. Today, I managed to force it to detect the station in a very strange way after reconnecting it to my laptop. After executing the command `sudo lsusb -v`, it suddenly and automagically detected the station, and then an image appeared on both monitors connected to the station.

I checked the udev scripts to see how the docking station is detected, and noticed that the scripts expect a device with Vendor ID 17e9. Executing the following command should show a similar result to mine:

```
$ lsusb | grep 17e9
Bus 004 Device 006: ID 17e9:6006 DisplayLink Dell Universal Dock D6000
```

Unfortunately, when I disconnect the docking station and reconnect it, the result is empty, which is very strange. Only when I run the aforementioned command `sudo lsusb -v` and then check again with the command `lsusb | grep 17e9` does it show the expected result.

I also noticed that this docking station connected to the ASUS laptop is not detected as a Thunderbolt device, i.e. the `sudo boltctl list --all` command does not show the station in the list. I remember that on my previous Dell Precision laptop, this station appeared as a Thunderbolt device. Maybe there is a problem here? What do you think, @elguero?

I have attached the diagnostic information from dmesg and the next steps I took. [displaylink.log.gz](https://github.com/user-attachments/files/19159773/displaylink.log.gz)

piotrminkina, 2025-03-10:
I am not the first person to have discovered this strange problem. https://unix.stackexchange.com/questions/749970/usb-c-devices-not-being-detected-until-i-run-sudo-lsusb-v

piotrminkina, 2025-03-10:
I am attaching the results of `lsusb -t -v` in different situations.

1. no docking station [lsusb-1-no-docking-station.log](https://github.com/user-attachments/files/19160047/lsusb-1-no-docking-station.log)
2. connected docking station [lsusb-2-connected-docking-station.log](https://github.com/user-attachments/files/19160054/lsusb-2-connected-docking-station.log)
3. forced redetect via `sudo lsusb -v` [lsusb-3-forced-lsusb-v.log](https://github.com/user-attachments/files/19160061/lsusb-3-forced-lsusb-v.log)

piotrminkina, 2025-03-10:
I made a udev rule that executes the command `lsusb --verboose` after every addition or change of any USB device. It works for me. From now on, when I reconnect the docking station, an image appears on the external screens. This should be considered a workaround for the main problem, the cause of which is unknown to me.

```udev
# /etc/udev/rules.d/01-lsusb-hacks.rules
ACTION=="add|change", SUBSYSTEM=="usb", DRIVERS=="usb", RUN+="/bin/sh -c 'lsusb --verbose >/dev/null'"
```

elguero, 2025-04-28:
¡Hola!  Before digging into this too much more, I wonder if this is still happening on the latest release?

piotrminkina, 2025-05-19:
The same problem with DisplayLink 1.14.10-1 on Fedora 41 Workstation on kernel 6.14.6-200.fc41.x86_64 and the same hack forces detection of other docking station features.

elguero, 2026-11-11:
¡Hola!

There have been some updates made to UDEV rules. Do you know if this is still an issue?