displaylink-rpm/displaylink-rpm #295: Fedora 41 kernel 6.13 evdi compilation failure

jvo203, 2025-02-26:
DKMS automatically tried to re-compile the evdi kernel module after today's automatic kernel update from 6.12 to 6.13 in Fedora 41. Attached is a full compilation log.

[evdi.log](https://github.com/user-attachments/files/18980224/evdi.log)

scorpp, 2025-02-26:
same problem with Fedora 40

brigante91, 2025-02-26:
I solved the compilation of evdi 1.14.8 with this command on kernel `Linux fedora 6.13.4-200.fc41.x86_64`

`CFLAGS="-isystem./include -isystem./include/uapi -Werror -Wextra -Wall -Wno-error=missing-field-initializers -Wno-error=sign-compare -Wmissing-prototypes -Wstrict-prototypes -Werror=discarded-qualifiers" make -C module`

elguero, 2025-02-26:
Hola!

Which release is installed on your system?  That should have already been addressed.

func0der, 2025-02-26:
[make.log](https://github.com/user-attachments/files/18989697/make.log)

Same problem here. Running evdi/1.14.7 and compiling for kernel 6.13.4-200.fc41.x86_64

gcalmettes, 2025-02-26:
same error on `6.13.4-200.fc41.x86_64`, and `evdi/1.14.7`

liloumuloup, 2025-02-26:
Hello,

I got the same problem on Fedora 40. 

I had to rollback to 6.12 because i can't work properly without this. 😔 

Here is how i did my rollback in case anyone here needs it : 
* reintall the kernel in latest version :
```
~ > cd $(mktemp -d) \
  && koji download-build --arch=x86_64 --arch=noarch kernel-6.12.13-100.fc40 \
  && rm *debug*.rpm \
  && rm *uki*.rpm \
  && sudo dnf install *
```

* reinstall the displaylink package on latest release :
```
~ > sudo dnf reinstall ~/Downloads/fedora-40-displaylink-1.14.7-4.github_evdi.x86_64.rpm
```

I found these commands here : https://docs.fedoraproject.org/en-US/quick-docs/kernel-installing-from-koji/

jvo203, 2025-02-26:
The version installed: Fedora 41 RPM (https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v6.1.0-2/fedora-41-displaylink-1.14.7-4.github_evdi.x86_64.rpm) from "[v6.1.0-2 (re-released)](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v6.1.0-2)", which is the most up-to-date release version.

cyprox972, 2025-03-01:
For anyone else having this problem fedora by default keeps 3 kernels as backups when the kernel is updated. So you can just choose to boot with 6.12 and display link works like normal, when a proper fix is released you can start using 6.13 again. When you boot press ESC after bios/UEFI and you should enter grub, here you can choose 6.12 instead of 6.13.
If you like you can also use grubby to change the default kernel from a terminal.
This worked for me at least and very easy workaround

DrMaxNix, 2025-03-03:
This seems to be fixed by the new [v6.1.0-3](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v6.1.0-3) release (`displaylink-1.14.8-1`) 🎉

cyprox972, 2025-03-03:
Can confirm it also works for Fedora 41 with the new [v6.1.0-3](https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v6.1.0-3) release

jvo203, 2025-03-03:
Thank you for fixing it.

closed 2025-03-03 by jvo203