displaylink-rpm/displaylink-rpm #296: Screens not refreshing properly

DMario93, 2025-03-06:
Hello and thanks to whoever will help me out 🙏 

**My problem:** External screens seem to not refresh properly leaving trails. See photo and video below to see what I mean by trails.

[image]
https://youtube.com/shorts/kdtKRc3kDDY?si=VNpoO_e-G8qeWZ4V

**My setup:** 
- Fedora 41 (kernel 6.13.5)
- NVIDIA GeForce RTX 4070
- Intel Meteor Lake-P (integrated GPU)
- NVIDIA Drivers: NVIDIA UNIX x86_64 Kernel Module  570.86.16
- DP Link driver: 1.14.8
- Dock: DELL D6000

Additional info:

To check whether the problem was caused by the gpu/gpu driver, I tried to switch between using the NVIDIA card or the integrated one as default, unfortunately the problem persisted in both cases. 

The problem does not present on the display of my laptop.

Also, I noticed this configuration for the driver:

[image]

Could the absence of any acceleration method be the cause? I'm honestly quite ignorant on the subject so just trying to reason logically about it - apologies if makes no sense whatsoever 😂.

elguero, 2025-04-28:
¡Hola! @DMario93 Are you still having issues on the latest release?

DMario93, 2025-04-29:
Hi @elguero thanks for checking in! 

DKMS build fails with kernel 6.14.3 (which I think is tainted as it induces a kernel panic at boot time) and 6.14.4.

6.13.11 is the last kernel I have on which the driver works and the problem reported in this thread is still present.

I'm going to update to Fedora 42 right now. I'll keep you posted

elguero, 2025-10-23:
Hola! The folks a Synaptics are working on something that sounds like your issue:

https://github.com/DisplayLink/evdi/issues/524#issuecomment-3380239598

austins, 2025-12-18:
Same here with Fedora 43 and 6.17.12-300.fc43.x86_64. It's slow to refresh, leaving artifacts, but eventually fully refreshes if static content is displayed.

dra66y, 2025-12-29:
It does more or less fix it, but it adds a ton of cpu overhead

[image]

This is on a ryzen ai max 395, so it has plenty of power and is still using that much cpu

dra66y, 2025-12-29:
The high cpu usage that I reported only happens when there's movement. Moving the mouse, playing a video, moving a window, etc. If the screen is still, there is no cpu usage. According to the "fix" it's flushing the CPU cache to ram constantly, which seems like it would explain high cpu usage.

I'm also on fedora 43