displaylink-rpm/displaylink-rpm #299: Error when updating Fedora 41 to kernel 6.13.7

ZeddieXX, 2025-03-19:
Was on 6.13.6 and during the typyical "sudo dnf update", there was a kernel update to 6.13.7, which seem to have broke booting the OS.

I tried to reinstall the kernel by using "sudo dnf reinstall kernel\*" (searched this in Google), I noticed it produced an error (I may have missed it the first time it updated):

>>> Running post-transaction scriptlet: kernel-core-0:6.13.7-200.fc41.x86_64                                                                    
>>> Non-critical error in post-transaction scriptlet: kernel-core-0:6.13.7-200.fc41.x86_64                                                      
>>> Scriptlet output:                                                                                                                           
>>> Sign command: /lib/modules/6.13.7-200.fc41.x86_64/build/scripts/sign-file                                                                   
>>> Signing key: /var/lib/dkms/mok.key                                                                                                          
>>> Public certificate (MOK): /var/lib/dkms/mok.pub                                                                                             
>>>                                                                                                                                             
>>> Autoinstall of module evdi/1.14.7 for kernel 6.13.7-200.fc41.x86_64 (x86_64)                                                                
>>> Cleaning build area... done.                                                                                                                
>>> Building module(s)....(bad exit status: 2)                                                                                                  
>>> Failed command:                                                                                                                             
>>> make -j12 KERNELRELEASE=6.13.7-200.fc41.x86_64 all INCLUDEDIR=/lib/modules/6.13.7-200.fc41.x86_64/build/include KVERSION=6.13.7-200.fc41.x86
>>>                                                                                                                                             
>>> Error! Bad return status for module build on kernel: 6.13.7-200.fc41.x86_64 (x86_64)                                                        
>>> Consult /var/lib/dkms/evdi/1.14.7/build/make.log for more information.                                                                      
>>>                                                                                                                                             
>>> Autoinstall on 6.13.7-200.fc41.x86_64 failed for module(s) evdi(10).                                                                        
>>>                                                                                                                                             
>>> Error! One or more modules failed to install during autoinstall.                                                                            
>>> Refer to previous errors for more information.                                                                                              
>>> /usr/lib/kernel/install.d/40-dkms.install failed with exit status 11.                                                                       
>>>                                                                                                                                             
>>> [RPM] %posttrans(kernel-core-6.13.7-200.fc41.x86_64) scriptlet failed, exit status 11         

I noticed it mentioned evdi, and it jogged my memory that I installed this before, but had to retrace my steps to try to remember how I got my DisplayLink adapter working.  After some googling (and reading my old notes), I remember it was an RPM I downloaded from this site.  Thankfully, I have a folder I created called "Manually installed RPMs" so I can re-run them to uninstall if I needed to, which has the correct DisplayLink RPM file I used to install (fedora-41-displaylink-1.14.7-4.github_evdi.x86_64.rpm).

The latest release is 1.14.8 so it's not all that much newer.  Not sure if that works with the newest kernel (and going forward, newer kernel versions), but wanted to at least document it here.

After uninstalling the RPM, I reran the kernel install and the initramfs file is no longer missing (and no error during the kernel update).  I have yet to reboot just yet, but without any errors, I'm guessing it should finally boot with the newest kernel.

I only use the adapter once in a rare while so I will leave it uninstalled until the next time I need to use it.

Any way to get this driver in Fedora's repo so it can get updated along with the standard dnf updates?

feroxib, 2025-03-24:
having some trouble here too.
I also noticed the problem with 6.13.7-200.fc41.x86_64 and no luck with 6.13.8-200.fc41.x86_64 either.

Any advice how to get going again?

ClaudiuVapit, 2025-03-28:
Same situation for me on Fedora 41, but I've got it working after remove - install the rpm package with dnf ( sudo dnf remove displaylink -y      ---      sudo dnf install ./fedora-41-displaylink-1.14.8-1.github_evdi.x86_64.rpm ) in the middle if the installation process the external display will power on but right after that everything will freeze  and nothing seems to work getting out of that, had to force power off my laptop and then after start again it works fine :)

elguero, 2025-04-28:
¡Hola! @ZeddieXX Are you still having issues with the latest release?

domerich, 2025-04-29:
I can confirm 13.5 was the last working Kernel

bencardoen, 2025-05-15:
Not sure if this data point helps, but kernel 6.14.5 works for me, (Fedora 40, latest release 1.14.9-2).

ClaudiuVapit, 2025-05-15:
displaylink driver version "fedora-42-displaylink-1.14.9-2.github_evdi.x86_64" work perfectly Fedora 42 - kernel 6.14.6-300.fc42.x86_64, with no sign of any problem.

firewallzer0, 2025-05-18:
Since updating to Fedora 42 I have not been able to get the displaylink monitors to work, but they will work if I run Fedora 41 kernel that I have not yet removed. I tried 
`fedora-42-displaylink-1.14.9-2.github_evdi.x86_64.rpm` 
as well as 
`fedora-42-displaylink-1.14.10-1.github_evdi.x86_64.rpm`

With no luck.

I looked at the journalctl for `displaylink-driver.service` and saw this

```
May 17 21:23:30 fedora systemd[1]: displaylink-driver.service: Scheduled restart job, restart counter is at 119.
May 17 21:23:30 fedora systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
May 17 21:23:30 fedora modprobe[20994]: modprobe: FATAL: Module evdi not found in directory /lib/modules/6.14.6-300.fc42.x86_64
May 17 21:23:30 fedora systemd[1]: displaylink-driver.service: Control process exited, code=exited, status=1/FAILURE
May 17 21:23:30 fedora systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
May 17 21:23:30 fedora systemd[1]: Failed to start displaylink-driver.service - DisplayLink Driver Service.
May 17 21:23:36 fedora systemd[1]: displaylink-driver.service: Scheduled restart job, restart counter is at 120.
May 17 21:23:36 fedora systemd[1]: Starting displaylink-driver.service - DisplayLink Driver Service...
May 17 21:23:36 fedora systemd[1]: displaylink-driver.service: Control process exited, code=killed, status=15/TERM
May 17 21:23:36 fedora systemd[1]: displaylink-driver.service: Failed with result 'signal'.
May 17 21:23:36 fedora systemd[1]: Stopped displaylink-driver.service - DisplayLink Driver Service.
```

When I ran a `mlocate` on evdi
 I do find it in `/usr/lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko.xz`
and in `/lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko.xz`

modinfo makes it look like it is working:
```
root@fedora:~# modinfo evdi
filename:       /lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko.xz
license:        GPL
description:    Extensible Virtual Display Interface
author:         DisplayLink (UK) Ltd.
import_ns:      DMA_BUF
import_ns:      DMA_BUF
depends:        
name:           evdi
retpoline:      Y
vermagic:       6.14.6-300.fc42.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        66:32:7F:A5:3D:85:2C:D9:EA:BE:46:2C:03:BB:6A:BA:08:5F:9E:80
sig_hashalgo:   sha512
signature:      8B:31:E1:16:14:CB:38:0E:D1:A0:D5:5B:D3:46:85:D6:30:E7:09:F8:
                E8:E5:9C:26:96:56:50:E7:AE:BF:06:0B:27:CC:7E:3F:20:76:A7:7C:
                1A:99:BC:D3:88:F2:30:06:6D:26:C2:16:27:F7:BD:CF:2E:D9:F4:C9:
                F6:70:39:5D:46:62:2F:18:C3:2E:94:B2:81:95:EC:89:7B:26:C7:C5:
                F0:86:CC:84:E8:B8:FF:6A:1E:FF:2A:A0:92:6A:DE:F0:DD:73:F3:46:
                92:29:9B:98:84:DE:B2:B0:7C:E4:D4:B6:A5:6F:EB:84:D1:3F:F6:E9:
                E3:7A:BF:0A:6D:39:D2:23:0C:61:8C:A0:BD:5F:AC:5F:19:78:9F:FF:
                20:33:4C:D5:0C:74:43:6E:4C:0F:EC:A3:ED:DD:B6:B7:40:85:AB:7C:
                0E:24:B8:4E:3B:35:16:55:EE:E3:93:A3:B7:5A:DA:4F:D9:82:9C:8C:
                57:C8:6E:53:8C:C7:AD:E8:E2:18:F5:8B:76:B9:0E:8C:C7:1A:30:8F:
                57:95:C4:7F:20:70:CD:82:FF:DE:4F:0E:FF:68:C1:E2:BC:03:6F:8B:
                8A:5D:EB:80:9A:7B:41:C5:DA:2E:4B:AD:84:4F:5F:2F:51:97:95:5F:
                CB:82:10:DE:01:07:81:14:48:95:A0:99:EE:E1:97:9C
parm:           initial_loglevel:Initial log level (int)
parm:           initial_device_count:Initial DRM device count (default: 0) (ushort)
```

but when I try to load the module...

```
root@fedora:~#` insmod evdi
insmod: ERROR: could not load module evdi: No such file or directory

root@fedora:~# insmod /lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko.xz
insmod: ERROR: could not insert module /lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko.xz: File exists

[... 15 lines cut ...]
.:ccccccccccccc;KMMc;cc;xMMc;ccccccc:.    Shell: bash 5.2.37
,cccccccccccccc;MMM.;cc;;WW:;cccccccc,    Display (VSCF433): 1920x1080 @ 60 Hz in 24" [External]
:cccccccccccccc;MMM.;cccccccccccccccc:    Display (VX2260WM): 1920x1080 @ 60 Hz in 22" [External]
:ccccccc;oxOOOo;MMM000k.;cccccccccccc:    Display (VSCF433): 1920x1080 @ 60 Hz in 24" [External]
cccccc;0MMKxdd:;MMMkddc.;cccccccccccc;    Display (VG259QR): 1920x1080 @ 165 Hz in 25" [External] *
ccccc;XMO';cccc;MMM.;cccccccccccccccc'    DE: KDE Plasma 6.3.5
ccccc;MMo;ccccc;MMW.;ccccccccccccccc;     WM: KWin (Wayland)
ccccc;0MNc.ccc.xMMd;ccccccccccccccc;      WM Theme: Breeze
cccccc;dNMWXXXWM0:;cccccccccccccc:,       Theme: Breeze (Dark) [Qt], Breeze [GTK3]
cccccccc;.:odl:.;cccccccccccccc:,.        Icons: breeze-dark [Qt], breeze-dark [GTK3/4]
ccccccccccccccccccccccccccccc:'.          Font: Noto Sans (10pt) [Qt], Noto Sans (10pt) [GTK3/4]
:ccccccccccccccccccccccc:;,..             Cursor: breeze (24px)
 ':cccccccccccccccc::;,.                  Terminal: konsole 25.4.1
                                          CPU: AMD Ryzen 9 5900X (24) @ 4.95 GHz
                                          GPU: NVIDIA GeForce RTX 3080 Ti [Discrete]
                                          Memory: 7.02 GiB / 62.69 GiB (11%)
                                          Swap: 0 B / 40.00 GiB (0%)
                                          Disk (/): 133.07 GiB / 173.47 GiB (77%) - btrfs
                                          Disk (/mnt/alpha): 1.42 TiB / 1.82 TiB (78%) - fuseblk
                                          Disk (/mnt/backups): 225.80 GiB / 1.36 TiB (16%) - fuseblk
                                          Disk (/mnt/holodeck1): 89.91 GiB / 915.82 GiB (10%) - ext4
                                          Disk (/mnt/holodeck2): 804.31 GiB / 931.51 GiB (86%) - fuseblk
                                          Local IP (enp6s0): 192.168.101.50/24
                                          Locale: en_US.UTF-8
```

bencardoen, 2025-05-18:
@firewallzer0 I remember seeing an issue with modules failing in some recent kernel versions on Fedora due to weird issues with xz decompression, that can be a lead?
https://discussion.fedoraproject.org/t/cant-boot-kernel-6-14-x-on-fedora-41-machine-with-zfs-disks/152289

firewallzer0, 2025-05-18:
So that was last night, I just booted up and everything is working today. I currently have `fedora-42-displaylink-1.14.9-2.github_evdi.x86_64.rpm` installed. Think I am going to leave it alone for now. 

So I guess a reboot fixed my issue. 🤦

mira-miracoli, 2025-05-26:
ok, so I was able to fix it, but in an ugly way:
I needed to unpack the evdi module 
~~~
unxz /lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko.xz
~~~
after every reboot, I have to do the following
~~~
insmod /lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko
/usr/src/kernels/6.14.6-300.fc42.x86_64/scripts/sign-file sha256 ./MOK.priv ./MOK.der /lib/modules/6.14.6-300.fc42.x86_64/extra/evdi.ko
~~~

elguero, 2025-10-23:
Hola. Is it okay to close this issue out?

@mira-miracoli You shouldn't have to do that everytime.  It does sound like maybe your DKMS key has not been enrolled through Mok Manager.

elguero, 2025-11-23:
¡Hola!

I am going to close this issue out.

If you still need assistance, please try the latest release and submit newer logs or messages in a new issue.

Thank you!