displaylink-rpm/displaylink-rpm #304: On Fedora 42 no screen with DisplayLink 6.1.1 with DELL D6000

domerich, 2025-04-28:
This is the follow up ticket with the current version I get the same error as previously.
This is a clean install of Fedora 42 on a Latiude 5480 (yes old but running quietly)

`>>> Running post-install scriptlet: displaylink-0:1.14.9-2.github_evdi.x86_64                                                                                                                                       
>>> Non-critical error in post-install scriptlet: displaylink-0:1.14.9-2.github_evdi.x86_64                                                                                                                         
>>> Scriptlet output:                                                                                                                                                                                               
>>> Warning: The unit file, source configuration file or drop-ins of displaylink-driver.service changed on disk. Run 'systemctl daemon-reload' to reload units.                                                     
>>> Job for displaylink-driver.service failed because the control process exited with error code.                                                                                                                   
>>> See "systemctl status displaylink-driver.service" and "journalctl -xeu displaylink-driver.service" for details.                                                                                                 
>>>                                                                                                                                                                                                                 
>>> [RPM] %post(displaylink-1.14.9-2.github_evdi.x86_64) scriptlet failed, exit status 1  `

elguero, 2025-04-28:
¡Hola!  Would you be able to provide the logs?

Thank you.

domerich, 2025-04-29:
> ¡Hola! Would you be able to provide the logs?
> 
> Thank you.

buen dia, can you access this? https://pastebin.com/EgGVRakn

elguero, 2025-04-29:
Yes, I can access the pastebin.

Your system has secure boot enabled. Do you recall setting up a cert to be used for signing with DKMS?

I see this
```shell
[   31.555863] Loading of module with unavailable key is rejected
[   36.802736] Loading of module with unavailable key is rejected
[   41.861481] Loading of module with unavailable key is rejected
[   47.064684] Loading of module with unavailable key is rejected
[   52.177376] Loading of module with unavailable key is rejected
[   53.378291] evm: overlay not supported
[   57.320257] Loading of module with unavailable key is rejected
[   60.808665] input: Dell Mouse MS5320W Mouse as /devices/virtual/misc/uhid/0005:413C:250A.0003/input/input17
[   60.808918] input: Dell Mouse MS5320W as /devices/virtual/misc/uhid/0005:413C:250A.0003/input/input18
[   60.809390] hid-generic 0005:413C:250A.0003: input,hidraw2: BLUETOOTH HID v0.35 Mouse [Dell Mouse MS5320W] on 00:21:6a:fa:99:3d
[   62.591877] Loading of module with unavailable key is rejected
``` 
A module is not being loaded because the signing key is not enrolled, which I have a feeling it is the `evdi` module.

If you look at the README, it describes how to enroll the MOK key for the first time.  After that is done for the first time, when future updates are installed, DKMS will use that key to sign the updated module and it will be loaded by the kernel.

https://github.com/displaylink-rpm/displaylink-rpm?tab=readme-ov-file#secure-boot-on-fedora

Let me know how that goes.

domerich, 2025-04-30:
So I tried with the keys but it fails

 mokutil --import /var/lib/dkms/mok.pub
Failed to access kernel trusted keyring: Required key not available

elguero, 2025-04-30:
When using sudo, it should be the password you use to login with.

Does /var/lib/dkms/mok.pub exist?

Disabling secure boot will definitely work and proves that this was the issue.

domerich, 2025-05-01:
so I get this

@fedora:~$ sudo ls /var/lib/dkms
evdi  mok.key  mok.pub

then I entered

@fedora:~$ sudo mokutil --import /var/lib/dkms/mok.pub
input password: 
input password again: 

It seems ths is not your sudo password but you create a password for your mok key, whatever that stuff is.

Then I did the reboot and entered the password (working a German Layout you will have to know the US keyboard layout).

domerich, 2025-05-01:
Then I did another reboot, enabled secure boot, and everything worked.

Thanks man. Not sure why I did not work before. 

[image]

closed 2025-05-01 by domerich