displaylink-rpm/displaylink-rpm #305: [Feature]: xorg package niot needed for Asahi linux

schmidtw, 2025-04-30:
### Are you using the latest DisplayLink driver from Synaptics?

Yes

### Are you using the latest EVDI module?

Yes

### If you are using a DisplayLink device, have you checked
'troubleshooting' on Synaptics website?

Yes

### Is this a kernel space issue? Which kernel space issue?

No

### Is this a user space issue? DisplayLink Manager

No

### If you are having an issue upgrading, have you tried to
do a clean install of the driver?

Yes

### Environment

```markdown
- OS: Fedora 42 Asahi Remix
- Version: 42
- Kernel: 6.14.2-401.asahi.fc42.aarch64+16k
```

### DisplayLink Log

n/a

### DKMS Build Log

n/a

### Steps To Reproduce

Rebuild the rpm.

### Detailed Description

I found that that this `Requires` is not needed for Fedora 42 Asahi Remix & still res my ults in a working driver.  (I was able to test with my `DisplayLink Plugable USBC 6950M`)

https://github.com/displaylink-rpm/displaylink-rpm/blob/340055df11872a76be9d12e67749b4aeef716795/displaylink.spec#L64

This makes since since F42 + Asahi is a Wayland based OS.

mrinaldhillon, 2025-06-06:
@schmidtw  With #307  you can install without Xorg using dnf --setopt=install_weak_deps=False. Flag is only needed till Xorg package is available in Fedora repos.

elguero, 2026-04-28:
¡Hola! This hard require was adjusted. It should be resolved in the latest releases.