displaylink-rpm/displaylink-rpm #308: [Bug]: Fedora 42 - Only 1 Display is recognized

HujinoKun, 2025-06-19:
### Are you using the latest DisplayLink driver from Synaptics?

No

### Are you using the latest EVDI module?

Yes

### If you are using a DisplayLink device, have you checked
'troubleshooting' on Synaptics website?

No

### Is this a kernel space issue? Which kernel space issue?

No

### Is this a user space issue? DisplayLink Manager

No

### If you are having an issue upgrading, have you tried to
do a clean install of the driver?

Yes

### Environment

```markdown
- OS: Fedora 42 fresh install

- Kernel:

uname -a
Linux yuusha 6.14.11-300.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Jun 10 16:24:16 UTC 2025 x86_64 GNU/Linux
```

### DisplayLink Log

```
Deprecated feature: CLEAN (/usr/src/evdi-1.14.10/dkms.conf)
Creating symlink /var/lib/dkms/evdi/1.14.10/source -> /usr/src/evdi-1.14.10
Deprecated feature: CLEAN (/var/lib/dkms/evdi/1.14.10/source/dkms.conf)
Sign command: /lib/modules/6.14.6-200.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module(s).... done.
Signing module /var/lib/dkms/evdi/1.14.10/build/evdi.ko
Deprecated feature: CLEAN (/var/lib/dkms/evdi/1.14.10/source/dkms.conf)
Installing /lib/modules/6.14.6-200.fc41.x86_64/extra/evdi.ko.xz
Running depmod.... done.
Executing post-transaction command................... done.
Module evdi/1.14.9 is not installed for kernel 6.14.6-200.fc41.x86_64 (x86_64). Skipping...

Module evdi/1.14.9 for kernel 6.14.4-200.fc41.x86_64 (x86_64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.14.4-200.fc41.x86_64/extra/evdi.ko.xz
Running depmod.... done.

Module evdi/1.14.9 for kernel 6.14.5-200.fc41.x86_64 (x86_64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.14.5-200.fc41.x86_64/extra/evdi.ko.xz
Running depmod..... done.

Deleting module evdi/1.14.9 completely from the DKMS tree.
Executing post-transaction command....................................... done.
Module evdi/1.14.10 for kernel 6.14.6-200.fc41.x86_64 (x86_64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.14.6-200.fc41.x86_64/extra/evdi.ko.xz
[... 120 lines cut ...]
# CC [M]  .module-common.o
  gcc -Wp,-MMD,./..module-common.o.d -nostdinc -I/usr/src/kernels/6.14.11-300.fc42.x86_64/arch/x86/include -I/usr/src/kernels/6.14.11-300.fc42.x86_64/arch/x86/include/generated -I/usr/src/kernels/6.14.11-300.fc42.x86_64/include -I/usr/src/kernels/6.14.11-300.fc42.x86_64/include -I/usr/src/kernels/6.14.11-300.fc42.x86_64/arch/x86/include/uapi -I/usr/src/kernels/6.14.11-300.fc42.x86_64/arch/x86/include/generated/uapi -I/usr/src/kernels/6.14.11-300.fc42.x86_64/include/uapi -I/usr/src/kernels/6.14.11-300.fc42.x86_64/include/generated/uapi -include /usr/src/kernels/6.14.11-300.fc42.x86_64/include/linux/compiler-version.h -include /usr/src/kernels/6.14.11-300.fc42.x86_64/include/linux/kconfig.h -include /usr/src/kernels/6.14.11-300.fc42.x86_64/include/linux/compiler_types.h -D__KERNEL__ -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -fno-inline-functions-called-once -fmin-function-alignment=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -fno-builtin-wcslen -Wall -Wextra -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wno-error=unterminated-string-initialization -Wframe-larger-than=2048 -Wno-main -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-unterminated-string-initialization -Wno-array-bounds -Wno-stringop-overflow -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -fzero-init-padding-bits=all -Wunused -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -Wno-unused-parameter -g  -fsanitize=bounds-strict -fsanitize=shift    -DMODULE  -DKBUILD_BASENAME='".module_common"' -DKBUILD_MODNAME='".module_common.o"' -D__KBUILD_MODNAME=kmod_.module_common.o -c -o .module-common.o /usr/src/kernels/6.14.11-300.fc42.x86_64/scripts/module-common.c   ; /usr/src/kernels/6.14.11-300.fc42.x86_64/tools/objtool/objtool --hacks=jump_label --hacks=noinstr --hacks=skylake --ibt --orc --retpoline --rethunk --sls --static-call --uaccess --prefix=16  --link  --module .module-common.o
# cmd_gen_objtooldep .module-common.o
   { echo ; echo '.module-common.o: $(wildcard /usr/src/kernels/6.14.11-300.fc42.x86_64/tools/objtool/objtool)' ; } >> ./..module-common.o.cmd
# cmd_gen_objtooldep evdi.mod.o
   { echo ; echo 'evdi.mod.o: $(wildcard /usr/src/kernels/6.14.11-300.fc42.x86_64/tools/objtool/objtool)' ; } >> ./.evdi.mod.o.cmd
# LD [M]  evdi.ko
  ld -r -m elf_x86_64 -z noexecstack --no-warn-rwx-segments --build-id=sha1  -T /usr/src/kernels/6.14.11-300.fc42.x86_64/scripts/module.lds -o evdi.ko evdi.o evdi.mod.o .module-common.o
# BTF [M] evdi.ko
  if [ ! -f /usr/src/kernels/6.14.11-300.fc42.x86_64/vmlinux ]; then printf "Skipping BTF generation for %s due to unavailability of vmlinux\n" evdi.ko 1>&2; else LLVM_OBJCOPY="objcopy" pahole -J -j8 --btf_features=encode_force,var,float,enum64,decl_tag,type_tag,optimized_func,consistent_func,decl_tag_kfuncs --lang_exclude=rust --btf_features=distilled_base --btf_base /usr/src/kernels/6.14.11-300.fc42.x86_64/vmlinux evdi.ko; /usr/src/kernels/6.14.11-300.fc42.x86_64/tools/bpf/resolve_btfids/resolve_btfids -b /usr/src/kernels/6.14.11-300.fc42.x86_64/vmlinux evdi.ko; fi;
Skipping BTF generation for evdi.ko due to unavailability of vmlinux
make[2] : on quitte le répertoire « /var/lib/dkms/evdi/1.14.10/build »
make[1] : on quitte le répertoire « /usr/src/kernels/6.14.11-300.fc42.x86_64 »

# exit code: 0
# elapsed time: 00:00:03
```

### Steps To Reproduce

Use same environment

### Detailed Description

after installing the displaylink-rpm package, my dock (dell d6000) only recognizes one screen out of two

timo-a, 2025-06-24:
I had the same issue with my thinkpad dock, and since the latest fedora update (now `kernel-6.15.3-200.fc42.x86_64`) it works again! I am still on `displaylink.x86_64 1.14.9-2.github_evdi` however.

HujinoKun, 2025-06-25:
> I had the same issue with my thinkpad dock, and since the latest fedora update (now `kernel-6.15.3-200.fc42.x86_64`) it works again! I am still on `displaylink.x86_64 1.14.9-2.github_evdi` however.
> 
> (commands used to find the versions: `rpm -qa kernel` and `dnf list installed "displaylink*"`)

Hello,
You're lucky, for my part it still doesn't work.

mrinaldhillon, 2025-07-05:
Same issue here. DisplayLink Dell Universal Dock D6000 with D6.1.1-1 release on Fedora 42.

Kipzrw, 2025-07-18:
> Hello! Not sure if this is your guys' case or not, but I've experienced this issue when one of the monitors was connected with HDMI and the other with DP - the DP monitor worked, the HDMI one didn't. Dell D6000 dock. Laptop is Zephyrus G14 (2020), if that matters. Fedora 42, GNOME 48 Wayland, 6.15.4-200.fc42.x86_64, evdi 1.14.10-1
> 
> The issue was present only when the docking station was connected to my laptop via the USB-C port that supports DP alt-mode. Both monitors work fine on the other USB-C port that doesn't support alt-mode. However I cannot charge my laptop with that one, so for now the workaround in my case is to connect the HDMI monitor straight to laptop. I don't know how to confirm if the alt-mode is interfering or not, so take this with a grain of salt...
> 
> I also feel like I should mention that there appear to be some workarounds to force detect the monitors on devices with thunderbolt in [#292](https://github.com/displaylink-rpm/displaylink-rpm/issues/292), so you could try those as well. The workarounds did not work for me, but I also do not have thunderbolt.

Interesting, I'll try some of the thunderbolt workarounds. For context. I've got the same setup with one hdmi and one DP monitor. It's the dp one that doesn't work for me.

mrinaldhillon, 2025-07-18:
Hi this probably is Fedora 42 issue. The same issue is reproducible on my
setup when Monitor is directly connected via HDMI port. So, better to reach
out on fedora community. I am using Asahi Fedora 42 on Macbook Pro.

Thank You
Mrinal Dhillon
+18573108902

Kipzrw, 2025-07-19:
> Hi this probably is Fedora 42 issue. The same issue is reproducible on my
> setup when Monitor is directly connected via HDMI port. So, better to reach
> out on fedora community. I am using Asahi Fedora 42 on Macbook Pro.
> 
> Thank You
> Mrinal Dhillon
> +18573108902
> […](#)

Any idea where best to log this bug? I'm very new to using fedora.

mrinaldhillon, 2025-07-19:
https://discussion.fedoraproject.org/

HujinoKun, 2025-10-31:
Up 

@elguero Would it be possible to update the rpm package with the latest evdi update, please?

elguero, 2025-11-23:
¡Hola!

A branch was created to use the newer EVDI driver and the older DLM.

If you could give that a try.  I am going to close this issue out.

If you still need assistance, please open a new issue with the latest logs or messages.

Thank you.

closed 2025-11-23 by elguero